Phil1988/FreeDi #1: [Bug] Restarting Klipper after/during a print causes screen to stop updating correctly

chabad360, 2024-09-17:
**Describe the bug**
If Klipper gets restarted (after a `SAVE_CONFIG` or emergency stop), the display stops updating correctly. I can still click around and change things in Klipper, but if I start a print, it doesn't go to the print screen. If I change the target temperature, it doesn't update the actual temperature.

**To Reproduce**
Steps to reproduce the behavior:
1. Run a print
2. Restart Klipper
3. See if the screen updates at all

**Console output**
```
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
_BootPage: len: 1 data[0]: ee
ActiveScreen: = BootPage
Connected to web services!

G-Code file path is: ~/gcode_files
Printer volume: 325x325x325
Klipper version: v0.12.0-297
X3seriesLCD start
self.lcd.data_update()
An error occurred while calculating the print progress: 'result'
An error occurred while calculating the print progress: 'result'
printjob.name.txt="TR/Binky_Rapid PLA+_1h49m.gcode"
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: complete
Printjob completed
extract_printjob_values: self.printer.file_name: TR/Binky_Rapid PLA+_1h49m.gcode
Exception in thread Thread-4 (periodic_update):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/main.pyx", line 143, in main.X3seriesLCD.periodic_update
  File "build/lcd.pyx", line 923, in lcd.LCD.data_update
  File "build/main.pyx", line 818, in main.X3seriesLCD.lcd_callback
  File "build/main.pyx", line 345, in main.X3seriesLCD.extract_printjob_values
KeyError: 'result'
```

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.01 stable
 - OS: `Armbian 24.2.0-trunk with bleeding edge Linux 6.7.5-edge-rockchip64`
 - Source: CChen616 Image
 - Mainboard: stock Mainboard X-6

chabad360, 2024-09-17:
After reading the log, I tried to restarting X3seriesLCD after clearing the print state, which worked. But it's still a bug, and I'm not sure if it's just two things that are closely related.

Phil1988, 2024-09-19:
Thanks @chabad360 
As I said in discord, I will have a look at it today :)

Phil1988, 2024-09-19:
Hey @chabad360 
unfortunately I cant reproduce this on my side.

Thats how I tested it:
1. I started a print and aborted it by pushing the "emergency stop" on the screen
2. I started a print, and pushed the emergency stop on the screen after it forwarded to the "print completed screen"
3. I started a print with an intentional gcode-error to make it abort and pushed the emergency stop on the screen after it forwarded to the "print aborted screen"

In all cases the screen forwarded to a menu where I was able to "restart firmware" and after pushing that button it worked properly.

Can you please give me more information, maybe also the gcode where it happened and check the terminal while it happens to show me?
For the terminal, please to it similar as you did already, but while the bug happens:
[see here](https://github.com/Phil1988/FreeDi/wiki/Bugs!)

Phil1988, 2024-09-20:
Tried these 3 combinations but this time hitting the emergency button on the Mainsail UI.
Works for me in all 3 cases.

jarretluft, 2024-10-03:
Hi @Phil1988,
I am finding I have the same problem as @chabad360 describes above, only I get it on every print I send to the X-plus3.

What I have been able to debug so far, is that after a fresh boot of the linux emmc and the screen is working properly, I go into debug mode and see that it is outputting a 'self.lcd.data_update()' every 1-2 seconds.

I do a few things on the touch screen to test that it's responsive, I turn on the heated bed briefly, check that the temperature starts to update on screen, then turn it back off.
Then I try to send a print to the printer via OrcaSlicer, it generates the following error:

> printer_callback triggered
> printer_callback triggered
> self.lcd.data_update()
> An error occurred while calculating the print progress: 'result'
> An error occurred while calculating the print progress: 'result'
> printjob.name.txt="Small+Baby+Dragon_PLA_3h32m.gcode"
> Printer state: printing
> Ongoing print detected
> extract_printjob_values: self.printer.file_name: Small+Baby+Dragon_PLA_3h32m.gcode
> Exception in thread Thread-4 (periodic_update):
> Traceback (most recent call last):
>   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.11/threading.py", line 975, in run
>     self._target(*self._args, **self._kwargs)
>   File "build/main.pyx", line 143, in main.X3seriesLCD.periodic_update
>   File "build/lcd.pyx", line 1058, in lcd.LCD.data_update
>   File "build/main.pyx", line 818, in main.X3seriesLCD.lcd_callback
>   File "build/main.pyx", line 345, in main.X3seriesLCD.extract_printjob_values
> KeyError: 'result'
> 

And after this error, the 'self.lcd.data_update()' stops firing, which I suspect is what stops the screen from refreshing beyond this point.

I can restart the X3SeriesLCD service, and the .update() calls start firing again, but only until I try to send another print directly from the slicer, at which point the service errors again, and the screen stops updating any of the values.

A complete debug log can be found here, where I adjust heated bed values on screen to test before trying to send a print from the slicer.
['X3SeriesLCD Debug Log'](https://pastebin.com/EsQ2NWkw)

I am running a Qidi x-plus3 with all stock hardware, the v1.01 of the LCD firmware, and used your FreeQIDI walkthrough to do the rest.  Klipper is v0.12.0-307-g9426485bb, Linux is Armbian-unofficial 24.2.0-trunk bookworm, Fluidd is v1.30.4

Is the actual code behind the X3ServiceLCD or the code behind the actual touch screen firmware available somewhere? I'd be happy to dig deeper and see if I can pinpoint where this might be failing, but I only see precompiled releases in the github repo.

Thanks for any assistance you can provide in troubleshooting this.

Phil1988, 2024-10-07:
Thanks @jarretluft for the detailed log, this helps and I think I see, where this might come from.

The API request to moonraker fails.
It doesnt get any 'result' for the requested file.

Your file name is **Small+Baby+Dragon_PLA_3h32m.gcode**
I assume it has spaces at each "+" sign, so the real name might be "Small Baby Dragon_PLA_3h32m.gcode".
Can you confirm this?

Please try these URLs with your favorite browser and show me the result:
http://_printers_IP_/server/files/metadata?filename=Small+Baby+Dragon_PLA_3h32m.gcode
http://_printers_IP_/server/files/metadata?filename=Small Baby Dragon_PLA_3h32m.gcode
(replace the **_printers_IP_** with its actual IP adress)

If my quess is right, please try to upload a gcode with **no spaces** the same way it does fail now.
(eg name it "**Small_Baby_Dragon_PLA_3h32m.gcode**")

And please tell me if it works this way.

Btw: the "self.lcd.data_update()" every 2s is the normal behavior to push updated values to the screen.
The X3ServiceLCD code and the touch screen firmware isnt available and I dont think that I will publish it.

jarretluft, 2024-10-07:
Thanks for this additional context @Phil1988, I think you've got it partly figured out.

To clarify, the filename is "Small+Baby+Dragon_PLA_3h32m.gcode", with the plus characters in the name.  This was just what the original STL was named, and carried forward into the name of the gcode file after slicing.

When visiting both:
/server/files/metadata?filename=Small+Baby+Dragon_PLA_3h32m.gcode
and
/server/files/metadata?filename=Small Baby Dragon_PLA_3h32m.gcode
I get a json 404 metadata not available response from the api

However, after reading some issue reports on the moonraker github issue tracker, it appers moonraker expects third party API clients to convert these special characters to URL encodings, so the actual API endpoint for this file's metadata through moonraker is:
http://printers_IP/server/files/metadata?filename=Small%2BBaby%2BDragon_PLA_3h32m.gcode
Where %2B replaces the '+' characters, per this [encoding reference](https://www.w3schools.com/tags/ref_urlencode.asp?_sm_au_=iVVDMg0TSmrMV6Dm)
When visiting this encoded URL on my own server, I get a proper JSON response with the file metadata:

`{"result": {"size": 35399508, "modified": 1728309048.029999, "uuid": "18f8b616-5fcb-4d10-9b09-9aa7ffbccc9b", "slicer": "OrcaSlicer", "slicer_version": "2.1.1", "gcode_start_byte": 19636, "gcode_end_byte": 35383021, "layer_count": 133, "object_height": 26.65, "estimated_time": 12621, "nozzle_diameter": 0.4, "layer_height": 0.2, "first_layer_height": 0.25, "first_layer_extr_temp": 230.0, "first_layer_bed_temp": 60.0, "chamber_temp": 0.0, "filament_name": "Qidi Bambu PLA - Black", "filament_type": "PLA", "filament_total": 16176.44, "filament_weight_total": 48.25, "thumbnails": [{"width": 32, "height": 32, "size": 407, "relative_path": ".thumbs/Small+Baby+Dragon_PLA_3h30m-32x32.png"}, {"width": 300, "height": 300, "size": 12330, "relative_path": ".thumbs/Small+Baby+Dragon_PLA_3h30m-300x300.png"}], "print_start_time": null, "job_id": null, "filename": "Small+Baby+Dragon_PLA_3h30m.gcode"}}`

So, I guess i'm wondering if you are currently doing any URL encoding of characters in file names? And if not, would you be able to? 

> Btw: the "self.lcd.data_update()" every 2s is the normal behavior to push updated values to the screen.

yeah, I get this, and this refresh frequency makes sense, but why do the data_update() calls stop when an error is encountered? is there a way to gracefully fail back to the main screen so that current temperatures and other info on screen continues to be accurate, even if print-screen metadata can't be presented?

Also, as a suggestion, is it possible when getting an error response from moonraker to put into the debug log the actual API call that was being made that failed? might help troubleshoot future issues where an unexpected 'result' is received from moonraker.

Anyway, thanks again for your insight on this!

Phil1988, 2024-10-07:
Good catch @jarretluft :)

This also explains @chabad360 error with the "TR/Binky_Rapid PLA**+**_1h49m.gcode".
I am currently not doing any URL encoding, but I will include it to cover this bug.

I will just need a couple of days to get time working on it.

For your question regarding the data_update() stop:
I am not really sure how to answer this :D
Yes, it would be possible to establish a fallback but the use case doesnt seem legit for me.
In these cases the printer status is out of sync with the screen, eg. printing something, but the screen would show the main menu. That doesnt make sense to me.

And you are right. A better error logging would be handy.
My main goal was to make the screen firmware working and publish it to others that might like it.
Because making the screen firmware took a huge amount of my spare time, I couldnt yet include everything I would like to.

closed 2024-10-10 by Phil1988