Phil1988/FreeDi #2: [Bug] Exception errors on running start.py, touchscreen not working

jarretluft, 2024-09-26:
Hi,
I have a Qidi X-Plus 3 with 32mb emmc.

I have followed the step by step guide you produced for FreeQIDI to get to a standard bookworm install with klipper, moonraker, fluidd and crowsnest.

I have then followed the wiki to install FreeDI screen.
I am using the **X3seriesLCD_firmware_v1.01.tft** screen firmware

I have followed the **"installation guide for existing systems"** to install FreeDI
All of this completed without any issues, but trying to run `python3 start.py` I get the following errors and my touch screen is unresponsive, but I do see all the menu items show up:

> Moonraker host address:
> http://127.0.0.1:80
> Connecting to UDS /home/mks/printer_data/comms/klippy.sock
> Connected!
> Exception in thread Thread-1 (run):
> Traceback (most recent call last):
>   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.11/threading.py", line 975, in run
>     self._target(*self._args, **self._kwargs)
>   File "build/lcd.pyx", line 1271, in lcd.LCD.run
> IndexError: bytearray index out of range
> Connected to web services!

Any ideas what could be wrong here?
Thank you!

jarretluft, 2024-09-26:
After further troubleshooting, I think this is an issue with the cable that connects the touch screen overlay to the LCD.  I have removed the LCD, removed/reattached the cable and reinstalled, and things are working as expected now.  Sorry for the bother.

closed 2024-09-26 by jarretluft

Phil1988, 2024-09-26:
I am happy you find the error and that you are using the firmware.
I hope it gives you a nice user experience and you like it!