Phil1988/FreeDi #3: LCD screen stay on connecting klipper

fun74fr, 2024-10-07:
**Describe the bug**
After following "Installation guide for existing systems" when i reboot the printer the LCD screen stay on Connecting Klipper.
I can connect to printer by fluidd and everything works.
Actually only the screen stuck on connecting klipper

**Screenshots or picture**
[image]

[image]

**Console output**
[image]

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.01 stable
 - OS: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_edge_6.7.5.img.xz
 - Source: https://github.com/billkenney/update_max3_plus3
 - Mainboard: stock Mainboard X-4/X-6

Phil1988, 2024-10-07:
I might have gotten it wrong, but your build from https://github.com/billkenney/update_max3_plus3 is not a vanilla mainline klipper, is it?
Looks like it has some modifications to run xindi with the stock screen firmware.

Problem is, that this uses the same serial port and they interfere.
You can either switch back to the stock screen firmware or get a mainline klipper.

My tutorial for that is almost a year old and since 9 month also here:
https://github.com/Phil1988/FreeQIDI

I'll wait for your response, but I am pretty sure this is not a bug related to my software :)

fun74fr, 2024-10-07:
Thank you for your answer

I reput my old EMMC from qidi and after do your tutorial to go to vanilla and come back to you if i have trouble

Phil1988, 2024-10-07:
Please also come back if you dont have trouble :)
The feedback is important for me and others that might come across this problem.

Phil1988, 2024-10-07:
btw: You can also take my [full image](https://github.com/Phil1988/FreeDi/releases/tag/v1.00) and save some time :)

fun74fr, 2024-10-07:
I have download your complete image but i think i made something wrong because i only see in fluidd the armbian version and not mcu and other.
But actually its 1 o'clock in the morning so i go to bed and retry tomorrow morning.

Phil1988, 2024-10-08:
my image doesnt even have fluidd included, so yeap! - I also think you made something wrong :D

Phil1988, 2024-10-11:
Hey @fun74fr,
have you been successful?

If not, tell me whats the problem and I happily try to help you.

If everything work for you, please tell me that also and I would appreciate a star here on git so this project gets more attention :)

fun74fr, 2024-10-11:
No i haven't success.
So for the moment i run my printer with the method of bill kenney.

Next week i retry with your method and keep you informed.

fun74fr, 2024-10-25:
HI
I have made the modification today with your debian image. (now it works)
I have also made the freedi on my screen and it works.
but i'm actually trying to adjuste the Z-offset but when i touch the arrow for upper it goes down and the invert on down button.

Phil1988, 2024-10-28:
Thanks for the feedback.
This issue is then closed.

For the z-offset, please open a new one and add pictures where you explain what it does.
Thanks :)

closed 2024-10-28 by Phil1988