Phil1988/FreeDi #4: Full Image Download

Unicorn3D, 2024-10-11:
''I created an image with a fresh and pure system for y'all.
Unfortunetly its 7.28GB (~5.6GB when flahed, so 2.3GB free disk space). This exceeds the maximum file size here on github, so I cant share it with you. If you have an idea, how I can make it availible for you, reach out to me.''

Answer. You could upload the files to google drive and share the link?

would like to try it out. Thanks for your work

Phil1988, 2024-10-11:
Even better!
Its already here in the "Releases" section.

I found a way, but forgot to update the wiki page.
Updated now:
[see here](https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems)

Please tell me how it worked out here after you made it :)

Unicorn3D, 2024-10-12:
I found it but now I have the following problem. when I enter ls /dev/serial/by-id/* I get the following:ls: cannot access '/dev/serial/by-id/*': No such file or directory
mks@mkspi:~$

Unicorn3D, 2024-10-12:
[image]

Unicorn3D, 2024-10-12:
here is the proof that it worked
[image]

Unicorn3D, 2024-10-12:
[image]

Unicorn3D, 2024-10-12:
[image]

Phil1988, 2024-10-13:
Lets check a few things please.

what does this return:
`lsusb`
and
`ls /dev/serial/by-path/*`

Please show me your [mcu MKS_THR] and [mcu] section.

Have you rebooted the system?

Unicorn3D, 2024-10-14:
[image]

Unicorn3D, 2024-10-14:
[moonraker.log](https://github.com/user-attachments/files/17360285/moonraker.log)
[klippy.log](https://github.com/user-attachments/files/17360286/klippy.log)

Unicorn3D, 2024-10-14:
so. now i've redone everything. it's running again. then i heated the hotend, suddenly no more power. printer restarts, again no connection to the toolboard. can it be that the hotend is defective? i have installed v2.3 or something like that.

Phil1988, 2024-10-14:
Yeap, the toolhead should have been visible via lsusb, which is not for you.
You should see a device "OpenMoko".
Most likely your cable to the toolhead or the board itself might be defective.

Maybe you can check that on your side (checking the wire connection).

Unicorn3D, 2024-10-14:
but the toolhead works. i have checked everything... if i redo everything everything works again. Now I have restarted, the toolhead is not accessible again

Unicorn3D, 2024-10-14:
as soon as i reflash the emmc everything works again

Unicorn3D, 2024-10-14:
ok probably the toolhead board is broken. i have now installed another one that i still had there. that works. now i have one more question.
if i order the new board from qidi does it also work? that is a bit different. v1.2 

[image]

Phil1988, 2024-10-14:
In general other hardware is not a breaking thing for my software as it uses official moonraker API.
As long as your klipper config is working and you have every mandatory section included in your printer.cfg there should not be any problem.

So in short: 
This toolheadboard shouold work,

I am closing this issue now as your issue is not related to the software :)

closed 2024-10-14 by Phil1988