Phil1988/FreeDi #5: [Bug] Config upload

Unicorn3D, 2024-10-11:
can you please upload your config files? that would help a lot. unfortunately the folder is empty.

Phil1988, 2024-10-11:
Hey @Unicorn3D 

I currently only have a basic config for the X-Max3.
Do you own a X-Plus3?

If yes, you need to adjust the `position_max:` at the "# Drives" section.
I hope you find it very much organized so you can find all you need.
If you have a working basic config for it, it would be nice to share it, so we can publish it here for others :)

If you have problems, reach out to me here again.
If you like it, please report that also back and I would appreciate a star here on git :)

I changed the topic name, because its not really a [bug] in my opinion.

closed 2024-10-14 by Phil1988

Unicorn3D, 2024-10-14:
no a x-max 3

Phil1988, 2024-10-14:
That makes it even easier fo you.
As you already flashed my full image, the config is already setup for you there.

You can also find the config seperately here:
https://github.com/Phil1988/FreeDi/tree/main/config_section

Have fun :)