Phil1988/FreeDi #6: [Feature request]

jarretluft, 2024-10-12:
Hey @Phil1988, this is more of a question to see if you have explored this at all than an actual issue.

I'm wondering, do you know if is it possible to put katapult bootloader on the mainboard MCU as well to allow for easier flashing of klipper updates in the future? (as opposed to having to access the SD card slot on the mainboard anytime we want to update klipper on the mainboard and toolhead mcu's)

I see from the katapult project, that it lists STM32F series chips, and when exploring the options in the make menuconfig in katapult I see options for STM32F401,  which is what moonraker lists as the chip of the mainboard MCU.

Let me know if you have any thoughts on this, or if you explored this and determined it is not possible.
Thx!

Phil1988, 2024-10-13:
Yes I digged into it and I was unable to flash it that way.
I am not sure what the cause of that.
If you make it work, I would appreciate a guide how you succeeded.
I have no plans to check it again on my side.

A kind warning:
Please only to this kind of things if you are familiar with the procedere and you know what you are doing.
Its not at all connected with this project here, but one guy reached out to me who tinkered a bit this his printer and brocked the mainboard.
I him and the mainboard is working again, but if the dump/flashing fails it can cost you an new mainboard (currently around USD 240).

I am closing this issue now, because I will not support this feature in the near future (simply because the risk is too high).
Feel free to discuss it here anyway :)

closed 2024-10-13 by Phil1988

jarretluft, 2024-10-22:
I appreciate that you don't want to support this due to the risk. :)

Would you be willing to share more information on your attempts?

When you tried, what did you try and how did it fail? Like, did you compile katapult with deployment enabled then rename deployment.bin to X_4.bin and try to flash it through the SD card?
Or something else?

I'm curious how you and the other person got their mainboard working again from a bricked state - do you know what the 6 pin header above the MCU chip is? I'm trying to find a way to stlink into the chip and restore a backup if something goes wrong, but the schematic doesn't give any info on that specific header :(

I tried reaching out to cchen616 on the PLUS3 discussion area, asking about a way to software flash the STM32F4 like how their own firmware updates seem to do it, but did not get a response :( 
[https://github.com/QIDITECH/QIDI_PLUS3/discussions/38](https://github.com/QIDITECH/QIDI_PLUS3/discussions/38)

My progress so far, is i've been able to read the current flash out into a backup.bin file, and I can connect and flash via stm32flash (i flashed the backup back onto the chip to confirm that it would work) but this only works after I manually reset the MCU and enter bootloader mode by holding down the BOOT-0 button on the mainboard while resetting.  

I was not able to get USB DFU mode to find the chip through the usb c port that the schematic shows is wired to the STM32 chip.  I hooked up to my laptop connecting to that USB port but the STM32CubeProgrammer tool didn't find any DFU.

I can build katapult with deployment option enabled for the STM32F401, but I hesitate to try and rename deployer.bin to X_4.bin and flash via sd card, because I have no clear path to recovery if that replaces the bootloader and stmflash32 is no longer able to connect for me to flash my backup back to the chip.  I would really like to know if the 6 pin header above the MCU could be used to re-flash it in a worst case scenario, before I try anything further!

Would love to get more info from you on what you tried that didn't work, if you're willing to share.

jarretluft, 2024-10-24:
Well, I got past my hesitation and flashed with the katapult bootloader and it worked.
So my stm32f401 mcu is now running katapult.  When I did make menuconfig I selected for it to build the deployment app, and in the /out directory found both the katapult.bin and the deployment.bin files were created.  It was the second one, deployment.bin that I renamed to X_4.bin and put on the SD card for the initial flash.  This is the one that replaces the original QIDI code in the first 32kb bootloader area of flash memory - the katapult.bin only puts the code at the start of main flash memory after the bootloader area.

The good news also, is that even after flashing to katapult, stm32flash will still connect to it when resetting the MCU with boot-0 held down, which means I can easily copy my full backup of the QIDI original firmware back should I ever need to.

Given that progress, I unfortunately still can only flash the klipper software using flashtool.py after I double press the reset button to put it into the katapult bootloader mode.  Not quite the hands-off, software only approach I was going for.

I am now trying to find out what method I could use to get it into bootloader mode and ready to accept new firmware without the manual double-tap of the reset.
With the RP2040 i have to pass it a command to put it in bootloader mode, before katapult can flash new klipper updates to it, and was hoping for something similar with the STM32 mcu.

I will probably have to reach out to the katapult team on github to see if there are any options for getting into bootloader mode via software commands.

Phil1988, 2025-07-13:
Can you guys @urche0n-82 @DrFate09 tell me what benefit you have with katapult on the mainboard MCU?
I mean as long as flashing needs physical access to the board, I dont see much benefit over using a microSD to flash it.

urche0n-82, 2025-07-13:
The benefit is that once katapult is on the mainboard mcu you **can** flash without physical access.

However, to achieve this you need to use an altered version of katapults flashtool.py that properly puts the serial device into bootloader mode.

A pull request to add the feature to the main katapult repo is here but still hasn’t been accepted and merged.

https://github.com/Arksine/katapult/pull/142

I continue to use this method successfully to flash my mcu with Klipper updates.