Phil1988/FreeDi #7: [Bug] Start and End G-Code ORCA SLICER

Unicorn3D, 2024-10-14:
What exactly do I have to enter as Start and End G-Code in Orca Slicer?

[image]

Phil1988, 2024-10-14:
Hello @Unicorn3D ,
I think this is getting out of hand :)

I am happily helping anyone with problems related to the software and also further related to the OS and klipper config.
But it seems as this is your first point of contact for any problems.
I can not give a non-profit "everything support" here ^^
I hope you understand it and you please only open issues in the future that are either totally related to my software or you cant fix after you really tried and searched for a solution yourself.

I didnt do anything fancy for Orca.

Start gcode
```
G28
G0 Z50 F600
M190 S[first_layer_bed_temperature]
G28 Z
G29 ; mesh bed leveling ,comment this code to close it
G0 X0 Y0 Z50 F6000
M109 S[first_layer_temperature]
M83
G0 Z5 F1200
G0 X{first_layer_print_min[0]} Y{max(0, first_layer_print_min[1] - 2)} F12000
G0 Z0.2 F600
G1 E3 F1800
G0 Z0.3 F600
G1 X{min(first_layer_print_min[0] + 30,print_bed_max[0])} E6 F600
```

End gcode:
`PRINT_END`

closed 2024-10-14 by Phil1988

Unicorn3D, 2024-10-14:
ok i understand thanks for your help. but that was just a question because the problem still occurs. it shows the following problem. isn't that due to the firmware?

Unicorn3D, 2024-10-14:
it could have been that you are aware of the problem. we are now a total of 5 people who are experimenting with your firmware. as soon as i have found the problem i will let you know.

Unicorn3D, 2024-10-14:
so it makes some mistake with the z offset. i have now set the minimum distance of stepper Z to a minus value so that i can set the first layer. maybe you will find out something in the future. but it does something wrong with the z offset. it is at 0 after setting and the nozzle is still 1.5mm away from the bed.