Phil1988/FreeDi #8: [Bug]

hypdal, 2024-10-17:
Hi,

first of all, thank you for your great work!

I have a Qidi X-MAX3 with 32mb emmc and installed FreeDi according to your "Installation guide for fresh systems".

I noticed that after the update the fan2 no longer works. This is the fan that cools the mainboard.
Is it actively temperature-controlled? Where can this be configured?
With the original firmware, it runs continuously, so I could simply connect it to 24V.

 - X3seriesLCD firmware version: v1.02
 - Version: v0.12.0-289-g08a1c9f12
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: FreeDi image
 - Mainboard: stock Mainboard Qidi X-MAX3

Best regards
   Chris

Phil1988, 2024-10-18:
Hey and thanks for the feedback.
This is indeed very ... "interessting".
The mainboad fan, physically connected to the fan2 mainboard pins is an always on fan - there is nothing to switch it on or off.

Are you **really sure**, that with the original firmware the mainboad fan does work?
It could also be a defective fan, which is my guess as I saw a couple of users with the same problem.

It would be nice if you can check that with a replacement fan and maybe check the pins of the mainboard with a multimeter if you know what you are doing and you have the knowledge not to damage your mainboard and other electronics.

Btw: How did you know about this project?
Im asking because its still a bit "under the radar" as I havent published it actively and find it interessting how you came here :)

jhook-met, 2024-10-18:
Hi, I actually have the same issue. I checked both of the 24V sockets at the top of the board and neither of them have power.

I'll answer your other questions as well.
I ran across this project just poking around after discovering the freeQIDI repo.

The Wiki was great, the only thing that took me a while to find was how to flash the screen in the installation guide.

hypdal, 2024-10-18:
Many thanks for the quick response and @jhook-met for the fast solution! Yes, a defective fan could be ruled out and pin:PC4 controls this fan (I don't know yet whether I want this as a heater fan :-)

I had bookmarked your FreeQidi guide and when my EMMC memory arrived I was very happy to find your new project FreeDi (although I have already built and tested a RasperryPi with TFT and KlipperScreen).

Your wiki was really helpful, the only thing missing is the hint at the end to set the [mcu MKS_THR] serial: /dev/serial/by-id/... (although of course this is in all tutorials and I had simply forgotten - maybe I was too tired)

Phil1988, 2024-10-19:
Thank you @hypdal and @jhook-met for your feedbacks.

Regarding the onboard fan: It works for me and I dont have PC4 configured.
I guess you do have a different mainboard?
Can you please tell me the mainboard version and revision, so that we can help others better in the future?
For me, I am having an X-6 mainboard (I will check if I find a revision number somewhere next week).

About the tutorials:
@hypdal 
I noticed the the missing hint for the [mcu MKS_THR] section and I added it last week.
See [here](https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#flashing-the-thr-toolhead) at the **"Flashing the THR toolhead"** chapter.
Probably it was later then you following the guide.

@jhook-met 
Yes you are right.
Looks like I only put it in the **Installation guide for existing systems** and not at the **Installation guide for fresh systems**.
Thanks for your feedback. I will add it there and also rework the wiki and guide section the coming week.

I will leave this issue open for now, as I would like to know the reason for having this PC4 pin for some models.

jhook-met, 2024-10-20:
My main board is an X-6 V1.0
[image]

Phil1988, 2024-10-20:
Thats interesting, because mine is also an  X-6 V1.0
[image]

Where exactly is your mainboard fan connected to?
Mine is here:
[image]

hypdal, 2024-10-20:
Hi,
I also have the mainboard version X-6 1.0, but the name on the fan is FAN2

[image]

Phil1988, 2024-10-20:
That is super weird!
So QIDI has managed to label different hardware identically :(

@hypdal 
Am I allowed to use your picture for a the guide section?

I will close this issue report now as we know the source of this issue.

closed 2024-10-20 by Phil1988

hypdal, 2024-10-20:
@Phil1988 

you can certainly use the image freely for your needs.

Phil1988, 2024-10-20:
Thanks a lot!
I hope it may help others.

Phil1988, 2024-12-22:
@hypdal and @jhook-met :
I got a message from a kind user via discord.

Please change your fan config to this:
```
[temperature_fan MCU_fan]
sensor_type: temperature_host #This is what the MKS's Temperature sensor is known as.
pin: PC4                      #Set this to your fan pin
max_temp: 80.0                #Set below 85C, MCU can be damaged at 85C
target_temp: 45.0             #Set this to your preferred running temperature
min_temp: 0
shutdown_speed: 0.0
kick_start_time: 0.5
off_below: 0.19
max_speed: 1.0
min_speed: 0.0
control: pid
pid_Kp: 2.0
pid_Ki: 5.0
pid_Kd: 0.5
pid_deriv_time: 2.0
```

Reason is, that the heater_fan is based on the nozzle temp.
With this solution it's linked to the host cpu temp (RK3328), what should be the right way to do it :)

markblokpoel, 2025-10-28:
> [@markblokpoel](https://github.com/markblokpoel) This only applies for the X-Smart 3 as the Plus and Max are using the FAN1 port for the sidefan.
> 
> For you I recommend to use this, but with PA8:
> 
> [FreeDi/config_section/X-Plus3/printer.cfg](https://github.com/Phil1988/FreeDi/blob/6813607480dd301f92329138aea6c89f2e8eac36/config_section/X-Plus3/printer.cfg#L312)
> 
> Line 312 in [6813607](/Phil1988/FreeDi/commit/6813607480dd301f92329138aea6c89f2e8eac36)
>  ## Controllable Mainboard cooling fan 
> 
> In combination with this at the PRINT_START:
> 
> [FreeDi/config_section/Plus4/macros.cfg](https://github.com/Phil1988/FreeDi/blob/6813607480dd301f92329138aea6c89f2e8eac36/config_section/Plus4/macros.cfg#L463)
> 
> Line 463 in [6813607](/Phil1988/FreeDi/commit/6813607480dd301f92329138aea6c89f2e8eac36)
>  SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=Mainboard_fan TARGET=1                                              ; Decrease the mainboard target temp to fully turn the fan on 
> 
> And that at the PRINT_END:
> 
> [FreeDi/config_section/Plus4/macros.cfg](https://github.com/Phil1988/FreeDi/blob/6813607480dd301f92329138aea6c89f2e8eac36/config_section/Plus4/macros.cfg#L528)
> 
> Line 528 in [6813607](/Phil1988/FreeDi/commit/6813607480dd301f92329138aea6c89f2e8eac36)
>  SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=Mainboard_fan TARGET=45                                              ; Reset the mainboard fan temp 
> 
> It is superior then the implementation of mendeste as it does not only turn the fan off/on when the steppers are active but also when the actual CPU temp is too high (so it cools the steppers and the CPU whenever needed).

This is the way. Indeed I only took the pin from mendeste and the freedi config. Works a treat!