Phil1988/FreeDi #9: [Bug] storage size is limited to 6.7GB with 32GB EMMC

jhook-met, 2024-10-18:
**Describe the bug**
Flashed the provided image file to my 32GB EMMC. I am left with a 28.9GB unallocated partition on my EMMC that can't be accessed.

**Expected behavior**
All 32GB of storage available for use.

**Screenshots or picture**
[image]

**Base (please complete the following information):**
 - X3seriesLCD firmware version: V1.02
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: FreeDi image
 - Mainboard: stock Mainboard

hypdal, 2024-10-18:
Hi,

strictly speaking, this is of course not an error and I have solved this (as root) as follows (taken from my bash_history), because I also have a 32GB EMMC module:

wget https://raw.githubusercontent.com/canonical/cloud-utils/main/bin/growpart -P /usr/local/bin/
chmod +x /usr/local/bin/growpart
growpart /dev/mmcblk1 2
resize2fs /dev/mmcblk1p2

Please check in advance whether your partitioning scheme is identical (I assume so).

jhook-met, 2024-10-19:
> Hi,
> 
> strictly speaking, this is of course not an error and I have solved this (as root) as follows (taken from my bash_history), because I also have a 32GB EMMC module:
> 
> wget https://raw.githubusercontent.com/canonical/cloud-utils/main/bin/growpart -P /usr/local/bin/ chmod +x /usr/local/bin/growpart growpart /dev/mmcblk1 2 resize2fs /dev/mmcblk1p2
> 
> Please check in advance whether your partitioning scheme is identical (I assume so). Sorry, I saw your photo too late ;-)

This worked perfectly! Thanks

closed 2024-10-19 by jhook-met