Phil1988/FreeDi #11: [Bug] Z-Offset - Z position number not updating as expected

jarretluft, 2024-10-21:
**Describe the bug**
When running the z-offset option in the second page of the advanced menu, after initial homing it moves to the front-center of the bed and does a probe, then moves slightly to the right - I think to move the nozzle directly over the spot that the probe had just probed.

As this occurs, the screen just shows "Manual Measurement" then "11" in the box underneath.  I assume this box is supposed to report the current distance of the nozzle from the bed?

The first time after it finishes its initial probe, the number reports "11" until I press the "-1" button, then the number jumps from 11 to 5.  This suggests maybe the initial offset was 6mm, but the screen was showing 11mm.
As I proceed to get closer and closer to zero offset, the number goes down in whole digits.  When I get to zero, it stops showing the decimal movements, so I don't see on screen what the actual z offset is, or if it's changing at all.

Anything in the negative just shows "0" on screen.

**Expected behavior**
I guess I was expecting the box at the top to show the current Z position to the same precision as we are able to move (ie: to the 0.000th place).

Further to the above, is it possible to:

1. back the bed off the nozzle by 10-20mm after the "Accept" button is pressed? This would end the offset session with the nozzle more safely away from the bed.  (Same thing if "Abort" is selected)
2. Can the SAVE_CONFIG be automatically applied after a Z-offset session, similar to how it is automatically run after a "Bed Meshing" process in the same menu?

**Screenshots or picture**
Here is my screen at -1.500 offset right before I hit "Accept"
[image]

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.02 stable
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: FreeQIDI wiki install method
 - Mainboard: X-6

Obviously, if i'm misinterpreting how this screen should work and my expected behaviour above is not how it should work, please let me know :).

Phil1988, 2024-10-21:
Thanks for the feedback.

I will check it the next days.

The "11" is the z position in mm.
Might be a delay issue because the position is sent when the "Manual measurement" menu opens (maybe the movement is not finished or a 5mm decrease is performed at calling "PROBE_CALIBRATE").
I will wipe this out.

The negative z position should be visible and I dont see an error on the code basis.
Can you please do the same and when it is on a negative position check the data with your browser and send it here?
Do it like:
http://YourPrintersIPaddress/printer/objects/query?toolhead

The SAVE_CONFIG ist automatically applied after a Z-offset session in v1.03+ (not released tho).
I inserted a 10mm safety distance movement according to your suggestion and you will find it in 1.03.

jarretluft, 2024-10-22:
Thanks, i've done some further testing.

This time around, it never switched to 11mm, it just stayed at 150mm once the probe was done and it was ready for me to adjust the position:

This is the X3 menu screen showing 150mm Z position:
[image]

And this is fluidd screen at the same point, showing a manual probe dialog and a Z: position of 6.557mm
[image]

First time I press the "-1", the X3 screen changes from 150mm to 5mm, and the fluidd manual probe window goes from 6.557mm to 5.557mm.

Finally, once i'm finished adjusting and have moved my nozzle to where I want it, the X3 screen shows:
[image]

And the fluidd dialog shows:
[image]

For what it's worth, when I run your api call to get toolhead position, i get:
"position": [157.80093124994804, 56.752081249946194, 0.26749999997192475, 0.0]
This seems to align with what the fluidd dialog is showing, while the X3 screen is showing "0"

My apologies, in the previous report I had assumed it was going into negative space, but it was just fine tuning on the positive side of 0 that I could not see to the full resolution of the movements.

Hopefully the screenshots attached help illustrate what is happening, I can provide a video as well if that would be useful.

Phil1988, 2024-10-22:
Thanks that makes it clear, why I couldnt find an error related to the negative numbers (I cant see whats not there :D ).
I will fix this with the next release and update you here once finished/released.

leckert123, 2024-10-22:
Same issue. 

Please review whether it's possible to display negative values and decimal places on the screen for Z offset, not just whole, positive numbers.

Phil1988, 2024-11-06:
Its fixed, tested and released :)

Please check out the release (+changelog) :
https://github.com/Phil1988/FreeDi/releases/tag/v1.03
and
https://github.com/Phil1988/FreeDi/wiki/Updating-FreeDi

closed 2024-11-06 by Phil1988