Phil1988/FreeDi #12: [Feature request]X Smart 3 Display Support

CanadianDreamer, 2024-10-27:
Enabling the display to support vanilla klipper on this model would be greatly appreciated.

Phil1988, 2024-10-27:
Thanks @CanadianDreamer 

Im totally open for this but I'm currently not working on it.

This feature would consume a lot of time for me because I must create a whole tft firmware from scratch dedicated to the X-Smart 3 display.
Reason is that this display has only 8MB flash instead of 16MB (the current firmware simply doesnt fit).

I will leave this feature request open to collect votings.
If the "critical mass" is reached I might create a tft firmware X-Smart 3.

So got and vote for it you X-Smart 3 users out there ;)

AaronMickDee, 2024-11-06:
Vote for yes!

CanadianDreamer, 2024-11-11:
Voting YES!!

robonxt, 2024-11-22:
Definitely voting yes to this! Thank you for the work done on the XMax and XPlus, would love to see my (fairly new) XSmart get the same treatment and love!

theclue, 2024-11-25:
Vote YES for my poor x-Smart 3 supplicating for a better screen firmware :)

Alessandro292, 2024-11-26:
Vote for yes!

42069-big, 2024-12-03:
Vote yes (sorry about the name, I made this account in middle school...)

Phil1988, 2024-12-04:
Can someone please zip his klipper configuration and attach it here? (basically all *.cfg files)
It should be as stock/generic as possible.

I want to have a look and need to consider necessary changes in my software to make it compatible.

AaronMickDee, 2024-12-04:
> I want to have a look and need to consider necessary changes in my software to make it compatible.

I've modified mine quite a bit so I'd be no help here, but your cfg files on this github were directly compatible.

Phil1988, 2024-12-04:
Thanks @AaronMickDee for your reply.
Im wondering about the compatibility:
It might work with no errors because of the same mainboard, but as far as I know, the X-Smart 3 has no heated chamber, no side cooling fans and no filter fan. 
Instead it has a chamber circulation fan.

Is that right and does it have any other components I did not mention?

eaman, 2024-12-09:
I vote yes.

Here is the OEM config backup of my X Smart 3:
[klipper_config.zip](https://github.com/user-attachments/files/18060165/klipper_config.zip)

Phil1988, 2024-12-09:
Thanks for the config! 👍 

I have a few questions.
Can you please clarify what exaktly these fans do:
Lets start with the funny one: what is the "borad" fan?  (isnt that guy spelled with a "T" at the end ? ;)
fan2  (commented "Large turbo fans on motherboards" but I dont trust this).
fan3 (commented "Activated charcoal blowing fan" but I dont trust this). You said its a chamber fan. Does it blow air out or is this the "chamber circulation fan"?

theclue, 2024-12-10:
Lemme see if I can help :)

- Yes, these guys at QIDI mispelled "board fan" with "borad fan" and never fixed that :D 

About the second and third point, I can do some hypothesis on what makers usually do on their printers (the same I did, actually):

- x-Smart doesn't have an activated charcoal fan, but they have a chamber fan blowing out from the chamber on the rear. Typical upgrades is to add a grill on that fan with an activated charcoal sheet on it. No electical modiications here.
- 
- Another typical upgrade is to wire a mini bentoo box with activated charcoal pellets inside the chamber with a fan. This fan can be connected to an unused pin in the motherboard (the board is the same of the bigger brothers of the family, which actually have this filter fan on stock).

I suspect that @eaman could have done some of those typical upgrades. 
Since they are totally common by x-Smart owners I suggest you to consider it in your custom UI for the display. This will be an incredibly added value vs the stock one (actually I pilot the fan I added within a gcode macro, but it's definitively not user friendly)

hope this helps!

eaman, 2024-12-10:
>     * Lets start with the funny one: what is the "borad" fan?  (isnt that guy spelled with a "T" at the end ? ;)
Are you referring to:

##主板上大涡轮风扇
[output_pin fan2]

To my knowledge there's no such thing, as the the mini turbo noisy fan on the mainboard is always on connected straight to a +- power plug on the board, not MCU controlled.
[image]

Port / plug n.11 on the attachet board picture

> 
>     * fan2  (commented "Large turbo fans on motherboards" but I dont trust this).
See previous one

> 
>     * fan3 (commented "Activated charcoal blowing fan" but I dont trust this). You said its a chamber fan. Does it blow air out or is this the "chamber circulation fan"?

##活性炭吹风扇
[output_pin fan3]

It blows air out, it's a ~6cm normal fan on the back top-left.

Cheers

Phil1988, 2024-12-10:
Thanks for your replies.

@eaman :
Socket #11 on your picture is for some mainboards controllable (PC4), see:
https://github.com/Phil1988/FreeDi/wiki/Why-creating-a-backup-is-important#example-differences

So to summerize:
[output_pin fan3]
pin:PC9
Is for a outlet fan.

[output_pin fan2]
pin: PA8

Im still wondering what this is:
[output_pin fan2]
pin: PA8

It is connected to the green socket #10 in the posted mainboard mapping picture.
Is there even anything connected at the smart3?

eaman, 2024-12-11:
> [output_pin fan2]
> pin: PA8
> 
> It is connected to the green socket https://github.com/Phil1988/FreeDi/issues/10 in the posted mainboard mapping picture.
> Is there even anything connected at the smart3?

I checked again, that is free, it's not populated. 
Pictures:
[image]
[image]

Thank you for looking into it!

theclue, 2024-12-11:
it this may help, I'll check my printer this weekend and provied a photo of populated connections here 👍

Phil1988, 2024-12-11:
Thanks @eaman for checking that.
This helps :)

@theclue:
Under normal circumstances I would say "no need to check it", as eaman did it already, BUT: 
This is a Qidi printer where they developed/improved it continously.
So please check that too just to have one more data set :D

eaman, 2024-12-11:
Yeah probably QIDI made some revision of the X Smart 3: I got mine just last week, it has the bed probe on the right of the toolhead, the RJ45 is not exposed, it came with a brass nozzle. 

Maybe in case some other version had a fan on that socket #10 just provide a config to make it work, it would not affect printers like mine.

robonxt, 2024-12-11:
If we need more data, I can also take pictures of my printer. I don't have my configs though as the printer is currently dead due to me updating the kernel image (the board and the touchscreen wasn't happy haha)

Phil1988, 2024-12-12:
Here are some news:
I took the last 3 days to work on the firmware.
It looks promising and I will update you soon about the release of the first test version.
But: I am currently very limited in time. So this will be beta version.
This will be usable but not a polished as the current stable and probably not be polished until april 2025.

@robonxt :
With my project, you should be able to bring your printer back to life again :)

robonxt, 2024-12-14:
Thank you! Support was kind enough to send me a replacement EMMC, but alas, there's something wrong with the OS inside it as it's giving me errors that I've not seen during my setup. Probably need a fresh wipe and reinstall from QIDI's recovery image, but here's hoping FreeDi gets it working soon with all the python3 goodness!

Phil1988, 2024-12-14:
Good news!

I worked on the firmware for the past 5 days and just released the fresh designed beta v1.03 for the X-Smart 3:
https://github.com/Phil1988/FreeDi/releases/tag/v1.03b

I expect multiple things to not fully work.
Thumbnails will definetly be a problem as the software has to be modified.
Currently the thumbnails do have a different size for the bigger displays of the Plus/Max.

As I dont own this machine, I cant test it.
Therefore the real life appearance of the UI is somehow a surprise for all of us :)
Please give feedback!

AaronMickDee, 2024-12-14:
I’ll test it out once I get the machine setup. Excited!

robonxt, 2024-12-14:
I'll give it a go later when I have time. This weekend suddenly got more exciting 😂

AaronMickDee, 2024-12-14:
Flashed good. Seems stuck at the booting screen. Doesn’t move past this screen.

[image]
[image]
[image]
[image]

AaronMickDee, 2024-12-14:
OK, so far I've followed the install procedure for: https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-existing-systems using option A and B for downloading the program.

After running 'python3 start.py' the screen changes to "Connecting to Klipper...". The loading bar finishes to the end and just stops there.

Inside my terminal, I get "Exception in update_variable(): 'temperature' repeating itself over and over again.
My install was done a fresh Armbian build.

root@mkspi:~/X3seriesLCD# python3 start.py
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'

billkenney, 2024-12-15:
@AaronMickDee I had the same error and it was because I hadn’t renamed heater hot to heater chamber in my printer.cfg file. Make sure you make all of the changes in the instructions. https://github.com/Phil1988/FreeDi/issues/16#issuecomment-2453501184

AaronMickDee, 2024-12-15:
@billkenney The X-Smart 3 doesn't have a heated chamber, so there is no config setting to change or add.

billkenney, 2024-12-15:
> @billkenney The X-Smart 3 doesn't have a heated chamber, so there is no config setting to change or add.
> 
We had the same error and my problem was that I didn't rename one section. There are multiple other sections specified in the instructions that the smart3 does have. So you should check to make sure they are all named correctly. Or maybe the problem is that the firmware is looking for heater_generic chamber and it doesn't exist because the smart3 doesn't have a chamber heater. It is a beta version after all. 

From the instructions:

Make sure you have these sections (or rename existing) and they are correctly configured in your printer.cfg (or the included files):

[extruder]  -   (obvious... shouldnt need to change)
[heater_bed]  -   (obvious... shouldnt need to change)
[heater_generic chamber]
[output_pin partfan]
[output_pin sidefan]
[output_pin filterfan]
[output_pin caselight]
[filament_switch_sensor filament]

AaronMickDee, 2024-12-15:
@billkenney I have all of those except the heater_generic chamber. How would i go about adding that without Klipper throwing an error?

billkenney, 2024-12-15:
@AaronMickDee maybe try adding the below sections from my printer.cfg. If the firmware no longer throws a temp error, you know that's the problem and it will have to be addressed by the dev. 

[heater_generic chamber]
heater_pin:PB10
max_power:1.0
sensor_type:NTC 100K MGB18-104F39050L32
sensor_pin:PA1
control: watermark
max_delta: 1.0
min_temp:-100
max_temp:70

[verify_heater chamber]
max_error: 300
check_gain_time:480
hysteresis: 5
heating_gain: 1

AaronMickDee, 2024-12-15:
@billkenney It throws an error saying PB10 is used multiple times. It's the same pin that's used for my heater_bed

billkenney, 2024-12-15:
@AaronMickDee try changing it to PC8, that's what the bed is in my printer.cfg

AaronMickDee, 2024-12-15:
@billkenney Same issue. Weird that it's happening.

billkenney, 2024-12-15:
@AaronMickDee same issue as in the screen firmware is throwing the temp error or klipper is failing to start? If the screen firmware is no longer throwing the temp error (with either PC8 or PB10) then the problem seems like it might be that it's looking for the chamber heater section, regardless of any klipper errors.

AaronMickDee, 2024-12-15:
The screen just goes through either "Booting...." on fresh start, or "Connecting to Klipper..." if I firmware_restart or save_config

billkenney, 2024-12-15:
@AaronMickDee does it still throw the error you mentioned in your original post? 
Exception in update_variable(): 'temperature'

AaronMickDee, 2024-12-15:
Yes, inside the terminal. However, I got a little further. 
The screen now shows "Klipper is ready You should be forwarded to the main page...." with the options of Klipper Restart, Firmware Restart, System restart, System shutdown buttons.

Inside the terminal I get this:

Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'value'
Exception in update_variable(): 'value'
Exception in update_variable(): 'value'
ERROR in extract_printjob_values(): gcode file not known
Exception in thread Thread-6 (periodic_update):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/main.pyx", line 161, in main.X3seriesLCD.periodic_update
  File "build/lcd.pyx", line 905, in lcd.LCD.data_update
  File "build/main.pyx", line 845, in main.X3seriesLCD.lcd_callback
  File "build/main.pyx", line 358, in main.X3seriesLCD.extract_printjob_values
UnboundLocalError: local variable 'file_name' referenced before assignment

billkenney, 2024-12-15:
@AaronMickDee at least you made progress and the screen is able to connect to klipper. Hopefully your troubleshooting helps the dev pinpoint and fix the issue.

robonxt, 2024-12-15:
On the topic of flashing the firmware to the screen, will a future update allow users to not have to open up the printer to get to the screen's SD slot?

robonxt, 2024-12-15:
Interestingly, I'm getting similar errors, although it may be due to me not remembering how to update the toolhead:

```command
mks@mkspi:~$ sudo systemctl start X3seriesLCD.service
mks@mkspi:~$ cd ~/X3seriesLCD && python3 start.py -debug
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Exception in thread Thread-1 (run):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
Error state detected. Skipping further steps in update_variable()
    self._target(*self._args, **self._kwargs)
  File "build/lcd.pyx", line 1282, in lcd.LCD.run
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Error state detected. Skipping further steps in update_variable()
Connected to web services!

Error state detected. Skipping further steps in startWebServices()
Printer volume: None
Klipper version: None
Software version: v1.03 stable
X3seriesLCD start
sw-version sent!
self.lcd.data_update()
Error state detected. Skipping further steps in update_variable()
data.klipper_state != self.printer.klipper_state
klippy error
Error state detected. Skipping further steps in klippy_callback()
Error state detected. Skipping further steps in klippy_callback()
Error state detected. Skipping further steps in klippy_callback()
Error state detected. Skipping further steps in klippy_callback()
self.lcd.data_update()
Error state detected. Skipping further steps in update_variable()
self.lcd.data_update()
[...]
```

and it repeats. As mentioned before, it could be due to the toolhead not being updated, but I don't remember how to do that, is it just through kiauh?

AaronMickDee, 2024-12-15:
@Phil1988 Would it be beneficial at all to send you my mainboard and screen for development and testing?

Phil1988, 2024-12-16:
@AaronMickDee :
Please share your printer.cfg and a screenshot of your X3seriesLCD folder.

@robonxt :
For the tool head and MCU you could simply follow this:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way

For you it seems that you came from a stock/qidi OS and your xindi is still running.
Did you use the guide from [billkenney](https://github.com/billkenney) or similar?
Your serial port is still in use/blocked.
I recommend to do this from scratch:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems by following the complete guide (espacially https://github.com/Phil1988/FreeDi/wiki/Why-creating-a-backup-is-important)

robonxt, 2024-12-16:
@Phil1988 I used a fresh empty EMMC with your os build from release 

> For the tool head and MCU you could simply follow this:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way

Silly me, but I somehow did not see that page at all. Will give it a go!

As for the backup, I was able to recover my 8gb EMMC klipper config files, and I promptly wiped it afterwards. 

If it helps, I can upload them in a bit. Just let me know what files are more helpful (I did a copy of the whole mks user home directory)

Otherwise I guess I'll start from scratch again 😅

AaronMickDee, 2024-12-16:
I reverted back to my original config setup. but this is a 100% working config for qidi x-smart 3.

[qidi x-smart 3 printer.zip](https://github.com/user-attachments/files/18147132/qidi.x-smart.3.printer.zip)

Phil1988, 2024-12-16:
@AaronMickDee :
You have multiple errors in your printer config like wrong fan names, filament sensor named "fila" etc.
Please compare and adjust accordingly to:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-existing-systems#2-prepare-your-printer-configuration

If thats too much for you, I recommend flashing my pre-build image and change stepper max positions + homing macro according to your smart dimensions.

@robonxt :
Your config is likely also wrong.
I get that by the terminal output "klippy error" that only happens if there is an issue with the klipper service "klippy".
Your printer setup/config needs to be working before my software can communicate with the moonraker API and the screen :)

AaronMickDee, 2024-12-16:
@Phil1988 thats correct. I reverted back to this config after changing everything according to your link. 
I’ll give it another go here in a bit. 
can you confirm which part of that setup is required? The smart 3 doesn’t have a heated chamber. Is “side fan” the chamber exhaust fan and it’s ok to remove the filter fan?

AaronMickDee, 2024-12-16:
@Phil1988 I used the max3 config from this github, changed the toolhead address, and it worked. So A+. I'll dig around for some bugs.

AaronMickDee, 2024-12-16:
Here is my config setup that works - Just in case anyone else is having a hard time getting it to work: 
[config-20241216-154900.zip](https://github.com/user-attachments/files/18156915/config-20241216-154900.zip)

@Phil1988 I renamed "sidefan" to "boardfan" inside my config. Inside the UI it keeps the name "sidefan" and when trying to adjust the fan speed it throws an error inside klipper console: The value 'sidefan' is not valid for PIN.

When adjusting the fan speed via mainsail it doesn't update on the UI.

I'll keep poking around for bugs.

robonxt, 2024-12-16:
@Phil1988 at the moment, I'm stuck at the following error:

```command
mks@roboqxs3p:~$ sudo systemctl stop X3seriesLCD.service
mks@roboqxs3p:~$ sudo systemctl start X3seriesLCD.service
mks@roboqxs3p:~$ cd ~/X3seriesLCD && python3 start.py -debug
  Moonraker host address:
  http://127.0.0.1:80
  Connecting to UDS /home/mks/printer_data/comms/klippy.sock
  Connected!
  _BootPage: len: 1 data[0]: ee
  ActiveScreen: = BootPage
  Exception in update_variable(): 'temperature'
  Exception in update_variable(): 'temperature'
  G-Code file path is: ~/printer_data/gcodes
  Exception in update_variable(): 'temperature'
  Exception in update_variable(): 'temperature'
  Exception in update_variable(): 'temperature'
```

I'm really scratching my head with this. I've followed the steps according to FreeDI guide (using your img file), and I've taken a look at @AaronMickDee 's configs to see if there were any temperature-related changes.

robonxt, 2024-12-17:
@Phil1988 I would like to second https://github.com/Phil1988/FreeDi/issues/12#issuecomment-2543427828 as it seems like the display is asking for chamber temperature regardless if it's available or not. I cleaned up my config file (basing it off X-Max3's config), and modifying a few values. One reason why I think this error keeps popping up is because in the original X-Max3's config:

```cfg
[heater_bed]
heater_pin:PC8

[...]

[heater_generic chamber]
heater_pin:PB10
```

but for X-Smart3:

```cfg
[heater_bed]
heater_pin:PB10 # < it's changed to PB10

[...]

#[heater_generic chamber] # < there's no chamber heater
# commented out the rest of the chamber related stuff, plus the chamber related stuff in macros.cfg
[...]
```
I believe the tft firmware is the culprit here, as the printer is working fine, just the display is asking for the temperature for the chamber when it does not exist.

Is there a workaround to fool the display that the temperature value is getting passed?

AaronMickDee, 2024-12-17:
> Is there a workaround to fool the display that the temperature value is getting passed?

I'm not sure that's the problem. My display is working now. I've made quite a few changes to my config setup since my last post. Mainly just removing errors that the display throws (set_fan_speed an on output pin error, end_print vs print_end, etc).

Would you like my latest config backup? Try throwing it on your machine and see if it works.

AaronMickDee, 2024-12-17:
My latest config, with most bugs squashed. [config-20241216-203125.zip](https://github.com/user-attachments/files/18158897/config-20241216-203125.zip)

robonxt, 2024-12-17:
Thanks @AaronMickDee for your config. I tried yours out, but it seems like the display is just not happy. I'll keep trying it, and seeing if I could brute force it or something. 😅

Phil1988, 2024-12-17:
Hey guys,

so here is some more information for you:
FreeDi has a clean armbian base with pure klipper. (see more here https://github.com/Phil1988/FreeDi/wiki/Interaction-between-software-and-the-display)
To make the screen work, I need to pull/push information between klipper (via moonraker APII) and the screen.
This is what my software does.
On top of that the display needs to be able to use/show/send information, thats what my tft firmware does.

For the X-Smart 3 beta I "only" created the tft firmware (which took me 5 full days).
I did NOT change the software that connects the screen with klipper.
So this is the same as for the X-Plus/Max 3.
This means you currently need to have all the elements in the config (like chamber heater etc.) even if your machine doesnt have it.

Reason is because I created this software for the Plus/Max and I currently dont have the time to make it more flexible, but I have it on the road map.

**So for now:**
Use the elements that are mandatory in your config even if you dont have it.
The "**side fan**" is a blower fan, that cools the part (I guess you dont have that). On the Plus/Max its connected to pin **PA8**
The **"filter fan"** is used to blow out the cleaned air. This is what used to be **"fan3" on pin:PC9** according to https://github.com/Phil1988/FreeDi/issues/12#issuecomment-2531289061

I hope this helps.

@robonxt 
What exactly doesnt work for you?

@AaronMickDee :
Thanks for sharing!
It would be nice if you could confirm if everything works for you now.
If it does, can I place your config to the wiki to share for others?

AaronMickDee, 2024-12-17:
@Phil1988 I’ve done a few prints so far using the screen. Everything works so far. I’ve had to do some hacky things inside the config to get it work (duplicate pins).
Go ahead and share it.

Phil1988, 2024-12-17:
I plan to release a new version tomorrow that solves all of your problems (hopefully :D )
and has nice new features.

So stay tuned ;)

AaronMickDee, 2024-12-21:
@Phil1988 I see your making lots of changes. Any of those changes apply to us? How would we go about to updating without wiping our configs?

Phil1988, 2024-12-21:
Hey everyone!

I have very good news for you: 
**Official support for the X-Smart 3 for FreeDi v1.20 and later** 🚀

Check this out:
https://github.com/Phil1988/FreeDi/releases/tag/v1.20

Regarding your question @AaronMickDee:
You are right, there have been a lot of defelopment recently.

Please check out the wiki:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide

I spend a lot of time to cover all thinkable cases and writing guides for it.
If you still have questions, let me know.

This is my christmas present for all of you 🎅🎄

[1 more reply]

closed 2024-12-21 by Phil1988