Phil1988/FreeDi #14: LOAD Button inactive when machine is cold

leckert123, 2024-10-30:
LOAD and UNLOAD buttons are grayed out and do not operate when the hot end is cold. Pressing the buttons, when grayed out, does not trigger the heat up sequence.  The buttons can only be made active (blue) when first preheated manually. This is the same behavior as the arrow buttons.  Arrow buttons make sense to be gray and not active, unless it's hot, for safety.  The LOAD and UNLOAD need to be active all the time, so the heat up and extrude macro can be initiated.

aalm, 2024-10-30:
I think the display load button should also ask the target temperature for the macro it uses, if possible?

leckert123, 2024-10-30:
I agree with the request by aalm, but perhaps that needs to be a feature request? 

Funny thing was that there was, indeed, a bug, but it was not that the extruder didn't extrude when the preheated LOAD button was pressed. It was that the LOAD button should not have been grayed out.  Bug yes, wrong bug reported! :)

Phil1988, 2024-10-31:
Will be fixed in v1.03.

Regarding the feature request:
I dont think I will implement it, as it adds another menu in the firmware and takes more flash space.
You can always change that default value in the macro.cfg.

If you still think its usefull, please open a [Feature Request] and lets see if others vote for it.
Unless there is not much interest I dont think I will spend ~5h on implementing it for ~2-4 users :)

leckert123, 2024-10-31:
I'd love the feature, but if it's a ton of work - may be not worth doing. 

I think I'll follow your suggestion and modify the macro to 250C.  I use that value to flush the nozzle and load more than any other.  And if I'm flushing really high temp stuff, I'll just do it manually. It's truly not a big deal for me.

Phil1988, 2024-11-06:
Its fixed, tested and released :)

Please check out the release (+changelog) :
https://github.com/Phil1988/FreeDi/releases/tag/v1.03
and
https://github.com/Phil1988/FreeDi/wiki/Updating-FreeDi

closed 2024-11-06 by Phil1988