Phil1988/FreeDi #16: [Bug] SerialException Error

billkenney, 2024-11-02:
**Describe the bug**
Serial exception error. The screen will no longer get past the connecting to klipper screen. It worked a handful of times, but stopped working for some reason. I tried reinstalling the X3 software a few times but it didn't fix the problem. I switched back to the Qidi firmware and re-enabled the makerbase-client service it works fine. 

**To Reproduce**
Steps to reproduce the behavior:
Install the X3 firmware and software (with the install script). I used the printer.cfg and macros.cfg from your image, with some minor modifications. Reboot the printer and it won't get past the connecting to klipper screen. The tmp log shows a serial connection error.

**Console output**
Here's the output from the tmp log:
"/usr/lib/python3/dist-packages/serial/serialposix.py", line 595, in read raise SerialException( serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.02 stable
 - OS:
Armbian Bookworm
 - Source:
Self configured
 - Mainboard:
Stock max3 mainboard

Phil1988, 2024-11-03:
Thanks for for your report @billkenney 

I cant understand completely what you did and what happened.
It seems that there is a software in background blocking the serial port.
Your source is "self configured", can you give more details about that?

Additionally a terminal output would help me to maybe see more, see here:
https://github.com/Phil1988/FreeDi/wiki/Bugs!

billkenney, 2024-11-03:
Yeah I figured it may be a bit tricky to troubleshoot. I'll give it another go and see if I can pull together more info. 

I'm running the image here: https://github.com/billkenney/update_max3_plus3

I disabled the makerbase-client service, which allows xindi to communicate with the screen, and it worked initially. After reflashing the Qidi 4.3.15 tft firmware I re-enabled makerbase-client and everything works as it should.

billkenney, 2024-11-03:
flashed the firmware again, disabled makerbase-client service, installed the X3 software via the install script. i am getting an error during the install script, I assume this is because the wifi module is in the proper mode. I am no longer getting a serial exception error, now it's a temperature error. 
```
Finding vendor_id and product_id for RTL8188GU
vendor_id: 0bda, product_id: b711
Look for default devices ...
 Found devices in default mode (1)
Access device 003 on bus 002
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 255
Error: can't use storage command in MessageContent with interface 0; interface class is 255, expected 8. Abort

wifi install done!
```

here are the contents of /tmp/X3seriesLCD.log, the results of python3 start.py -debug are the same, the temperature exception keeps spamming the logs:
```
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'result'
Exception in update_variable(): 'result'
Exception in update_variable(): 'result'
Exception in update_variable(): 'result'
Exception in klippy_callback(): 'result'
Exception in klippy_callback(): 'result'
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'
Exception in update_variable(): 'temperature'
.......
```

when running python3 start.py -debug and i restart the X3seriesLCD service, i get this output:
```
Exception in thread Thread-1 (run):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/lcd.pyx", line 1275, in lcd.LCD.run
IndexError: bytearray index out of range
```

Phil1988, 2024-11-03:
Yeap, I hit the bull's eye with my assumption.
This guide results in a recent klipper but does the same hacky things like the stock firmware to make the stock screen work (key word is xindi).

Its not compatible as its not a pure system and please dont mix different approaches/guides - they are not compatible to each other.
See also this: https://github.com/Phil1988/FreeDi/issues/3

You can 
A: Stay at billkennys or cchen616s approach and have mods on your system to use the stock tft firmware.
B: Use my guide for a real clean system with pure and mainline software (Klipper, Mainsail etc.). 

If you want B, start here: https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems)

I will close this issue as its not a bug/issue.
A different OS with different software/modified software/hacks does not work with FreeDi.
Main goal of this project is to get rid of that hacky things and being able to use pure other software also in the future ;)

closed 2024-11-03 by Phil1988

billkenney, 2024-11-03:
@Phil1988 I am running vanilla klipper, moonraker, fluidd, mainsail, and all other software. My image uses uses xindi to communicate with the screen, but after installing the qidi firmware I deleted klipper/moonraker and reinstalled vanilla versions with kiauh. All of qidi's modifications to klipper/moonraker are gone, and you can update any software at any time without issue. My goal was also to get rid of qidi's hacky setup, and to use current mainline software with no restrictions on updating the software. It seems both of us have accomplished that goal, although it seems our setups are incompatible. I learned about your screen firmware from someone who posted an issue on my repo and had to at least give it a shot!

Phil1988, 2024-11-03:
I had a few issues from guys with persistent background applications blocking serial communication.
So please excuse me that I cant spend my time on fixing things that are not really connected or are coming from different approaches.

Regarding your issue, I think your printer.cfg is incorrect.
check this:
[so please excuse me](https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-existing-systems#2-prepare-your-printer-configuration)

And post the result of:
http://YourPrinters_IP_address/printer/objects/query?extruder&heater_bed&heater_generic chamber

I guess your hotend is called "hot" instead of "extruder" or a similar issue.

billkenney, 2024-11-03:
Yeah I understand completely. I think I may have actually forgot to change heater hot to heater chamber as it's set up in your printer.cfg. I'll give it another go at some point today and report back.

Phil1988, 2024-11-03:
Thanks for your understanding :)
I cross my fingers, that my guess is right and looking forward for your reply.

billkenney, 2024-11-03:
Yep that was the problem. Everything appears to work fine now. I'm not sure why I was getting a serial exception previously, because I know I had the heater set to chamber when that was happing. 

If you don't mind, I may add a step to replace the screen firmware with your firmware in my guide. Great work btw!

Phil1988, 2024-11-04:
@billkenney 
That is totally fine for me.
I worked hard to make this an overall improvement for every user.

Currently there are a few minor bugs in it and some of them are already fixed.
I'll soon publish V1.03 and it would be great if you can publish the additional step in your guide after this release.
I let you know here, when its up.

Phil1988, 2024-11-06:
Hey @billkenney 
The new release is just published.
Check it out at:
https://github.com/Phil1988/FreeDi/releases/tag/v1.03

billkenney, 2024-11-06:
Thanks for the heads up. I did some testing yesterday and your firmware and printer.cfg setup seems to be working well. Nice to have the z offset saved in the printer.cfg as it should be. I always had issues with the printer forgetting the z offset with Qidi's setup, and ended up switching to fluidd. I'll add your firmware to my guide!

Phil1988, 2024-11-07:
Thanks for the feedback, mate! :)
I'm happy that you like it and appreciate the cleanliness of the setup.
It's great to hear that the z offset is working well for you now and better as the stock system.
If you have any further suggestions or feedback, feel free to share.

Regarding my firmware: I appreciate your understanding in not uploading and sharing it on your side. I trust you wouldn’t do that, but I just want to ensure everything stays organized. Thanks for your support! :D

billkenney, 2024-11-08:
My plan was to link to your repository, explain that your screen firmware solves the z offset issue, your printer.cfg setup is much cleaner, and it actually eliminates the need to install the Qidi firmware entirely. And outline the steps to use your firmware for people using my image.

Phil1988, 2024-11-08:
That sound really great and make the interaction for your users perfect.
Thanks billkenney and if you have suggestions or find bugs, you know where to find me :)

billkenney, 2024-11-08:
I guess my only suggestion at this point would be to include instructions for those of us with the bltouch. I believe the below lines are the only changes that would need to be made to your printer.cfg. You could use a one-line command to do this with sed (I haven't tested this command yet): `sed -Ei 's/^(\[probe\])$/\[bltouch\]\n sensor_pin:^MKS_THR:gpio21\n control_pin:MKS_THR:gpio11\n stow_on_each_sample: False\n#\1/;s/^(pin:.*gpio21)$/#\1/' ~/printer_data/config/printer.cfg`

[bltouch]
sensor_pin:^MKS_THR:gpio21
control_pin:MKS_THR:gpio11
stow_on_each_sample: False
#[probe]
#pin: ^!MKS_THR:gpio21

Phil1988, 2024-11-09:
Thanks, thats nice!

I will soon update and fill the config section.
These snippets will be found there.

If its finished and you miss anything, just let me know.
Goal is to have a quick "look up" section for different setups :)