Phil1988/FreeDi #17: [Feature request] Support for the Qidi Q1 Pro

tschreiner, 2024-11-27:
Hi,

The Qidi Q1 Pro printer has become quite popular and uses (I think) a similar LCD. Could you implement support for this printer?

zakfromevil, 2024-12-02:
Hello. We have TJC4827X243_011

Phil1988, 2024-12-04:
Can someone please zip his klipper configuration and attach it here? (basically all *.cfg files)
It should be as stock/generic as possible.

I want to have a look and need to consider necessary changes in my software to make it compatible.

zakfromevil, 2024-12-04:
[20241204-191821.zip](https://github.com/user-attachments/files/18009615/20241204-191821.zip)
Qidi Q1 Pro klipper v0.12.0-378

Phil1988, 2024-12-06:
Thanks @zakfromevil .

I checked your config and it seems mostly compatible (but ofc not the display itself).
I have a few questions/hints regarding this section in the printer.cfg:
```
[fan_generic auxiliary_cooling_fan]
pin: U_1:PA8
shutdown_speed: 0.0
cycle_time: 0.0100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0

[fan_generic auxiliary_cooling_fan]
pin: U_1:PA8
shutdown_speed: 0.0
cycle_time: 0.0100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0

#[fan_generic chamber_circulation_fan]
#pin:U_1:PC9
#shutdown_speed: 0.0
#cycle_time: 0.100
#hardware_pwm: false
#kick_start_time: 0.100
#off_below: 0.0

[controller_fan board_fan_2]
pin: U_1:PC9
max_power: 1.0
shutdown_speed: 1.0
cycle_time: 0.01
fan_speed: 0.5
stepper: stepper_x, stepper_y, stepper_z, stepper_z1
heater: extruder, heater_bed, chamber

[heater_fan chamber_fan]
pin:U_1:PA4
max_power: 1.0
shutdown_speed: 0
kick_start_time: 0.5
heater: chamber
heater_temp: 35
fan_speed: 1.0
off_below: 0

[heater_fan hotend_fan]
pin:gpio25
max_power: 1.0
shutdown_speed:1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 65.0
fan_speed: 1.0
off_below: 0

[heater_fan hotend_fan2]
pin:gpio11
max_power: 1.0
shutdown_speed:1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 65.0
fan_speed: 1.0
off_below: 0

[fan_generic cooling_fan]
pin:gpio2
max_power: 1.0
shutdown_speed: 0
cycle_time: 0.0100
hardware_pwm: false
kick_start_time: 0.100
off_below: 0.0
```

1. `auxiliary_cooling_fan` exists twice
2. Why is the `chamber_circulation_fan` out commented?
3. Why is the `board_fan_z` connected to the pin of the `chamber_circulation_fan`?
4. What is the `chamber_fan` (is this the `chamber_circulation_fan` ?)
5. What is the `cooling_fan`? Is this a side blowing/auxillary colling fan on the case side?

zakfromevil, 2024-12-07:
1. My error )
2-3. I have put a plug instead of a chamber circulation fan  and put it on the rear side
4. Chamber fan is a chamber heater fan
5. It's a model cooling fan

Phil1988, 2024-12-07:
Sorry, but its still not clear to me :D 

What do you mean by the plug instead of the chamber circulation fan?

zakfromevil, 2024-12-07:
I can say that I removed chamber_circulation_fan, it is not needed control it on the printer screen

zakfromevil, 2024-12-08:
I can be a beta tester, if you made lcd  firmware for us :)

Phil1988, 2024-12-12:
Ok, so here is "good and bad news" for you:
The screen rotation needs to redesign the the whole layout.
I estimate the time for implementing it at roughly 2 full working weeks by myself.
Problem is: I do have a life and bills to pay :D

So currently I cannot affort that much time for the Q1 Pro and this will stay like this untill ~april 2025.

But here is the good part:
I plan to work on it at begin on April 2025.

This might not be the answer you hoped but I want to be honest and transparent.

zakfromevil, 2024-12-12:
Good news!

Phil1988, 2025-01-06:
As the Q1 Pro and Plus4 share some similarities, please note the current anouncement for the Plus4 development:
https://github.com/Phil1988/FreeDi/issues/46

Basically the Q1 Pro development faces the same challenges  plus a time intense creation of the screen firmware.

kinghat, 2025-01-10:
@Phil1988 if youre in the US i could probably send you an extra Q1 screen i have.

Phil1988, 2025-01-10:
Thanks @kinghat for this kind offer ❤️ 
The display model is clear to me, so I dont need it.

A Q1 Pro on hand would help :D
If you now start considering  ;)  - I am based in Germany.

Challenges being the klipper modifications made by qidi (convert it to a clean klipper) and that I need to design a UI in portrait mode.

zakfromevil, 2025-01-15:
Clean klipper already exist, open q1 project. Only LCD firmware needed, maybe based on KlipperLCD project.

tschreiner, 2025-01-15:
From which area are you from in Germany? I have a Q1 Pro that I don't use
that often. I can probably lend it for some time.

Phil1988, 2025-01-15:
> Clean klipper already exist, open q1 project. Only LCD firmware needed, maybe based on KlipperLCD project.

Well, it looks like you’ve found your solution 😊
Once you’ve finished the LCD firmware, feel free to reach out, and I’d be happy to forward any Q1 Pro users to your project.
That said, I think the amount of work needed to create such a firmware is significantly underestimated.
Just to give you some perspective, it took me around 1000h to develop my first beta.
It’s a lot more complex than it might seem at first glance.

@tschreiner Please get in touch with me via discord - maybe we can do something in this case :)

Phil1988, 2025-02-21:
@tschreiner 
Is your offer to make a Q1 Pro temporarily available to me still valid?
I would like to start development in about 2 weeks and a Q1 Pro could be helpful 😊

Phil1988, 2025-03-01:
I need testers when developing as I dont own the machine.

Testing needs to be done for:
- Usage of the display UI in general
- Checking macros
- Flashing the mcu/toolhead with klipper

I estimate the testing period will start in about 10 days

Dan3dprintsuk, 2025-03-09:
any news on freedi for qidi plus 4 qidi plus 4

Phil1988, 2025-03-09:
This is the ticket for the Q1 Pro (visible by the title 😉)

As mentioned [here](https://github.com/Phil1988/FreeDi/issues/183) everything official will be posted [here](https://github.com/Phil1988/FreeDi/issues/46).

Please stop this kind of cross posting (additionally at the wrong places) as it adds unneccesary time for me 🙏

DragonzaS, 2025-03-12:
Perfect Timing! I am about to reinstall the system with OpenQ1 for use with a MMU, but I can also be a tester.

DragonzaS, 2025-03-12:
No problem with the testing for flashing of the system. I don't have an eMMC adapter yet (should have today evening).

Phil1988, 2025-03-13:
Just for transparency to others:
We checked it together and the result is, it successfully failed :D

Reason for this test was to check if there is a way to flash the toolhead rp2040 without the need to solder or physical interaction on the toolhead.
It's not possible as the Q1 Pro has a UART serial connection and no USB serial connection.
Very unfortunate!

If there is a genius out there reading this and sees an option to update the toolhead without the need to attach a usb wire to the toolhead, please let us know :)

Phil1988, 2025-03-14:
> Hello. Why don't you want to consider this project? https://github.com/frap129/OpenQ1/blob/main/docs/Installation.md

On this project I can also read this:

[image]

What is wrong with what I said?
> without the need to **solder** or **physical interaction on the toolhead**
> If there is a genius out there reading this and sees an option to update the toolhead without the need to attach a usb wire to the toolhead, please let us know :)

zakfromevil, 2025-03-15:
I personally flashed it without soldering, Dupont connectors fit well into the holes, you just need to bend them a little, this is a one-time procedure.

Phil1988, 2025-03-15:
I just don't understand the point of your question/message.

I said you need to solder or at least have physicall access to flash your toolhead.
You commented/asked why I don't do: Exactly that.

I know how to flash the toolhead and I was asking for ideas/solutions for another way 🙂 
(Which is most likely not possible. But there is always someone smarter 😉)

Why am I even asking?
Because with the X3 series you only have to press a boot button to flash the toolhead, and people are already struggling with that.
Soldering/attaching custom wires may be great for you and I, but maybe not for everyone.

KIA1333, 2025-03-15:
I have received as a gift Q1 PRO from qidi as gift for design - if needed to test anything - not soldering related - i am open :)

Phil1988, 2025-03-15:
Thanks @KIA1333 !
I'm still developing the firmware for it :)

I'll state something here when needed

zakfromevil, 2025-03-19:
I have catapult bootloader on Toolhead and can flash firmwares for testing. And LCD firmware flashing without microSD from command line, example

DragonzaS, 2025-03-19:
For the tool head I found this on the Klipper website, not sure if you saw it already though
https://www.klipper3d.org/Bootloader_Entry.html
I already have katapult installed so can't try that, but anyone else could try one of these.

If it works, then there will be problem that the default startup sequence in RP2040 expects USB connection. It is how it was designed inside the chip, without previous custom bootloader flashing it won't be able to communicate over UART.

Phil1988, 2025-03-19:
> .venv/bin/nextion-fw-upload -b 115200 -ub 921600 /dev/ttyS1 ~/klipmi/hmi/openq1/Q1_UI_mod.tft
Can you give me a link to the nextion-fw-upload?

FreeDi already does support serial flashing for the tft but I'm interessted to see how others have done it 😊

The general downside is, that you have to flash the toolhead once (best with katapult) and some kind of soldering/USB-Dupont-attachment is involved.
So this is something that cant be avoided for everyone who wants to go FreeDi.

zakfromevil, 2025-03-22:
> > .venv/bin/nextion-fw-upload -b 115200 -ub 921600 /dev/ttyS1 ~/klipmi/hmi/openq1/Q1_UI_mod.tft
> 
> Можете ли вы дать мне ссылку на nextion-fw-upload? FreeDi уже поддерживает последовательную прошивку для tft, но мне интересно посмотреть, как это сделали 😊 другие
> 

https://github.com/straga/klipmi

IAmSHOCK, 2025-03-31:
Hi. I'm a software engineer and tinker, and I have the Q1 Pro. 
I'm happy to help in any way I can.

Phil1988, 2025-04-02:
Thanks a lot @IAmSHOCK for your kind offering :)

I am noting this down and I'm very sorry that my current progress is not yet that far that I can involve you guys more.
Im pretty sure that there is a ton of things to do as soon as the alpha is released :D

Phil1988, 2025-04-10:
@IAmSHOCK @KIA1333 @DragonzaS and everything interessted:
I am starting an alpha test phase soon (depending on when the first alpha firmware is done).

If you are interessted, you can join in :)
More information here:
https://github.com/Phil1988/FreeDi/issues/216

Phil1988, 2025-06-24:
@tschreiner @zakfromevil @kinghat @Dan3dprintsuk @KIA1333 @IAmSHOCK 
Good news! FreeDi v2.00 has just been released, and it now supports the Q1 Pro.

I have to admit, this one was a real challenge - especially since I don’t have a Q1 Pro on my bench! 😅 
Thanks so much for your patience and for helping with testing and feedback. 
I hope you enjoy the new support and I look forward to hearing how it works for you!

closed 2025-06-24 by Phil1988