Phil1988/FreeDi #22: [Suggestion] Change position_min for stepper_z

colin-small, 2024-12-17:
My suggestion is to change position_min for stepper_z in the X Max 3 printer config.

It is set to 0 in the config and I was unable to perform z offset calibration due to this. I needed to change it to -6 so I could set the z offset.

I checked my stock config and it was also set to -6.

Also, I keep reading that's people with klipper use a sheet of paper for setting z offset, but the x max 3 comes with a special sized thicker piece of some sort of plastic to set it with. It is thicker than paper. I used this one to set mine with FreeDi but I'm wondering if I should be using a standard sheet of paper?

Phil1988, 2024-12-17:
You are right and I changed it to `position_min: -2`.
Maybe I didnt Included it (yet) in the last release.
"-6" is too much for my taste.
Reason is that you can move the nozzle pretty deep into the print bed and cause damage.
A "-2" will also scratch it, but prevent further damage.

For the calibration:
Use any paper for a rough adjustment.
Print a 1 layer thick big part and do a live adjustment (baby stepping).
This makes it perfect and this baby stepping feature is included in my firmware :)

I will close this as soon as I made sure its really included in the latest release (that will come soon).

colin-small, 2024-12-17:
@Phil1988 Thank you! As long as -2 allows enough for me to set the offset then I will change it from -6 to -2 as I agree with your feelings on it.

I will try this baby step feature! I was always unsure of why QiDi required this specific sized plastic sheet to handle the z offset when everything I see documented elsewhere uses a standard sheet of paper.

Thanks for clearing this up for me!

Phil1988, 2024-12-20:
Hey @colin-small :
I just checked the printer.cfg and I can see I set the position according to my recommendation.
[image]

I am not sure if I upladed the wrong image, but this will be obsolete soon :)
(I am about to release a new version)

I close this issue now.
Happy printing !

closed 2024-12-20 by Phil1988