Phil1988/FreeDi #24: [Bug] Error during firmware update process: stat: path should be string...

hcrohland, 2024-12-20:
FreeDi service starts, shows status, but does not execute commands. 
Apparently it cannot update firmware:

```
mks@qidi:~/FreeDi/FreeDiLCD$ python3 start.py -debug
Detected and set baud is 921600
Installed firmware v0 is smaller than required v1.20. Flashing new version...
Error during firmware update process: stat: path should be string, bytes, os.PathLike or integer, not NoneType
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
_BootPage: len: 1 data[0]: ee
ActiveScreen: = BootPage
Connected to web services!

Exception in thread Thread-1 (run):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/lcd.pyx", line 1437, in lcd.LCD.run
IndexError: index out of range
G-Code file path is: ~/printer_data/gcodes
Printer model: x-plus3
Printer volume: 280x280x280
Klipper version: v0.12.0-289
Software version: v1.20 stable
X3seriesLCD start
sw-version sent!
self.lcd.data_update()
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: standby
Printer in standby
Forwarded to main page
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
...
```

 - FreeDi LCD Screen v1.03b-81-g119f5b8a 
 - OS: Armbian 24.11.2 bookworm
 - Source: FreeDi image with FreeDi 1.20 upgrade
 - Mainboard: stock X-Plus 3

Phil1988, 2024-12-21:
you are faster then me releasing it ^^

How did you update?

hcrohland, 2024-12-21:
:-)

```
git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi FreeDi
cd ~/FreeDi/
./update.sh
```

Phil1988, 2024-12-21:
Im wondering, where you got this information from ?

hcrohland, 2024-12-21:
https://github.com/Phil1988/FreeDi/wiki/Updating-FreeDi

hcrohland, 2024-12-21:
No problem if you are not ready yet. Then I will reactivate the old service.

But I am happy to test.

Phil1988, 2024-12-21:
I know.
Its just heavily under construction :)

You have been at the right point and I thing you have also put the [freedi] section to your printer.cfg?

Please do this now:
open your moonraker.cfg and go to the [update_manager FreeDi] section.

change:
`channel: dev`
to
`channel: stable`

remove the whole line:
`install_script: install.sh`

change:
`desc=FreeDi LCD Screen`
to 
`desc=FreeDi LCD`

Via terminal do this:
```
sudo systemctl stop FreeDi.service
sudo systemctl disable FreeDi.service
sudo rm /etc/systemd/system/FreeDi.service

git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi FreeDi
cd ~/FreeDi/
./update.sh
```

hcrohland, 2024-12-21:
Yes, now it works! Screen was updated automatically! 

I had a hard crash of the host system during the first attempt though. I don't know where it came from and after hard restart all worked fine.

Phil1988, 2024-12-21:
Great that it works.
You are now part of the auto update user base :D

Regarding your crash:
What exactly crashed and what did/n't work?

closed 2024-12-21 by Phil1988

hcrohland, 2024-12-21:
The Linux OS died without any log. So I don't know what happened.

Thanks for your work!

Phil1988, 2024-12-28:
Perfect! 
May have been a RAM issue or similar.

Im glad that everyhing is working for you :)