Phil1988/FreeDi #25: Upgrade to 1.20 failed on X Max 3

colin-small, 2024-12-21:
I was running the previous latest version of FreeDi, I saw the new release for 1.20 and I followed the update instructions on the wiki by modifying the printer config, cloning the repo, running the update script, and rebooting the printer. The screen showed it was flashing and the. It failed with the following message.

[image]

Not sure what I should do now.

Phil1988, 2024-12-21:
It might have been a transmission error during the update.
Have you shut the printer off and on again?

I havent seen this problem on my side but I am positive its nothing big.

For further help, can you give me this information:
**Base (please complete the following information):**
 - X3seriesLCD firmware version: [e.g. v1.00 stable]
 - OS: [e.g. Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17]
 - Source: [eg. self configured or FreeDi image]
 - Mainboard: [eg. stock Mainboard X-4/X-6]

If it all fails you still can update the firmware via SD card with your matching file from here:
https://github.com/Phil1988/FreeDi/tree/master/screen_firmwares

But I would like to know and solve it the right way ;)

colin-small, 2024-12-21:
I will try a simple restart again when I get home when I get home and let you know. Does this new version only affect the screen firmware or the entire FreeDi firmware itself?

colin-small, 2024-12-21:
It worked! Simple power down and power on fixed it.

[image]

colin-small, 2024-12-21:
But now klipper isn't working. It fails repeatedly if I try to restart it.

[image]

[image]

colin-small, 2024-12-21:
I restarted the firmware and now it seems to be working again.

Phil1988, 2024-12-22:
Im glad its solved and was no bug :)

Summary:
Reboot twice

once for restarting all the services and the second was needed because your toolhead caused a small problem (this happens sometimes, also with stock qidi. its not the most robust connection).

closed 2024-12-22 by Phil1988