Phil1988/FreeDi #26: [Bug] missing fan configs causing overheating issues

PHNX-666, 2024-12-22:
**Describe the bug**
[config_section/X-Max3](https://github.com/Phil1988/FreeDi/tree/master/config_section/X-Max3)
printer.cfg is missing fan settings causing major issues when printing at high temps

printer was overheating, clogging, shutting down mid print ... not playing nice at all

**To Reproduce**
Steps to reproduce the behavior:
clean install of freedi with printer.cfg supplied in above folder

**Missing config sections**

#mini hotend fan 2

[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

#main board fan

[heater_fan borad_fan]
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

Phil1988, 2024-12-27:
hotend fan 2 is new to me.
Seems that this another ondocumented change that qidi did.

The board fan can be found at the post install section:
https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section

I will have a closer look at it tomorrow when I'm back.

Phil1988, 2024-12-28:
I had a closer look at it and cant find 
```
[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
```

Could you be so kind and show a bunch of pictures of the inside of your toolhead?
I would like to see the fans, wiring and the board of the toolhead.
I bet you are having a new daughterboard at the toolhead :)

PHNX-666, 2024-12-28:
hi yeah , i will be able to do that tomorrow for you... they sent me a newer toolhead board with front and back covers with the 15mm fans after my toolhead board blew from overheating

Phil1988, 2024-12-28:
That makes total sense and is completely in line with what my crystal ball suspected :D

Im looking forward to the pictures - it's always a bit of a surprise to see what qidi has changed now :)

PHNX-666, 2024-12-30:
Sorry took me so long, forgot to take the pics before starting a long print, the new board has new fan connections at the bottom , the covers also came with a small adapter board, most likely for the original board as i didn't need it with the new one

[image]
[image]
[image]

MaZZly, 2025-01-05:
Just chiming in that my X-max 3 bought in ~June 2024 also has the second hotend fan, and in `printer.cfg` on original fluidd it is defined as:
```
[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0
```

Haven't yet gotten to flashing FreeDi onto mine, waiting for new EMMC to work with :)

colin-small, 2025-01-05:
> **Describe the bug**
>
> [config_section/X-Max3](https://github.com/Phil1988/FreeDi/tree/master/config_section/X-Max3)
>
> printer.cfg is missing fan settings causing major issues when printing at high [...]

I have this fan also and have been experiencing some heat issues. Does simply adding these two sections from my old config to the new config resolve the issue?

PHNX-666, 2025-01-05:
Yes

colin-small, 2025-01-05:
> Yes
> 
> 
> 
> 
> 
> 
> > *Describe the bug*
> 
> >
> 
> > config_section/X-Max3
> 
> 
> >
> 
> > printer.cfg is missing fan settings causing major issues when printing at
> 
> > high temps
> 
> >
> 
> > printer was overheating, clogging, shutting down mid print ... not playing
> 
> > nice at all
> 
> >
> 
> > *To Reproduce*
> 
> >
> 
> > Steps to reproduce the behavior:
> 
> >
> 
> > clean install of freedi with printer.cfg supplied in above folder
> 
> >
> 
> > *Missing config sections*
> 
> >
> 
> > #mini hotend fan 2
> 
> >
> 
> > [heater_fan hotend_fan2]
> 
> >
> 
> > pin:MKS_THR:gpio20
> 
> >
> 
> > max_power: 1.0
> 
> >
> 
> > kick_start_time: 0.5
> 
> >
> 
> > heater: extruder
> 
> >
> 
> > heater_temp: 50.0
> 
> >
> 
> > fan_speed: 1.0
> 
> >
> 
> > off_below: 0
> 
> >
> 
> > #main board fan
> 
> >
> 
> > [heater_fan borad_fan]
> 
> >
> 
> > pin:PC4
> 
> >
> 
> > max_power: 1.0
> 
> >
> 
> > kick_start_time: 0.5
> 
> >
> 
> > heater: extruder
> 
> >
> 
> > heater_temp: 50.0
> 
> >
> 
> > fan_speed: 1.0
> 
> >
> 
> > off_below: 0
> 
> >
> 
> > I have this fan also and have been experiencing some heat issues. Does
> 
> > simply adding these two sections from my old config to the new config
> 
> > resolve the issue?

Thanks. I just added it to my config. Is there anything else related that needs to be changed? Any macros or anything? Or does it just work after adding the two config sections?

Phil1988, 2025-01-05:
@colin-small please dont answer like this.
It makes this report unreadable :D 

Just for clarification:
The first few batches of printers had this daughterboard:
[image]

So they changed this for a later batch like @PHNX-666 got.
@colin-small just add the matchin fans like:

```
#tool head board fan
[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

#main board fan
[heater_fan borad_fan]
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0
```
to your printer.cfg and you are good to go. 
No macros or more needed.

closed 2025-01-05 by Phil1988

leckert123, 2025-01-05:
FWIW, Max3 on FreeDi 1.03. Have not yet done the 1.2x update.  Machine was purchased late May 2024, delivered 1st week of June 2024. Original cfg had fan2 as well.  Not including the PSU cooling fans underneath, my machine has the following fans and **EVERY ONE OF THEM WORKS** when printing:

1. Exhaust Fan - exhausts out rear duct of the enclosure
2. Rapido Cooling fan - mounted to right interior wall of enclosure
3. MCU Fan for Main board - formerly called "borad" fan
4. Printhead has 4 fans:
     1. Parts cooling fan - 5015 mounted to front printhead cover
     2. Hot End fan - 2510 mounted to side of the hot end 
     3. Hot End aux fan - 2006 mounted to interior of front printhead cover - earlier units did not have these
     4. Toolhead board cooling fan - 2006 - mounted inside rear printhead cover

Current Fan section in 1.03 Printer.cfg.   I don't recall needing to change these when installing 1.03.

#####################################################################
# Fans
#####################################################################
## Part cooling fan 
[output_pin partfan]
pin: MKS_THR:gpio2
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0

## Big side radial turbo fan
[output_pin sidefan]
pin: PA8
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0.00
scale: 255
shutdown_value: 0.0

## Activated charcoal blowing fan
[output_pin filterfan]
pin:PC9
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0.0
scale: 255
shutdown_value: 0.0

## Cold end  and toolhead housing fans
[heater_fan hotend_fan]
pin:MKS_THR:gpio1
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

## Printhead cover small aux 200
[heater_fan hotend_fan2] 
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

## Motherboard and MCU fan
[heater_fan board_fan] 
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

Phil1988, 2025-01-06:
> Current Fan section in 1.03 Printer.cfg. I don't recall needing to change these when installing 1.03.

Might be because I assisted you step by step back then and compared the printer.cfg for you :)

leckert123, 2025-01-06:
Thanks, buddy!