Phil1988/FreeDi #27: [Bugs] missing fan2 on XSmart3 + move z exceeds limits

robonxt, 2024-12-22:
**Describe the bug**
On fresh install of FreeDi 1.20 for XSmart3, during printing right after the heating up, I get a "missing fan2" and it cancels, which also throws another error as it tries to move the Z axis to 200, which is over the max limit for XSmart3.
**Expected behavior**
Print should start normally, without tossing error, cancelling print should move bed (z) to max without exceeding limits.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install XSmart3 v1.20 on EMMC 
2. Setup tool head, mcu, etc.
3. Try printing a model
4. See errors as it heats up, throw error, cancels, throw error again.

**Console output**
I don't have a output of the error at the moment.

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.20 stable
 - OS: from v1.20, no change
 - Source: FreeDi Image
 - Mainboard: stock XSmart3 motherboard X-6 V1.0

**Extra info** 
- Workaround for fan2 is to put the `fan2` part from QIDI's XSmart3 config.
- Workaround for z exceeds limits is to modify `CANCEL_PRINT` macro to use Z150 instead of Z200. Would recommend pulling from the config file instead of hardcoding values

Phil1988, 2024-12-23:
@robonxt 
I can only guess:
Is there a 
`M106 P2` in your gcode?
If yes, I am pretty sure the slicer does it incorrectly, based on the information I got here:
https://github.com/Phil1988/FreeDi/issues/12#issuecomment-2531289061

So here is my workaround:
Change your macro.cfg.

Change this:
```
[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    {% if (printer.gcode_move.position.z) < 200 %}                                                                 ; check that zhop doesn't exceed z max
        G90                                                                                                        ; absolute positioning
        G1 Z200 F600                                                                                               ; lower bed   
    {% endif %}

    G1 X{printer.toolhead.axis_maximum.x/2} Y{printer.toolhead.axis_maximum.y-40} F12000
    CLEAR_PAUSE
    SDCARD_RESET_FILE

    PRINT_END
    BASE_CANCEL_PRINT
```

to this:
```
[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:

	{% set max_z = printer.configfile.config["stepper_z"]["position_max"]|float %}
    {% if (printer.gcode_move.position.z) < (max_z / 2) %}                                                             ; check if the bed is low enough
        G90                                                                                                        ; absolute positioning
        G1 Z{max_z} F600                                                                                           ; lower bed   
    {% endif %}

    G1 X{printer.toolhead.axis_maximum.x/2} Y{printer.toolhead.axis_maximum.y-40} F12000
    CLEAR_PAUSE
    SDCARD_RESET_FILE

    PRINT_END
    BASE_CANCEL_PRINT
```

Change this:
```
########################################
# Fan Macros
########################################
# Set Fan Speed macro
# Mainly name castings for nicer web interface names and display functionality
[gcode_macro M106]
gcode:
    {% if params.P is defined %}
      {% if params.S is defined %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE={params.S|int}
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE={params.S|int}    
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE={params.S|int}
        {% endif %}
      {% else %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE=255
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE=255    
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE=255
        {% endif %}
      {% endif %}
    {% endif %} 

    {% if params.T is defined %}
      {% if (params.T|int) == -2 %}
        {% if params.S is defined %}       
          SET_PIN PIN=filterfan VALUE={params.S|int}
        {% else %}
          SET_PIN PIN=filterfan VALUE=255
        {% endif %}
      {% endif %}
    {% endif %}

    {% if params.P is undefined %}
      {% if params.T is undefined %}
        {% if params.S is defined %}
          SET_PIN PIN=partfan VALUE={params.S|int}
        {% else %}
          SET_PIN PIN=partfan VALUE=255       
        {% endif %}
      {% endif %}
    {% endif %}
```

to this:

```
########################################
# Fan Macros
########################################
# Set Fan Speed macro
# Mainly name castings for nicer web interface names and display functionality
[gcode_macro M106]
gcode:
    {% if params.P is defined %}
      {% if params.S is defined %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE={params.S|int}
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE={params.S|int}   
        {% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything		  
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE={params.S|int}
        {% endif %}
      {% else %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE=255
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE=255    
		{% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE=255
        {% endif %}
      {% endif %}
    {% endif %} 

    {% if params.T is defined %}
      {% if (params.T|int) == -2 %}
        {% if params.S is defined %}       
          SET_PIN PIN=filterfan VALUE={params.S|int}
        {% else %}
          SET_PIN PIN=filterfan VALUE=255
        {% endif %}
      {% endif %}
    {% endif %}

    {% if params.P is undefined %}
      {% if params.T is undefined %}
        {% if params.S is defined %}
          SET_PIN PIN=partfan VALUE={params.S|int}
        {% else %}
          SET_PIN PIN=partfan VALUE=255       
        {% endif %}
      {% endif %}
    {% endif %}
```

robonxt, 2024-12-23:
I'll give it a check. I'm leaning towards it being a older gcode. Will report back

robonxt, 2024-12-23:
> Is there a
> `M106 P2` in your gcode?

Yep, in the gcodes created by QIDISlicer v1.1.7 (also in v1.1.5 and possibly older versions) and on the stock XSmart3 printer profile, there are the following lines:

```gcode
[...]
; Filament gcode
M900 K0.031 T0.03
M107
M106 P2 S0 ; < fan2 like you said
M106 P3 S255
[...]
```

I made the changes and using the same file, the printer didn't throw any more errors.
Thank you!

closed 2024-12-23 by robonxt

Phil1988, 2024-12-28:
Yeah... this is a slicer error.
They set a not present fan to 0 :(

The workaround will be included in future images

theclue, 2025-01-07:
Had the same issue, these changes in the CANCEL_PRINT macro fixed the issue at print start, but the error is still tossed when trying to abort the print (the Z value is just a bit more than 175, but I suspect it's relative to current height in some sort).

This happens both from Mainsail stop button and the LCD