Phil1988/FreeDi #30: [Bug/Feature request] Change/Add run_current and hold_current for Z stepper

robonxt, 2024-12-23:
I noticed that if I cancel a print, the motors hold their place which is normal, but the z stepper motor become super hot and it started emitting a burning smell inside the printer. 

In the config, we have:
```gcode
[tmc2209 stepper_z]
[...]
run_current: 0.95
[...]
```

While the original XSmart3 config has:

```gcode
[tmc2209 stepper_z]
[...]
run_current: 0.2
#hold_current: 0.17
[...]
```

Should `hold_current` be included, or should `run_current` be changed to the lower value to prevent the motors from being burnt out?

For mine at the moment, I've lowered `run_current` to `0.2`, so it shouldn't heat up as much anymore.

Phil1988, 2024-12-28:
Hey @robonxt 

Thanks for your report.
This is correct:
```
[tmc2209 stepper_z]
uart_pin: PC5
run_current: 0.2
interpolate: True
stealthchop_threshold: 1200
```
So leave the` hold_current `out

It will be included to future images and the config section.
If you find new bugs or have feature requests, you are welcome :)

closed 2024-12-28 by Phil1988

Phil1988, 2024-12-28:
Its included at the latest config.

Next release will have an image file with the corrected value.
Thanks for the report :)