Phil1988/FreeDi #32: [Bug] XSmart3 Display does not know correct printer model

robonxt, 2024-12-25:
**Describe the bug**
When reloading, saving settings, or restarting, the display shows the image for other printers + in advanced settings there is no option to select what printer it is (eg no XSmart3, only XPlus and XMax), causing future restarts to expect the wrong printer model.

**Expected behavior**
Display should be have option to select XSmart3 printer, and remember that it is the correct printer model.

**To Reproduce**
Steps to reproduce the behavior:
First display bug:
1. Go to 'Advanced Settings' -> 'Next Page' -> 'Settings' -> 'Next Page' until 'printer model' is shown.
2. Notice how there is no XSmart3 to select.

Second display bug:
1. Go to 'Advanced Settings' -> 'Next Page' until 'firmware restart' is shown.
2. Press 'firmware restart'
3. Notice how the UI is not displaying the correct printer model image

**Screenshots or picture**
https://github.com/user-attachments/assets/429ccccb-f9aa-4770-b92e-ec3c1a374d3b

**Base (please complete the following information):**
 - FreeDiLCD firmware version: v1.201-beta
 - OS: from v1.20, no change
 - Source: FreeDi Image
 - Mainboard: stock XSmart3 motherboard X-6 V1.0

Phil1988, 2024-12-28:
Thanks for the detailed bug report.
It was easy to find the cause and I have a few questions:
At 'Advanced Settings' -> 'Next Page' -> 'Settings' -> 'Next Page', there is X-Plus 3 and Q1-Pro.
Did you ever touch one of them?
Does the boot screen say "X-Smart 3" or "X-Plus 3" ?

Its confusing, but if you select "Q1-Pro" at the advances setting, you may see the right boot screen and logo.
Meanwhile I am fixing this and will publish a new screen firmware soon.

robonxt, 2024-12-28:
> Edit: Wait!.. you can select a "X-**Max** 3"?

My bad! I meant XPlus and XPro.

> Did you ever touch one of them?
> Does the boot screen say "X-Smart 3" or "X-Plus 3" ?

If I had fresh installed the display firmware, and have never gone into the advanced settings, yes. Otherwise it defaults to the first selected printer.

It also seems to "remember" the past selected printer, as I had to flash the display firmware multiple times before it "resets" back to the XSmart printer image.

Phil1988, 2024-12-28:
> My bad! I meant XPlus and XPro.
You meant X-Plus 3 and Q1-Pro right? ;)

> It also seems to "remember" the past selected printer, as I had to flash the display firmware multiple times before it "resets" back to the XSmart printer image.

Yes this "remembering" is normal as I save the setting to EEPROM for later starts. 
Otherwise every user would need to select the printer model after each start.
So as soon as you touch/select a printer the settings will be stored.

Im pretty sure that if you select "Q1-Pro" it should be visibly ok for you.
Would be nice if you can check that.

Are you in FreeDi dev channel (in moonraker)?

robonxt, 2024-12-28:
> Im pretty sure that if you select "Q1-Pro" it should be visibly ok for you.
> Would be nice if you can check that.

I'll give it a test soon.

> Are you in FreeDi dev channel (in moonraker)?

If you mean dev for the FreeDi update in the moonraker.conf, no.

If you mean dev as in beta for FreeDi in the printer.cfg, yes.

Phil1988, 2024-12-28:
Thanks @robonxt :
That is the right way!

Just wanted to make sure you didnt mess the moonraker.conf :)
leave it as it is 👍

Phil1988, 2024-12-28:
Beta v1.202 is pubilshed.
Besides this fix I reduced the fw size from 11460KB to 8640KB.

Please update FreeDi via 1-Click moonraker update and report if the issue is solved for you :)
[image]

robonxt, 2024-12-29:
Seems to be working fine, as the options to select other printers are no longer showing, only XSmart. 
There is another bug regarding the display but I'll submit it as a separate bug.
Thanks for the fixes!

closed 2024-12-29 by robonxt