Phil1988/FreeDi #33: Screen still shows The System Starts Abnormally after upgrade

IamBrad70, 2024-12-26:
Hi, this probably is less of a bug and more user error. I used the fresh install method and updated with your most recent image. I then followed the instructions for updating the main board and the extruder. All that is working and I can connect through mainsail/fluidd w/o any issues. However, my display is still showing the The System Starts Abnormally message. I also tried to re-run the Install.sh script in the FreeDi folder which ran successfully but I still don't have a display. 

One item to note that might be causing this is. I previously tried to install the update using Bill Kenney's method but was having issues with that one. So, I don't know if something got install on the display that needs to be uninstalled. Any help would be appreciated. 

Thank you, 
Tobias

rosenrot00, 2024-12-28:
Hi, which printer do you have?

Could you provide /tmp/FreeDiLCD.log?

Phil1988, 2024-12-28:
Hi @IamBrad70 :
This is most likely not caused by you doing something wrong.

Other then the printer model it would be nice if you can show the output of the install.sh.
I assume you have power cycled already?

Please also show the output from (via terminal):
`ls -la  /etc/systemd/system/`

and
`ls -la  ~/FreeDi/screen_firmwares/`

Thanks :)

IamBrad70, 2024-12-28:
Hi and thanks for the responses.

I have the X-Plus 3. I have attached the requested screenshots. For the
Install.sh I reran it and included all outputs. Looking at the
FreeDiLCD.log it looks like it fails on the firmware update. Is there a way
to run that manually?

Thank you for your assistance.

Phil1988, 2024-12-28:
@IamBrad70 : Please open this with your browser at github.
I cant see any attachments, so please check that :)

There are multiple ways to help you, but first we must know the problem :)

IamBrad70, 2024-12-28:
Sorry, I'm still new to github. Hopefully this will work. 

[ls screenshots.pdf](https://github.com/user-attachments/files/18266766/ls.screenshots.pdf)
[FreeDiLCD.log](https://github.com/user-attachments/files/18266767/FreeDiLCD.log)
[Install.sh Output.pdf](https://github.com/user-attachments/files/18266768/Install.sh.Output.pdf)

rosenrot00, 2024-12-28:
> Sorry, I'm still new to github. Hopefully this will work.
>
> [ls screenshots.pdf](https://github.com/user-attachments/files/18266766/ls.screenshots.pdf) [...]

Thank you very much. Github isn't always that easy.

This information helps. Could you hit the refresh button in Moonraker and then update FreeDi? This should trigger the firmware update of the screen. Please report back if it worked.

IamBrad70, 2024-12-28:
Hi, by the refresh button I assume you mean the FreeDi refresh under the power menu? I clicked that and then ran install.sh again. I still didn't get the white update screen. I also double checked all the connections on the main board and on one of the restarts it came up to the Qidi main screen, but the touch screen was unresponsive. On the next restart it was back to Starts Abnormally screen. The log file still seems the same. 
[image]
[FreeDiLCD.log](https://github.com/user-attachments/files/18266965/FreeDiLCD.log)

rosenrot00, 2024-12-28:
@IamBrad70 actually I meant on the webgui as shown here:
[image]

IamBrad70, 2024-12-28:
No worries at all. Thank you for all your help. That seems to have fixed it! 

Also, I really appreciate all your work. This upgrade was very smooth and pretty easy to follow.

Phil1988, 2024-12-29:
Yes, it was a file name error.
The update fixed that :)

Thanks for the detailed terminal output, that helped here!

Side catch:
The directory for the wifi module was wrong at the install.sh:
`sudo cp wifi/rtl8710bufw_SMIC.bin /lib/firmware/rtlwifi/`
its fixed now for future users:
`sudo cp $FREEDI_LCD_DIR/wifi/rtl8710bufw_SMIC.bin /lib/firmware/rtlwifi/`

@IamBrad70 
If you want to use wifi and its not working for you right now, please do this:
`sudo cp ~/FreeDi/FreeDiLCD/wifi/rtl8710bufw_SMIC.bin /lib/firmware/rtlwifi/`

Then reboot please :)

closed 2024-12-29 by Phil1988