Phil1988/FreeDi #34: [Bug] X-Max 3 Screen wont boot v1.20

jhook-met, 2024-12-28:
**Describe the bug**
After following instructions to update to version 1.20 the screen gets hung up on the boot screen. I have tried reinstalling via micro sd using multiple cards and redownloads. Checked all connections at the main board. 

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.20 stable
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: udpated from v1.03 stable via install instructions.
 - Mainboard: stock Mainboard X-6

rosenrot00, 2024-12-28:
Hi @jhook-met, let's figure out what goes wrong.

Could you provide the output of /tmp/FreeDiLCD.log?

As you used several SD cards, I assume that the update process of the TFT succeeded?

jhook-met, 2024-12-28:
[install.sh output.txt](https://github.com/user-attachments/files/18266902/install.sh.output.txt)
[FreeDiLCD.log](https://github.com/user-attachments/files/18266903/FreeDiLCD.log)

rosenrot00, 2024-12-28:
> [install.sh output.txt](https://github.com/user-attachments/files/18266902/install.sh.output.txt) [FreeDiLCD.log](https://github.com/user-attachments/files/18266903/FreeDiLCD.log)

Thanks a lot. I assume you added the [freedi] section in the printer.cfg, right? If so, would you mind sharing your printer.cfg? This would help to figure out where the issue originates from.

jhook-met, 2024-12-28:
[config-20241228-124417.zip](https://github.com/user-attachments/files/18267051/config-20241228-124417.zip)
The printer URL I wasn't sure if  it needed changed to my printer IP address, but I've tried that as well.

rosenrot00, 2024-12-28:
Could you remove line 164 from your printer.cfg? The "step_pulse_duration" entry is there twice and that blocks FreeDi. We are fixing that on our side in one of the next releases.

Please let me know if this fixes the issue for you.

jhook-met, 2024-12-28:
> Could you remove line 164 from your printer.cfg? The "step_pulse_duration" entry is there twice and that blocks FreeDi. We are fixing that on our side in one of the next releases.
> 
> Please let me know if this fixes the issue for you.

That took care of it! Appreciate the quick support.

closed 2024-12-28 by jhook-met

Phil1988, 2024-12-29:
I fixed and tested it to be not so critical at the config and taking the latest option value by default.
It will be included in the next released version.