Phil1988/FreeDi #35: [Bug] Display does not update on mainsail print in progress

robonxt, 2024-12-29:
**Describe the bug**
The display on the XSmart3 does not update to reflect the printing in progress page.

**Expected behavior**
Display should show in progress print no matter if print was started from the display or from the web interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Mainsail web interface.
2. Start a print.
3. Observe display not updating to reflect printing progress

**Base (please complete the following information):**
 - FreeDiLCD firmware version: v1.210 beta
 - OS: from FreeDi image v1.20
 - Source: FreeDi image
 - Mainboard: stock XSmart3 motherboard X-6 V1.0

Phil1988, 2024-12-29:
Thanks for your report!

I need to see why its not updating and have a question:
Does it work of you start the print from the touchscreen?

In all cases where it doesnt work, can you please show the terminal output if you start the programm manually with debug flag?

Do it like:
```
sudo systemctl stop FreeDi.service
cd ~/FreeDi/FreeDiLCD && python3 start.py -debug
```

Im just interessted in the output that happens right after you started the printjob.

After doing the screenshot, you can stop the program and debug output by typing "Ctrl + C" at the terminal.
Then you can restart the service by either:
`sudo systemctl start FreeDi.service`
Or reboot the system

PHNX-666, 2024-12-30:
This is also happening with my max3 , if i use mainsail or my standalone klipperscreen to start a print...stays on the main menu, temps etc dont update either.... doesnt bother me much as i dont use the built in screen much.. but would be nice if it synced every time... its fine if i start from the screen

robonxt, 2024-12-30:
> Does it work of you start the print from the touchscreen?

@Phil1988 It does, however I've noticed that I'm getting errors with possibly the printer limits, similar to the v1.03b config. I can't test it right now, though, as a print is currently running.

Here is the terminal output after I uploaded/started a print from QIDI Slicer and ran following commands:
```command
sudo systemctl stop FreeDi.service
cd ~/FreeDi/FreeDiLCD && python3 start.py -debug
```

```terminal
Detecting baud...found 921600
comok 1,1073-0,TJC4827X243_011R,165,10201,A3B03401CD3CAD50,16777216-0
Firmware is up to date on v1.210
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
_BootPage: len: 1 data[0]: ee
ActiveScreen: = BootPage
Connected to web services!

G-Code file path is: ~/printer_data/gcodes
Printer model: x-smart3
Printer volume: 177x185x175
Klipper version: v0.12.0-289
Software version: v1.21 stable
X3seriesLCD start
sw-version sent!
self.lcd.data_update()
printjob.name.txt="Toys/Low-Poly Lion v2.gcode"
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: printing
Ongoing print detected
extract_printjob_values: self.printer.file_name: Toys/Low-Poly Lion v2.gcode
Error: 404 Client Error: Metadata not available for <Low-Poly Lion v2.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=Low-Poly%20Lion%20v2.gcode
Error: 404 Client Error: Metadata not available for <Low-Poly Lion v2.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=Low-Poly%20Lion%20v2.gcode
Error: 404 Client Error: Metadata not available for <Low-Poly Lion v2.gcode> for url: http://127.0.0.1:80/server/files/metadata?filename=Low-Poly%20Lion%20v2.gcode
[...]
Max retry duration reached. Returning None.
Exception in thread Thread-4 (periodic_update):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "build/main.pyx", line 237, in main.X3seriesLCD.periodic_update
  File "build/lcd.pyx", line 1007, in lcd.LCD.data_update
  File "build/main.pyx", line 944, in main.X3seriesLCD.lcd_callback
  File "build/main.pyx", line 464, in main.X3seriesLCD.extract_printjob_values
TypeError: 'NoneType' object is not subscriptable
printer_callback triggered
printer_callback triggered
printer_callback triggered
[...]
```

Phil1988, 2024-12-30:
@robonxt 
Thanks for the terminal output 👍 

It seems that it's a parsing error with the special char "/" causing a problem.
Does this also happen if you start with a "normally gcode file name?
(I will have a look at it tomorrow or the day after)

Can you also explain what you mean by "I'm getting errors with possibly the printer limits, similar to the v1.03b config." ?
I think I didn't understood what you are telling me :)

@PHNX-666 
Could you also provide a terminal output like robonxt did?

The normal behaviour is that the screen does show the current print job, no matter how you start the print job.
I will have a look at it on my end too.

robonxt, 2024-12-31:
> It seems that it's a parsing error with the special char "/" causing a problem. Does this also happen if you start with a "normally gcode file name? (I will have a look at it tomorrow or the day after)

@Phil1988 I believe it is due to how QIDI Slicer does uploads, as I am able to type in a file name with such as `Toys/Low-Poly Lion v2.gcode` in the upload dialog for it to move it into the Toys folder with the name of `Low-Poly Lion v2.gcode`

> Can you also explain what you mean by "I'm getting errors with possibly the printer limits, similar to the v1.03b config." ?

I just tried printing a new model from the display, seems to be working fine, so probably a older version problem that I thought was still present

PHNX-666, 2024-12-31:
> @PHNX-666 Could you also provide a terminal output like robonxt did?
> 
> The normal behaviour is that the screen does show the current print job, no matter how you start the print job. I will have a look at it on my end too.

Ive just updated to 1.20 tonight , ill let you know if anything changes , if not ill do a terminal output

robonxt, 2025-01-03:
From what I can tell, the display doesn't seem to like folders, much less sub-sub-folders. 
I also wonder if the display is not happy because I deleted the `.thumbs` folders for everything in a attempt for it to reload the thumbnails?

Phil1988, 2025-01-03:
My software does not work with sub-folders yet.
I did not saw this option, as the I implemented the use of the gcodes an their thumbnails.

So for now this simply doesnt work and will be implemented
But it is not a top priority for now and will take some time.

In the meantime I tested several scenarios invoking print starts from the WebUI (Mainsail and Fluidd) as well as from console.
It all worked as long as the gcode were _not_ in sub folders.

 If anyone experience something else, please let me know.
 
 I change the title now as it is not a bug for me, but sub folder support is more of a new feature.

robonxt, 2025-01-03:
Sounds good. No rush on this feature, can work around it 👍
Will keep this open

robonxt, 2025-01-11:
Can confirm that gcodes in subfolders are displaying and printing on both display and from the web interface.

robonxt, 2025-01-11:
Ok. it does work, however any changes I made to the gcodes on the web interface such as moving, adding, or deleting doesn't update all correctly on the display. 

I've also noticed that it prints out the full filepath to the file, and if it is too long (example: folder1/folder2/folder3/super_long_file_name_that_is_cool.gcode), it shows the full path to the file and cuts off the actual file name. Perhaps scrolling it on a single line as well as removing the folder(s) from the name would work?

robonxt, 2025-01-11:
> You mentioned, that changes on the web interface (moving/deleting) doesn't update on the display.
> Can you tell me when this does work and when not?

My guess would be due to me overwriting older gcodes with newer files that have the FreeDi supported image resolutions. Previously they were all using the QiDI's recommended thumbnail settings, so perhaps the display isn't loading them correctly, which causes the display to hang and not show those files.

Either that, or it's not loading my gcodes that were uploaded prior to FreeDi, or that I was too hasty and did not wait enough for the display to generate the thumbnails for new gcodes.

Phil1988, 2025-01-11:
> My guess would be due to me overwriting older gcodes with newer files that have the FreeDi supported image resolutions. 

Just to make this clear: This is not a must but recommended for better looks (only)

> Previously they were all using the QiDI's recommended thumbnail settings, so perhaps the display isn't loading them correctly, which causes the display to hang and not show those files.

In general: FreeDi can see the thumbnails (only) that Mainsail is seeing.
If you see them in mainsail but not on you display, can you share your slicer settings with the old config and at least one gcode where this problem happened?
I want to have a look at it.

> 
> Either that, or it's not loading my gcodes that were uploaded prior to FreeDi

That cant be the case. 
FreeDi is open for all: new files, old files, new users, ideas :D
If its a valid gcode its not a problem.

> or that I was too hasty and did not wait enough for the display to generate the thumbnails for new gcodes.

Well it takes ~90ms to extract, generate and send 8 thumbnails to the screen, so you must be a really hasty person not to wait for this :D

robonxt, 2025-01-11:
> If you see them in mainsail but not on you display, can you share your slicer settings with the old config and at least one gcode where this problem happened?

Here is the original thumbnail format I had: 
`205x205/QIDI, 140x140/QIDI, 110x110/PNG`

Here's a old gcode that has the same thumbnail format:
[wolfinal.gcode.txt](https://github.com/user-attachments/files/18385522/wolfinal.gcode.txt) (I renamed the file extension to `.gcode.txt`. just remove the `.txt` to turn it back into gcode.)

Since then, I've made changes to the slicer config already to match the values according to the [wiki page](https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section#:~:text=For%20X%2DSmart%203%20users%3A)

> > Either that, or it's not loading my gcodes that were uploaded prior to FreeDi
> That cant be the case.
> FreeDi is open for all: new files, old files, new users, ideas :D
> If its a valid gcode its not a problem.

I should have clarify that it doesn't load them correctly, and if they do, sometimes with thumbnails and sometimes without. Most likely had to do with me removing the existing `.thumbnails` folder and FreeDi has to regenerate them all.

> it takes ~90ms to extract, generate and send 8 thumbnails to the screen, so you must be a really hasty person not to wait for this :D 

Pretty sure it takes much longer than 90ms for me to move/update/delete files to the printer and then run to the other side of the house to switch the different pages on the display XD

Phil1988, 2025-01-11:
# Analysis of Your G-Code: Strange Issues and Observations

I tested your G-code and found many strange things—I don’t even know where to start 😅  

---

## 1. No Hover Thumbnail

When uploading your G-code, a thumbnail is generated:  
[image]  

This is my example file list in Mainsail:  
[image]  

However, when I hover over it, no thumbnail is displayed.  
Compare this to my "Print_Finished1" example:  
[image]  

If I duplicate this G-code file in Mainsail, the thumbnail disappears entirely (this issue doesn’t occur with “correct” G-codes):  
[image]  

It seems there’s something wrong with this file.  
This might explain why the thumbnail cannot be shown or generated:  
[image]  
[image]  

---

## 2. No "QIDI" Thumbnail Format, but QOI

> The original thumbnail format I used was: `205x205/QIDI, 140x140/QIDI, 110x110/PNG`

I have never seen "QIDI" as a file format.  
I’m using QIDISlicer 1.1.6, and these are the file formats I can see:  
[image]  

If I slice with only the "QOI" thumbnail format, no thumbnail is displayed (which is expected because Mainsail doesn’t recognize this custom image file).  

If I use both "QOI" and PNG formats, everything works fine:  
[image]  
[image]  
[image]  

Orca uses the correct "QIDI standard" file format, COLPIC:  
[image]  

I didn’t encounter any issues with this format.
So "QOI" (on QIDISlicer) and "COLPIC" (on Orca) works, if at least one PNG is provided (which is a must for Mainsail and FreeDi).  

I have no idea how this "QIDI" thumbnail format is being generated.  
If it really worked for you, my only guess is that it might have been transferred via QIDI's proprietary binding, "Klipper (via QIDI)". However, this is just speculation.  

The nice QIDI developers added a "Moonraker" option (after my feature request), allowing support for the common Moonraker port alongside the default QIDI port (10088):  
[image]  

But perhaps this "QIDI" format is simply incorrect and results in a faulty G-code file.  

---

## Summary

Your old G-code files seem faulty to me, made by the "QIDI" file format.
I recommend using the standard file format PNG only.

Custom formats like "QIDI" or "QOI" add unnecessary file size and look bad in Mainsail. 😊

Im waiting for your response to see what is working and what not when you are usings "un-faulty" gcodes ;)

robonxt, 2025-01-11:
Yes, I was mainly using QidiSlicer 1.1.6 and 1.1.7 and it was set to Klipper (Qidi) she those gcodes were generated. This was before I used FreeDi. After switching over, there were one or two gcodes that were uploaded using Klipper (Qidi) but was then switched over to Klipper (Moonraker). However it was only recently that I changed the image thumbnail settings to your recommended config, which is why I still have quite a few old gcodes with the Qidi image format. 

I am in the process of reuploading gcodes that uses the new thumbnail, which is why the newer and updates gcodes are showing, but the older gcodes that has the Qidi thumbnail format aren't loading properly.

robonxt, 2025-01-11:
I believe these were the settings I used, bit it's been a while so I am not too sure: https://github.com/SoftFever/OrcaSlicer/issues/5973#issuecomment-2209318078

Phil1988, 2025-01-28:
Check this out for subfolder support 😉
[FreeDi v1.40 (software and image)](https://github.com/Phil1988/FreeDi/releases/tag/v1.40)

closed 2025-01-28 by Phil1988