Phil1988/FreeDi #36: The screen always turns on with a blue error message

astur35, 2024-12-30:
Very good afternoon, I have done your tutorial to the letter but I can not get the screen to work, I always get the blue screen.
My system is complete in the same version.

mcu(stm32f401xc)
Version: v0.12.0-289-g08a1c9f12
Load: 0.01, Awake: 0.00, Freq: 84 MHz, Temp: 27°C 
mcu MKS_THR(rp2040)
Version: v0.12.0-289-g08a1c9f1
Load: 0.01, Awake: 0.00, Freq: 12 MHz, 
Host(aarch64, 64bit)
Version: v0.12.0-289-g08a1c9f12-dirty
OS: Armbian-unofficial 24.2.0-trunk bookworm
Distro: armbian
Load: 0.11, Mem: 267.0 MB / 911.1 MB , Temp: 44°C
end1 (192.168.0.15) : Bandwidth: 3.4 kB/s , Received: 3.0 MB , Transmitted: 3.0 MB 

The installation was done several times from 0 with the image, and the result is always the same, with the screen that does not work, it stays in blue.

mks@mkspi:~/FreeDi/FreeDiLCD$ ^C
[image]
[image]
[image]

astur35, 2024-12-30:
After updating several dependencies, I got the screen to start installing and working. Thank you very much for the great work.

[image]

Phil1988, 2024-12-30:
Hey @astur35 ,

If you run into problems, don't hesitate to open an issue right away.
Not need to spend time alone and wasting time, I am happy to help you.
It's likely that others may have the same problem so fixing it on my side is the goal.

I'm very glad you were able to solve it!
Now it's time for me to understand what the cause was ;)

Your report was very good btw!

I assume you took the 
"FreeDi_v1.20_X-Plus3_Armbian24.2.0_Bookworm6.6.17.rar"
Image from here:
https://github.com/Phil1988/FreeDi/releases/tag/v1.20

Is that right?

I had a file name error in this image (for the x-Plus 3 only) and it will be fixed in the next release.

I'm am wondering why Moonraker update manager said "invalid".

Can you explain a little what you did to solve it?

I guess it had something to do with the "invalid" state at the update manager, right?

I pretty sure that what you did triggered the FreeDi update via Moonraker update manager, which loads the latest/correct file for you and installed the screen firmware.

closed 2024-12-31 by Phil1988

astur35, 2025-01-02:
Good morning, sorry for the delay in responding. The installed version is FreeDi_v1.20_X-Plus3_Armbian24.2.0_Bookworm6.6.17
The problem was solved by updating FreeDi in the Update Manager. When you go to update it, two options appear, one soft and one hard. I tried several times with the soft one but it didn't do anything. I didn't dare to perform the hard update because I didn't know what would happen. When I thought that the most that would happen would be to have to reinstall everything again, and being a process that takes more time to remove the eMMc than to install, I did a hard update and that's when it started to install the correct screen taking about 5 or 8 minutes, and being blocked in the startup image of the X-MAX 3 once the installation was finished. Turn off the printer, turn it back on and it booted up the first time.

Translated with DeepL.com (free version)

Phil1988, 2025-01-02:
Thank you for clarifying this.
This makes totally sense and is what I thougt.
Thanks!