Phil1988/FreeDi #37: [Bug] X-Smart 3 display not at ttyS1?

JKorang, 2024-12-31:
Hey all,

I'm trying to get this installed on my X-Smart 3 and I'm running into a number of issues.  Chief among them is that, best as I can tell, my display is not seen by the system.  On stock firmware, I can confirm that the display was fully functional. 

Display: X-4.3 V2.0
MCU: X-6 V1.0

I have not tried the pre-built image for FreeDi.  Instead, I am trying to get this running on top of a base install of Armbian, built from `Armbian-unofficial_25.02.0-trunk_Mkspi_bookworm_edge_6.12.4`.  Toolhead and MCU are flashed with the latest Klipper, v0.12.0-404-g80d185c94, and the printer itself is functional.  

I tried to use the `install.sh` script, but it doesn't seem usable on new systems without a few changes.  

- FreeDi.service assumes the default user is `mks`.  In my case, I have a different, unique user
- Wifi firmware does not properly copy. This appears fixed per https://github.com/Phil1988/FreeDi/issues/33#issuecomment-2564663362
- freedi.py assumes the default user is `mks`, and that the display is available on `ttyS1`
- On first run(s), the installer assumes that you already have FreeDi.service deployed, then fails when evaluating whether the service file exists.

Out of the box, FreeDi is looking for a device at /dev/ttyS1 and I don't appear to have anything present there.

```
cat /proc/tty/driver/serial
serinfo:1.0 driver revision:
0: uart:16550A mmio:0xFF110000 irq:21 tx:236184 rx:4343411 RTS|DTR
1: uart:unknown port:00000000 irq:0
2: uart:16550A mmio:0xFF130000 irq:22 tx:17837 rx:0 RTS|DTR
3: uart:unknown port:00000000 irq:0
4: uart:unknown port:00000000 irq:0
5: uart:unknown port:00000000 irq:0
6: uart:unknown port:00000000 irq:0
7: uart:unknown port:00000000 irq:0
```

I replaced every instance of `/dev/ttyS1` with `/dev/ttyS2` but that doesn't look to let me communicate with the display.  When pointing at `ttyS1`, I only get I/O errors out of FreeDi.  When pointing at `ttyS2`, FreeDi attempts to connect but can't see what version is flashed to the display, and seemingly never determines what baud it should communicate at.  Per `/tmp/FreeDi.log`,

/dev/ttyS1 defined:
```
Traceback (most recent call last):
  File "/home/jkorang/klippy-env/lib/python3.11/site-packages/serial/serialposix.py", line 398, in _reconfigure_port
    orig_attr = termios.tcgetattr(self.fd)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (5, 'Input/output error')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jkorang/FreeDi/FreeDiLCD/start.py", line 11, in <module>
    main()
  File "/home/jkorang/FreeDi/FreeDiLCD/start.py", line 7, in main
    x = X3seriesLCD()
        ^^^^^^^^^^^^^
  File "build/main.pyx", line 143, in main.X3seriesLCD.__init__
  File "build/lcd.pyx", line 494, in lcd.LCD.startLCD
  File "/home/jkorang/klippy-env/lib/python3.11/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/home/jkorang/klippy-env/lib/python3.11/site-packages/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
```
/dev/ttyS2 defined: 
```
Detecting baud...not found baud, reverting to initial 921600
Installed firmware v0 is smaller than required v1.21. Flashing new version...
Detecting baud...not found baud, reverting to initial 921600
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/jkorang/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
...
```

I ran into a bit of a snag with printer.cfg, which looks largely undocumented?  It took me a bit to realize the code was looking at _only_ printer.cfg for the `[freedi]` entry.  My config structure uses a lot of includes, so it was not initially detected.

```
[freedi]
printer_model: x-smart3
serial_port: /dev/ttyS2
api_key: :shushing_face: 
klippy_socket: /home/jkorang/printer_data/comms/klippy.sock
channel: stable
```

The most recent thing I've tried was to flash the display via mSD card.  My display is now running v1.21, but it boot loops presumably trying to talk to the system.  Any ideas on how to proceed or where to look next for troubleshooting?

Phil1988, 2025-01-01:
Thanks for your detailes bug report.
Dynamic user is now integrated in the install script and the wifi path bug was already fixes as you found :)

The real issue here is why your mainboard doesnt use /dev/ttyS1.
No one with a stock mainboard had it on a different port than this.
I am pretty sure that your ttyS1 is not enabled properly and that ttyS2 is not correct for you.

Can you please use the latest install.sh and start it again?
Then please show me the complete output.

I am positive we can fix this soon :)

JKorang, 2025-01-02:
Hello,

I appreciate all the help here, and I understand what you mean regarding fixing this.  I ran the latest install script and I am in the same position as before. 

```
Detecting baud...not found baud, reverting to initial 921600
Installed firmware v0 is smaller than required v1.21. Flashing new version...
Detecting baud...not found baud, reverting to initial 921600
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/jkorang/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
```

And, the output from the script:
```
Do you use the stock Mainboard? (y/n)
y
Starting the installation...
Creating a symbolic link for freedi.py from /home/jkorang/FreeDi/klipper_module to /home/jkorang/klipper/klippy/extras...
Successfully installed freedi.py to /home/jkorang/klipper/klippy/extras.
Restarting Klipper service...
[sudo] password for jkorang: 
Klipper service restarted successfully.
Installation complete.
Setup dtbo for serial communication...
dtbo install done!
Customise the armbianEnv.txt file for serial communication...
Line not found. Adding the line.
armbianEnv.txt customization comlpleted.
Activating Klipper virtual environment and installing Python packages...
Klipper environment python version: 3.11.2
Arranging Python requirements...
Requirement already satisfied: pip in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 1)) (24.3.1)
Requirement already satisfied: configparser in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 2)) (7.1.0)
Requirement already satisfied: requests in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 3)) (2.32.3)
Requirement already satisfied: pyserial>=3.5 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 4)) (3.5)
Requirement already satisfied: Pillow in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 5)) (11.0.0)
Requirement already satisfied: numpy in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 6)) (2.2.1)
Requirement already satisfied: matplotlib in /home/jkorang/klippy-env/lib/python3.11/site-packages (from -r /home/jkorang/FreeDi/requirements.txt (line 7)) (3.10.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from requests->-r /home/jkorang/FreeDi/requirements.txt (line 3)) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from requests->-r /home/jkorang/FreeDi/requirements.txt (line 3)) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from requests->-r /home/jkorang/FreeDi/requirements.txt (line 3)) (2.3.0)
Requirement already satisfied: certifi>=2017.4.17 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from requests->-r /home/jkorang/FreeDi/requirements.txt (line 3)) (2024.12.14)
Requirement already satisfied: contourpy>=1.0.1 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (1.3.1)
Requirement already satisfied: cycler>=0.10 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (4.55.3)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (1.4.8)
Requirement already satisfied: packaging>=20.0 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (24.2)
Requirement already satisfied: pyparsing>=2.3.1 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (3.2.0)
Requirement already satisfied: python-dateutil>=2.7 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (2.9.0.post0)
Requirement already satisfied: six>=1.5 in /home/jkorang/klippy-env/lib/python3.11/site-packages (from python-dateutil>=2.7->matplotlib->-r /home/jkorang/FreeDi/requirements.txt (line 7)) (1.17.0)
Python requirements installed from requirements.txt.
Installing required packages for input shaping (if not already installed)...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-13).
libopenblas-dev is already the newest version (0.3.21+ds-4).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
System dependencies installed successfully.
Adding FreeDi section to moonraker update manager...
Moonraker configuration file /home/jkorang/printer_data/config/moonraker.conf found
The section [update_manager FreeDi] already exists in the file.
Permit Moonraker to restart FreeDi service...
[... cut ...]
Hit:5 http://deb.debian.org/debian bookworm-updates InRelease                                                                           
Hit:6 http://deb.debian.org/debian bookworm-backports InRelease
Hit:1 http://fi.mirror.armbian.de/apt bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Setup complete!
Please restart your system for the changes to take effect.
```

Phil1988, 2025-01-02:
Thanks for the outputs!

The install.sh seems to work fine for you now👍

Have you rebooted the system already?

Did you change the serial port in your printer.cfg to 
`serial_port: /dev/ttyS1`? 
if not, please do this now and reboot.

JKorang, 2025-01-02:
Thanks, I had not yet rebooted.  I just saw the actual changes to `install.sh` and see why that might matter lol.

After rebooting, ttyS1 looks populated now!  FreeDi looks to see the display, but it sits at the loading screen and never advances. 

```
Detecting baud...found 921600
Firmware is up to date on v1.21
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/jkorang/printer_data/comms/klippy.sock
Connected!
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
Exception in update_variable(): 'enabled'
```

Phil1988, 2025-01-02:
This is a different issue :)

Your printer.cfg is likely wrong.
Please attach it here so I can have a look.

I am closing this ticked here, because the origin issue is solved, but you can answer to wipe the other problem out :)

closed 2025-01-02 by Phil1988

JKorang, 2025-01-02:
Thanks.  My config:

```
[freedi]
printer_model: x-smart3
serial_port: /dev/ttyS1
api_key: :P 
url: 127.0.0.1
baudrate: 921600
klippy_socket: /home/jkorang/printer_data/comms/klippy.sock
channel: stable
```

Phil1988, 2025-01-02:
yeap - thats it!

Glad we could solve it :)