Phil1988/FreeDi #39: dev/serial/by-id* : no such file or directory

bronsn, 2025-01-01:
Qidi Xsmart3

Hello,

today i went step by step through your installation guide.

Mcu was on Klipper 0.12.0... , 
Toolhead was on 0.9..

So i did in your Guide #2 and flashed the Mainboard MCU and THR Toolhead.
Aftrer flashing the Toolhead i shut down the system like you said, turned the printer on again and got the same Error message  
MKS_THR unable to connect.

So i putty into the sys type : ls /dev/serial/by-id/* 
and got 
ls: cannot access '/dev/serial/by-id/*': No such file or directory

Now i´m a bit confused

bronsn, 2025-01-01:
omg i´m an Idiot..
i removed the SD Card with the X_4-bin , now i botted again with the Card it works :).

Perhaps u should make a notice in the guide for Noobs like me 📦 
Ty

Phil1988, 2025-01-01:
Haha - plus your link is incorrect.

I'm not sure if I should note that:
If I really do a note on everything you should not do than the guide will even more expand :D

Better simply follow what the guide is telling you to do ;)

closed 2025-01-01 by Phil1988

Phil1988, 2025-01-02:
@bronsn 
Please check this again.
I checked it on my side and I am sure, that no SD card is present in the mainboard sd-slot - and its working fine.

Might have been a failed MCU flash on your first attempt?