Phil1988/FreeDi #40: Problem with lcd communication.

kadabr3, 2025-01-01:
Hi. I have a problem with the display communicating with Klipper. I installed everything according to the instructions (I think). I already had a clean Klipper installed, I just wanted to start the display. I uploaded the display software using an SD card. However, the display does not connect to Klipper. Below I am sending the debugging information and configuration. Please advise what might be wrong.

**install.sh output:**
`FreeDi git:(master) ./install.sh
Do you use the stock Mainboard? (y/n)
y
Starting the installation...
Creating a symbolic link for freedi.py from /home/mks/FreeDi/klipper_module to /home/mks/klipper/klippy/extras...
Successfully installed freedi.py to /home/mks/klipper/klippy/extras.
Restarting Klipper service...
Klipper service restarted successfully.
Installation complete.
Setup dtbo for serial communication...
cp: cannot stat 'dtbo/rockchip-mkspi-uart1.dtbo': No such file or directory
dtbo install done!
Customise the armbianEnv.txt file for serial communication...
Line found. Replacing the line.
armbianEnv.txt customization comlpleted.
Activating Klipper virtual environment and installing Python packages...
Klipper environment python version: 3.11.2
Arranging Python requirements...
Requirement already satisfied: pip in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 1)) (24.2)
Requirement already satisfied: configparser in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 2)) (7.1.0)
Requirement already satisfied: requests in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 3)) (2.32.3)
Requirement already satisfied: pyserial>=3.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 4)) (3.5)
Requirement already satisfied: Pillow in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 5)) (10.3.0)
Requirement already satisfied: numpy in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 6)) (1.26.2)
Requirement already satisfied: matplotlib in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 7)) (3.8.2)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 3)) (3.4.1)
Requirement already satisfied: idna<4,>=2.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 3)) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 3)) (2.3.0)
Requirement already satisfied: certifi>=2017.4.17 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 3)) (2024.12.14)
Requirement already satisfied: contourpy>=1.0.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (1.2.1)
Requirement already satisfied: cycler>=0.10 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (4.53.0)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (1.4.5)
Requirement already satisfied: packaging>=20.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (24.1)
Requirement already satisfied: pyparsing>=2.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (3.1.2)
Requirement already satisfied: python-dateutil>=2.7 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (2.9.0.post0)
Requirement already satisfied: six>=1.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from python-dateutil>=2.7->matplotlib->-r /home/mks/FreeDi/requirements.txt (line 7)) (1.16.0)

[notice] A new release of pip is available: 24.2 -> 24.3.1
[notice] To update, run: python -m pip install --upgrade pip
Python requirements installed from requirements.txt.
Installing required packages for input shaping (if not already installed)...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-13).
libopenblas-dev is already the newest version (0.3.21+ds-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
System dependencies installed successfully.
Adding FreeDi section to moonraker update manager...
Moonraker configuration file /home/mks/printer_data/config/moonraker.conf found
The section [update_manager FreeDi] already exists in the file.
Permit Moonraker to restart FreeDi service...
"FreeDi" is already present in the moonraker.asvc file. No changes made.
Changing permissions to enable nmcli commands without sudo (necessary for setting wifi via screen)...
Adding the user mks to the 'netdev' group...
'auth-polkit=false' is already present in /etc/NetworkManager/NetworkManager.conf.
User mks has been successfully configured to run nmcli commands without sudo.
Installing WiFi...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
Hit:3 http://security.debian.org bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://dlm.mariadb.com/repo/mariadb-server/11.4/repo/debian bookworm InRelease
Hit:6 https://dlm.mariadb.com/repo/maxscale/latest/apt bookworm InRelease
Hit:7 https://www.deb-multimedia.org bookworm InRelease
Reading package lists... Done
[... 55 lines cut ...]
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Error: HTTPConnectionPool(host='127.0.0.1', port=80): Max retries exceeded with url: /server/info (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffff837d35d0>: Failed to establish a new connection: [Errno 111] Connection refused'))

************************************************

 FreeDiLCD git:(master) python3 start.py -debug
Detecting baud...found 921600
comok 1,1281-0,TJC8048X250_011R,165,10201,13B134016643E64F,33554432-0
Installed firmware v0 is smaller than required v1.21. Flashing new version...
Updating firmware... 0%
ERROR: Unexpected response during data transfer: b''
Detecting baud...not found baud, reverting to initial 921600
Moonraker host address:
http://127.0.0.1:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Error: HTTPConnectionPool(host='127.0.0.1', port=80): Max retries exceeded with url: /server/info (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffff86aef690>: Failed to establish a new connection: [Errno 111] Connection refused'))`                       `

[image]

[image]

[image]

Phil1988, 2025-01-02:
Hello @kadabr3,

It looks as the output wasn't the first time you started the install.sh. Is that true?
Why did you install the display firmware via SD card?

I had a path error in this script which I fixed about the time when you used it. You might have just missed the changes.

Let's fix the display communication first and then see if WiFi is working (because there was an error while installing it).

Please do this:
`sudo cp ~/FreeDi/FreeDiLCD/dtbo/rockchip-mkspi-uart1.dtbo /boot/dtb/rockchip/overlay/`

I recommend rebooting the system after this.

Phil1988, 2025-01-02:
I have reviewed and internally discussed your report again and there are some strange things .
You said you have a clean klipper, but I dont believe your system is really clean.
You also mentioned that the screen is not working, but I can see that you used it at least once to change your printer model to X-Plus 3.

Please explain how you made your clean OS in detail.
I have the suspicion that something blocking your serial port.
Please also tell me your printer model (X-Max 3 or X-Plus 3) and attach your printer.cfg (as zip).

kadabr3, 2025-01-02:
> I have the suspicion that something blocking your serial port.

I checked the port and it looks like nothing is blocking it

[image]

Phil1988, 2025-01-02:
> ```
> PRETTY_NAME="Armbian-unofficial 24.2.0-trunk bookworm"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
> ID=debian
> HOME_URL="https://www.armbian.com"
> SUPPORT_URL="https://forum.armbian.com"
> BUG_REPORT_URL="https://www.armbian.com/bugs"
> ARMBIAN_PRETTY_NAME="Armbian-unofficial 24.2.0-trunk bookworm"
> ```

I see its on armbian bookworm, but where do you get that image from?
What did you do with it?
How did you install software on it?
There must have been a lot of config/install with all those modules/plugins installed.

> yes, the screen turned on once, but it still had no connection to the klipper, it did not display any data and it was impossible to interact with the printer in any way, it simply loaded its system. That's why I had the opportunity to change the printer type.

Without connection to klipper you have no access to any menu but the boot screen.
Thats why its weird.
For me it seems as it can connect, but something is also using the serial port and interfering.

> X-plus 3

> When I try to start debugging (`python3 start.py -debug`). The script tries to update the display, which it fails as you can see in the previous log and the screen displays this message. this message also appears after restarting the FreeDi.service.

Its surprising that it mentions "TJC8048X250_011R", which is exactly the screen model and the firmware it is made for.
You should not see this message but you do...

Additionally you have mutiple glitches all over the place on the menue.

Me and the log agree on "multiple access on port".

Please do this and show me what you get:
`dmesg | grep tty`
And:
`ls /dev/tty*`
And
`sudo cat /proc/tty/driver/serial`
And:
`ps aux | grep tty`

kadabr3, 2025-01-02:
I may have used your guide from the link 
[https://github.com/Phil1988/FreeQIDI?tab=readme-ov-file]
, but I don't remember exactly, it was some time ago.

```
**FreeDiLCD git:(master) ✗ dmesg | grep tty**
[    0.000000] Kernel command line: root=UUID=0fc51a61-947b-4223-9087-e9494733531f rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=623c2399-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[    0.001100] printk: legacy console [tty1] enabled
[    0.976172] ff110000.serial: ttyS0 at MMIO 0xff110000 (irq = 21, base_baud = 1500000) is a 16550A
[    0.979476] ff120000.serial: ttyS1 at MMIO 0xff120000 (irq = 22, base_baud = 1500000) is a 16550A
[    0.982664] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 23, base_baud = 1500000) is a 16550A
[    0.982839] printk: legacy console [ttyS2] enabled
[    7.263179] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[    7.274787] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.
[    7.285837] systemd[1]: Expecting device dev-ttyS2.device - /dev/ttyS2...
[    9.162076] systemd[1]: Found device dev-ttyS2.device - /dev/ttyS2.
[   10.173587] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
**➜  FreeDiLCD git:(master) ✗ ls /dev/tty***
/dev/tty    /dev/tty17  /dev/tty26  /dev/tty35  /dev/tty44  /dev/tty53  /dev/tty62    /dev/ttyS4
/dev/tty0   /dev/tty18  /dev/tty27  /dev/tty36  /dev/tty45  /dev/tty54  /dev/tty63    /dev/ttyS5
/dev/tty1   /dev/tty19  /dev/tty28  /dev/tty37  /dev/tty46  /dev/tty55  /dev/tty7     /dev/ttyS6
/dev/tty10  /dev/tty2   /dev/tty29  /dev/tty38  /dev/tty47  /dev/tty56  /dev/tty8     /dev/ttyS7
/dev/tty11  /dev/tty20  /dev/tty3   /dev/tty39  /dev/tty48  /dev/tty57  /dev/tty9
/dev/tty12  /dev/tty21  /dev/tty30  /dev/tty4   /dev/tty49  /dev/tty58  /dev/ttyACM0
/dev/tty13  /dev/tty22  /dev/tty31  /dev/tty40  /dev/tty5   /dev/tty59  /dev/ttyS0
/dev/tty14  /dev/tty23  /dev/tty32  /dev/tty41  /dev/tty50  /dev/tty6   /dev/ttyS1
/dev/tty15  /dev/tty24  /dev/tty33  /dev/tty42  /dev/tty51  /dev/tty60  /dev/ttyS2
/dev/tty16  /dev/tty25  /dev/tty34  /dev/tty43  /dev/tty52  /dev/tty61  /dev/ttyS3
**➜  FreeDiLCD git:(master) ✗ sudo cat /proc/tty/driver/serial**
serinfo:1.0 driver revision:
0: uart:16550A mmio:0xFF110000 irq:21 tx:10430254 rx:2918468 fe:25 RTS|DTR
1: uart:16550A mmio:0xFF120000 irq:22 tx:4959105 rx:352463 fe:1 RTS|DTR
2: uart:16550A mmio:0xFF130000 irq:23 tx:24401 rx:0 RTS|DTR
3: uart:unknown port:00000000 irq:0
4: uart:unknown port:00000000 irq:0
5: uart:unknown port:00000000 irq:0
6: uart:unknown port:00000000 irq:0
7: uart:unknown port:00000000 irq:0
**➜  FreeDiLCD git:(master) ✗ ps aux | grep tty**
root        1561  0.0  0.1  10676  1536 tty1     Ss+  12:13   0:00 /sbin/agetty -o -p -- \u --noclear - linux
root        1565  0.0  0.1  10300  1920 ttyS2    Ss+  12:13   0:00 /sbin/agetty -o -p -- \u --keep-baud 1500000,115200,57600,38400,9600 - linux
mks        20466  0.0  0.2  11144  2048 pts/1    S+   13:56   0:00 grep --color tty
➜  FreeDiLCD git:(master) ✗                 
```

Phil1988, 2025-01-02:
Ok.

I have one last idea what could interfere if everything is correct on your side:
Did you stop the FreeDi service before doing the manual -debug start?

Do it like:
```
sudo systemctl stop FreeDi.service
```

Check if the service is stoped:
```
sudo systemctl status FreeDi.service
```

Start the program manually
```
cd ~/FreeDi/FreeDiLCD && python3 start.py -debug
```

If it doesnt work (likely, bevause otherwise you wouldnt face issues here) I am sure that another background program is active.

I can tell for sure, that you did not follow my guide ([FreeQIDI](https://github.com/Phil1988/FreeQIDI)) but likely someone elses - I my best guess would be the great guide of @billkenney

Reason is, that xindi is running in background.
This might block serial communication and is definetly no part of any of my guides.

kadabr3, 2025-01-02:
Thanks for your help and interest. I'll try to fight this again, if I fail I'll re-install everything using your system image.
Thank you also for your effort in writing this software.

kadabr3, 2025-01-03:
Hi, I have last question about changing the port. I understand that port 80 in the Moonraker address is hardcoded and cannot be changed?

```
 FreeDiLCD git:(master) ✗ cd ~/FreeDi/FreeDiLCD && python3 start.py -debug
Detecting baud...found 921600
comok 1,1281-0,TJC8048X250_011R,165,10201,13B134016643E64F,33554432-0
Firmware is up to date on v1.21
Moonraker host address:
http://127.0.0.1:7125:80
Connecting to UDS /home/mks/printer_data/comms/klippy.sock
Connected!
Error: Failed to parse: http://127.0.0.1:7125:80/server/info
_BootPage: len: 1 data[0]: ee
ActiveScreen: = BootPage
Error: Failed to parse: http://127.0.0.1:7125:80/server/info
Error: Failed to parse: http://127.0.0.1:7125:80/server/info
Error: Failed to parse: http://127.0.0.1:7125:80/printer/info
Error: Failed to parse: http://127.0.0.1:7125:80/server/info
```

Phil1988, 2025-01-03:
Yes that is true.
For now only port 80 is supported.

kadabr3, 2025-01-03:
everything is working now

closed 2025-01-03 by kadabr3

Phil1988, 2025-01-03:
Great!

Custom port setting is included now and will be published in the next release :)