Phil1988/FreeDi #41: [Bug]

3D-Nova, 2025-01-02:
**Describe the bug**
Qidi XMAX 3 with X-6 1.0 board will not flash new OS image via USB drive. System boots as stock printer.

**Expected behavior**
Expected to flash via USB drive and boot with new OS

**To Reproduce**
Write image to USB drive using win32 disk imager
Once complete eject USB drive and plug into XMAX 3 USB port with system off
Turn on system and wait for boot

**Screenshots or picture**
If applicable, add screenshots to help explain your problem

**Base (please complete the following information):**
 - OS: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17
 - Source: FreeDi image v1.20
 - Mainboard: X-6 1.0

Phil1988, 2025-01-02:
Thanks for your report.

On my test, it worked flawless on the first attempt.
Please try to use the USB Stick on a different USB port.
Do this while the printer is shut off and turn it on once it's plugged in.
Start with the one where the WiFi dongle is currently plugged in and let me know if it works.

3D-Nova, 2025-01-02:
I tried on all usb ports and nothing still

3D-Nova, 2025-01-02:
I have tried using raspi imager, win32 imager, multiple different usb
drives, and tried without emmc installed I don't believe I'm missing
anything.

Phil1988, 2025-01-02:
I can confirm that it doesnt work on a quick test.

Unfortunately in this process, Win32 Disk Imager flashed my raid storage instead the USB drive.
I am now facing a data recovery. 
So I am out for the next time.

Sorry.

Phil1988, 2025-01-02:
Tested it thoroughly and it is working perfectly for me.

Flashed the latest full Image matching my printer using Win32 Disk Imager v1.0:
Image file: FreeDi_v1.20_X-Max3_Armbian24.2.0_Bookworm6.6.17.img
[image]

Here are my tests:
#### Test 1 Failed: 
Flashed the Image to an USB thumb drive. Reason for failing: USB thumb drive died.
Booted from EMMC.
Mainboard LED: green
[image]

#### Test 2 Failed: 
Flashed the Image to an microSD card using a USB microSD reader. Removed the EMMC. Inserted the USB microSD reader to any USB slot on the Mainboard.
Mainboard LED: blue
[image]

#### Test 3 **Worked**: 
Flashed the Image to an microSD card using a USB microSD reader. Left the EMMC inserted. Inserted the USB microSD reader to any USB slot on the Mainboard on 
(A) the USB3 slot, no wifi dongle used
Mainboard LED: green
[image]

(B) one of the USB2 slots, no wifi dongle used
Mainboard LED: green
[image]

(C) one of the USB2 slots, with wifi dongle inserted
Mainboard LED: green

(D) the USB3 slots using the extension to the top usb slot, with no wifi dongle inserted
Mainboard LED: green
[image]
[image]

Please test it like this:
1. Flash the latest FreeDi_v1.20_X-Max3_Armbian24.2.0_Bookworm6.6.17.img with Win32 Disk Imager v1.0.
2. Shut the printer off.
3. Leave the EMMC installed.
4. Insert the flashed USB drive to the USB3 slot.
5. Turn the printer on and wait a bit (the sd is slower then the EMMC
6. Check the Mainboard LED light color

3D-Nova, 2025-01-02:
Still cant seem to get it to work. Does the drive need a
specific formatting?

Phil1988, 2025-01-03:
The sd card was FAT32 before I flashed it with "Win32 Disk Imager v1.0".
Maybe you can give it a go with this?

Niceemil, 2025-01-06:
Hi ! i´m also stuck att this point, i have tried 2 different usb-sticks (fat32) and unplugged all other Usb cables from de motherboard, nothing happens. can i do something wrong if i put a Sd cart in the motherboard input ?

Phil1988, 2025-01-07:
You cant do anything wrong, but it wont help as the SD card is connected to the STM32F401 MCU not to the host RK3328.
Have you left in the EMMC while testing with the USB sticks?

Niceemil, 2025-01-08:
Yes the EMMC is still connected, will try it tonight without the EMMC.

Phil1988, 2025-01-08:
The EMMC was a must for me.
I did disconnect it once tho.

So MAYBE you have to try to boot it once with no EMMC attached... then if does some magic... and later the EMMC is needed?
I really dont know as my main focus is to publish the next release to wipe out the current bugs.

Ecaze81, 2025-01-10:
I tried several techniques, moving the emmc, making an 8gb partition on the usb key and nothing works, I ordered an adapter to do the emmc

Niceemil, 2025-01-11:
Hi again! I have now tried to remove the EMMC card to se any difference, but none showed

3 different USB sticks (one sd card to USB adapter)
Fresh formatted through /cmd command
Green light shines on the motherboard
Have used both balero and win32 software 

Also tried 1.03 release 1.20 and the newest one

Is the solution to also buy a EMMC adapter?

robonxt, 2025-01-11:
I would highly recommend getting the EMMC adapter, as using USB to boot would take longer, if at all (in my case, USB booting does not work on my X-Smart3.)

Phil1988, 2025-01-11:
> Is the solution to also buy a EMMC adapter?

If you're in a hurry and want to get things working quickly, then yes. 
You can use one of the options listed here:
https://github.com/Phil1988/FreeDi/wiki/Requirements

The issue for me is that I cannot replicate the problem. 
I trust your feedback that it's not working for you, but in order to find a solution, I need to be able to reproduce the issue. 
I've conducted some deeper tests on my end, but everything works fine here.

Next weekend, I'll get my hands on a different batch X-Max 3 to investigate this further. 
In the meantime, I’m fully dependent on your findings and feedback.

I also found someone whose experience supports my findings, so I'm confident that I don’t have a "unicorn" printer where it works:
https://github.com/QIDITECH/QIDI_PLUS3/issues/27#issuecomment-2081588087

I hope this help you guys!

Bubu0070, 2025-01-16:
Hi, I use X-smart 3 printer, but FreeDi not booting from USB stick. What am i doing wrong? I tried several USB flash drives.

Phil1988, 2025-01-16:
> What am i doing wrong?

You did not read this issue report, did you? :)
I invite you to do that now and then you know whats the latest findings.

Bubu0070, 2025-01-16:
`mks@mkspi:~$ dmesg | grep -i boot
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Kernel command line: root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian
[    0.003001] printk: bootconsole [uart8250] disabled
[    0.013607] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.015105] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.016645] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    2.670959] mmcblk1boot0: mmc1:0001 8GTF4R 4.00 MiB 
[    2.678064] mmcblk1boot1: mmc1:0001 8GTF4R 4.00 MiB 
[    5.758391] systemd[1]: Configuration file /lib/systemd/system/bootsplash-hide-when-booted.service is marked executable. Please remove executable permission bits. Proceeding anyway.
[   52.489546] irq 58: nobody cared (try booting with the "irqpoll" option)`

Bubu0070, 2025-01-16:
<~$ dmesg | grep -i boot
[ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[ 0.000000] printk: bootconsole [uart8250] enabled
[ 0.000000] Kernel command line: root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian
[ 0.003001] printk: bootconsole [uart8250] disabled
[ 0.013607] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[ 0.015105] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[ 0.016645] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[ 2.670959] mmcblk1boot0: mmc1:0001 8GTF4R 4.00 MiB
[ 2.678064] mmcblk1boot1: mmc1:0001 8GTF4R 4.00 MiB
[ 5.758391] systemd[1]: Configuration file /lib/systemd/system/bootsplash-hide-when-booted.service is marked executable. Please remove executable permission bits. Proceeding anyway.
[ 52.489546] irq 58: nobody cared (try booting with the "irqpoll" option)>

Phil1988, 2025-01-16:
> What else can I try?

Please show me
```
cat /boot/armbianEnv.txt
cat /proc/cmdline
```
and
```
sudo apt-get install u-boot-tools
fw_printenv
```

Bubu0070, 2025-01-16:
<$ cat /boot/armbianEnv.txt
cat: /boot/armbianEnv.txt: No such file or directory
mks@mkspi:~$ cat /proc/cmdline
root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian>

Bubu0070, 2025-01-16:
<$ sudo apt-get install u-boot-tools
[sudo] password for mks: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
u-boot-tools is already the newest version (2019.01+dfsg-7).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
mks@mkspi:~$ fw_printenv
Cannot parse config file '/etc/fw_env.config': No such file or directory>

Bubu0070, 2025-01-16:
[image]

Bubu0070, 2025-01-16:
thanks for trying to help

Phil1988, 2025-01-16:
Thanks for that information.
I dont see anything that looks promising to me yet.
I will get my hands on a stock machine soon and have a deeper look.
Meanwhile please be patient :)

Bubu0070, 2025-01-16:
Thanks, I wait patiently

McSneaky, 2025-01-16:
Tried flashing with balena etcher. Down are commands you asked to run

Noticed, that my motherboard has boot and rst button. They seem to not do anything tho. Can't get it booting from USB anyhow

I'll try to upload pictures too as soon as I get my phone working again..

**$ cat /boot/armbianEnv.txt**
```sh
cat: /boot/armbianEnv.txt: No such file or directory
```

**$ cat /proc/cmdline**
```sh
root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian
```

**dmesg | grep -i boot**
```sh
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Kernel command line: root=UUID=5dcae443-0796-4029-ae6c-bf5bd2a37067 console=uart8250,mmio32,0xff130000 console=tty0 rootflags=data=writeback rw no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0 bootsplash.bootfile=bootsplash.armbian
[    0.002996] printk: bootconsole [uart8250] disabled
[    0.013588] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.015073] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.016638] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    2.406654] mmcblk1boot0: mmc1:0001 A3A551 4.00 MiB 
[    2.412677] mmcblk1boot1: mmc1:0001 A3A551 4.00 MiB 
[    5.606001] systemd[1]: Configuration file /lib/systemd/system/bootsplash-hide-when-booted.service is marked executable. Please remove executable permission bits. Proceeding anyway.
[   57.447973] irq 58: nobody cared (try booting with the "irqpoll" option)
```

**sudo apt-get install u-boot-tools**
```sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
u-boot-tools is already the newest version (2019.01+dfsg-7).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
```

**fw_printenv**
```sh
Cannot parse config file '/etc/fw_env.config': No such file or directory
```

**Full dmsg**

<details>

<summary>CLICK TO EXPAND</summary>

```sh
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.16.20-rockchip64 (root@kenneth) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #trunk SMP PREEMPT Mon Jul 25 01:58:45 UTC 2022
[    0.000000] Machine model: Makerbase mks-pi
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x3fdb2040-0x3fdb3fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[... 470 lines cut ...]
[   57.448094]  handle_level_irq+0xd0/0x160
[   57.448101]  generic_handle_irq+0x30/0x48
[   57.448109]  rockchip_irq_demux+0xd8/0x258
[   57.448118]  generic_handle_domain_irq+0x3c/0x60
[   57.448126]  gic_handle_irq+0x98/0xc8
[   57.448135]  call_on_irq_stack+0x28/0x50
[   57.448143]  do_interrupt_handler+0xd8/0xf0
[   57.448151]  el1_interrupt+0x38/0x88
[   57.448160]  el1h_64_irq_handler+0x18/0x28
[   57.448166]  el1h_64_irq+0x74/0x78
[   57.448172]  arch_cpu_idle+0x18/0x28
[   57.448179]  default_idle_call+0x40/0x184
[   57.448187]  do_idle+0x1fc/0x270
[   57.448196]  cpu_startup_entry+0x28/0x68
[   57.448204]  rest_init+0xe0/0xf0
[   57.448210]  arch_call_rest_init+0x10/0x1c
[   57.448220]  start_kernel+0x9b4/0x9ec
[   57.448227]  __primary_switched+0xa0/0xa8
[   57.448235] handlers:
[   57.448239] [<0000000016de2f8b>] irq_default_primary_handler threaded [<000000009ba5b825>] regmap_irq_thread
[   57.448264] Disabling IRQ #58

```

</details>

Phil1988, 2025-02-04:
A few updates from my side...

First things first: There is no break through for USB booting from my side.
I had a small time frame and all I did was backing up the system for a later comparison.
I wasnt even able to test the USB boot :(

Meanwhile user @protens1 tried to bypass the EMMC reader.
He used a USB stick to mount and copy FreeDi onto his stock system.
By doing this he didnt need an EMMC reader but could use a random USB stick.
This is risky and I do not recommend it, because errors can result in the need of flashing the EMMC with a EMMC-Reader 😄
But for all brave users out there, here is his great guide:
https://github.com/Phil1988/FreeDi/wiki/Flashing-EMMC-via-USB-Stick-(No-EMMC-Reader-Needed)

Phil1988, 2025-06-24:
While developing FreeDi v2.00 I had my hands on different machines and I am now sure that this is nothing I can solve.
Reason is the UBOOT on the stock EMMC is not enabled to allow USB booting from a certain release.

So for some users it may work, for others it doesnt work.
As soon as you flashed the EMMC with FreeDi it is open for being able to boot from USB, but I understand that the intention is mostly to "have a test" on FreeDi while still having the stock OS.

As this is a dead end, I will close this issue and rename it, because its not a bug.

closed 2025-06-24 by Phil1988