Phil1988/FreeDi #42: [Bug] Display issue at live Z Offset Adjust

bronsn, 2025-01-02:
X-Smart3

**Describe the bug**
If you go into live Z Adjust during a print, the back button wont work, so the only option is to press cancel print and then "NO", after yo come back to the main Menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots or picture**
Took a Short:
https://youtube.com/shorts/aFeBWNXo1Ok?feature=share

**Base (please complete the following information):**
[image]

 - Mainboard: X-6

Phil1988, 2025-01-02:
Thanks for the report and the short :)

It is fixed and will be pushed/updated with the next release.

Phil1988, 2025-01-10:
@bronsn It is fixed with the new release.
Please check it and verify :)
https://github.com/Phil1988/FreeDi/releases/tag/v1.30

bronsn, 2025-01-13:
Yes this issue is fixed with the new patch.
Thank you for your great work.

Phil1988, 2025-01-13:
You're welcome 😊

closed 2025-01-13 by Phil1988