Phil1988/FreeDi #43: [Bug] Stepper Driver Extruder OTPW Flag Set

colin-small, 2025-01-03:
**Describe the bug**
Since upgrading to FreeDi, I have printed only in PETG. Today, I tried to print in ASA. I heated the chamber to 65 and the bed to 110 and the nozzle was 270.

About 35% into the print of a small part, I noticed this OTPW warning about the extruder stepper driver.

I monitored it and about 56% into the print klipper shutdown.

The toolhead temp when I was watching said around 99 temperature.

I don't typically print in ASA or utilize the chamber heater or super high plate temps, but when I did before with the stock firmware, I did not have this issue.

Is there a configuration value difference somewhere perhaps?

[image]

 
**Expected behavior**
The stepper driver extruder does not overheat and crash klipper.

**To Reproduce**
Not sure how to reproduce but I am pretty confident that if I try the print again it will have this issue.

**Screenshots or picture**
See above

**Console output**
Had to shutdown and I no longer see console information.

**Base (please complete the following information):**

X Max 3

 - X3seriesLCD firmware version: This is FreeDi 1.21
 - OS: Armbian-unofficial 24.2.0-trunk bookworm (FreeDi 1.21)
 - Source: FreeDi image
 - Mainboard: Stock mainboard X Max 3

Phil1988, 2025-01-03:
Did you really compared and adjusted the` run_current` of your stepper according to your backup?
Did you use a preset for ASA printing?
65°C chamber temp is pretty high. 
I would recommend to take it down to 50°C (thats my standard).

I had something similar the other day with a new FreeDi user.
His print failed, that was sliced with a custom print-profile that worked in the past (stock).
Took 3h to figure out that the new klipper v0.12 is simply a lot faster then v0.10 on stock Qidi.
He needed to enable cooling (PLA) and problem was solved. 
Now it is printing faster.
Maybe your extruder is pushed over the limit by a custom preset?

FreeDi does not change klipper.
If thats an issue for you, the source is not me, nor is it FreeDi :)
Blame candidates being Qidi, klipper or your print settings :D

colin-small, 2025-01-03:
The run current is 0.714 in the old config and the new one for the extruder stepper.

I did modify the Qidi ASA profile to have the higher temps for the chamber and plate. I tried checking the Activate temperature control option in Orca but it didn't seem to work. I manually had to turn the temperature up again for the chamber when it started printing.

I was thinking maybe these 65 was too high, I can try the default 50 and see if that works. What do you heat your bed and nozzle to for ASA?

I rarely print in ASA and I was just trying to print a mount for a Cartographer probe I have coming. I almost exclusively print in PETG.

colin-small, 2025-01-03:
I'm going to give it another try. I am going to use the official Qidi studio slicer with the default settings for everything except the wall count, supports, and infill. It defaults to chamber of 55 and a bed of 90 with a nozzle of 250 for first layer and 270 after. This should work fine, it is all of their defaults for anybody who owns the printer. It is also Qidi ASA which this profile is specifically made for. I'll let you know what happens.

colin-small, 2025-01-03:
I resolved the lines issue by disabling the chamber heater and letting the chamber just stabilize at a temperature with the bed and nozzle hot. I thought I also disabled the exhaust fan but it seems to have been running at 100% the entire time... 

Not sure why, having the chamber heater running seems to have caused the lines.. it's definitely related to temperature and possibly pressure..

Phil1988, 2025-01-04:
My ASA settings:
Nozzle first layer 250°, 270°C at the rest.
Bed at 90°C.
Chamber at 50°C.
Part fan off for the first 3 layers, betweeen 10-50% at the rest.
Exaust fan off!
Side blower fans (I have the stock on the right and [added one myself on the left](https://www.printables.com/model/724570-qidi-x-max-3-left-blower-fan-duct)) both off!

Please dont mix issues.
I can not assist everyone at every problem related to 3d printing.
Klipper/Mainsail/Print setting issues should be solved somewhere else.

Its not that I am not willing to help, but FreeDi itself consumes a lot of my spare time already and I simply cant.
I hope you understand :)

closed 2025-01-04 by Phil1988