Phil1988/FreeDi #46: [Feature request & Anouncement] Support for Plus4

Phil1988, 2025-01-06:
# Plus4 Support

I've been receiving more and more requests to support the **Plus4**, so I wanted to provide an update on the development progress for everyone interested.

---

## Current Status  
**FreeDi successfully installed on a Plus4** 
Thanks to some amazing testers, FreeDi is now running on Plus4! 
Here's what we’ve accomplished so far:
- FreeDi software is working on a clean armbian system.
- A compatible firmware for the display has been created and successfully transferred via serial (automated flashing).
- The display is operational.

---

## Challenges Ahead  
There are a few hurdles to full usability.
**The biggest challenge is that numerous klipper files have been modified on the stock system.**
These modifications needs to be integrated in a proper way for non klipper supported hardware/functions (that are necessary for the Plus4) such as smart_effector, qdprobe, chamber_heater etc.
Without this, there cant be mainline klipper support and no updates.
Until this is achieved I will not release anything officially.
FreeDi stands for an open system and I don't want to offer half measures.

These modifications have to be integrated in a proper way and this takes time - especially as I dont own a Plus4 and cant develop directly at it. 
Time zone dependent time shifts between me and testers slows it down additionally.

---

## Development Timeline  
Unfortunately, I can only contribute my analysis and development work on the X3 series to a limited extent.
Therefore, development will proceed at a slow pace until I find more time in April 2025.
I also cannot guarantee anything.

If you'd like to help speed things up, feel free to:  
- **Donate a Plus4 or Q1 Pro** 😄  
- Collaborate as a developer or experienced tinkerer.  

---

## Get Involved  
Developers and enthusiasts are welcome to contribute and work together on these challenges.
Help is welcome for comparing official klipper with "qidi klipper" and expanding offical klipper by the missing functionality.

Feel free to ask any questions or share your thoughts here! 
- Also if you are not a developer :)

RedF81, 2025-01-09:
Hi, 

I'm interested in your FreeDi for the Plus 4.
I am not a Klipper expert, not a developer, but a tinkerer, maybe I can help to lead the +4 into freedom. Even if it's just to be a beta tester.

Phil1988, 2025-01-09:
Thanks @RedF81 
Beta testers are welcome and an important role for me as I dont own this printer :)
I will notify you here once I need you as beta tester.

In the meantime I need more like "klipper experts" :)

Phil1988, 2025-01-29:
### Teaser how FreeDi would look like on a Plus4 😈

[image]

[image]

pmbroth, 2025-01-29:
Just posted this on Qidi Facebook Users forum.  :-)

rhiz0matic, 2025-01-29:
Question, how did you solve the smart_effector, qdprobe, chamber_heater custom hardware in klipper for the xplus3? just curious.

Nebur99, 2025-02-04:
> Developers and enthusiasts are welcome to contribute and work together on these challenges. Help is welcome for comparing official klipper with "qidi klipper" and expanding offical klipper by the missing functionality.

@Phil1988 
I stumbled over your project looking for info regarding installing vanilla klipper on the plus4. No idea how to contact you other than replying here publicly.

Already waddled through the messy qidi code (confs, qdprobe, smart_effector, ctlyd...). Goal is to encapsulate things like what is basically 'dual z probes' and automatic z offset calib in a klipper digestible way with a dedicated module.

My plan for now would have been to just go headless and avoid the complexities regarding the stock display (unless of course you share the display fw & sw requirements on the host).

### I would appreciate if could give me a jump start :
- Does the basic info in the wiki of this project and your repo 'FreeQidi' apply to the plus4? I mean th & board mcu flashing, os image etc.  (comparing qidi's board pdfs things seem to look very similar hardware-wise)

- Would you mind sharing the current progress? I looked in the dev branch but no plus4 related stuff was evident at first glance.

And - just in case - no worries about me bricking stuff and blaming you! ;-)
I'm not a 'normie' accidentally jumping into the deep end...

Nebur99, 2025-02-04:
> I hope this information helped you :)

It did! Thanks for the quick reply! 
Just had a look under the tool head cover - pcb seems to have an unpopulated header hope that's SWD. Will check the discord for more info.

IAmSHOCK, 2025-03-31:
Hi. I'm a software engineer and tinker, and I have the Q1 Pro. I'm happy to help in any way I can.

weed2all, 2025-04-01:
Any news on plus 4?

Phil1988, 2025-04-01:
Yes!
I am working on the screen firmware already.
I cant tell when there will be a stable but I expect an alpha testing phase soon :)

millerad7, 2025-04-01:
Is it possible to update everything in the Plus 4 to get away from the custom repos and just use Klipperscreen from a PI until you get FreeDi up and running for the Plus 4?  I have updated everything, software-wise, on my printer with the exception of Klipper and Moonraker so as not to brick my printer.  I would like to get these to mainline versions of the software, and if I need to use Klipperscreen until the screen firmware gets worked out, I am fine with that.

Phil1988, 2025-04-01:
> Is it possible to update everything [...] until you get FreeDi up and running for the Plus 4?

Sure this is totally possible.
When finished, I would probably recommend using my image with the provided configs and macros so that we can be sure everything is tailored correctly (even tho I expect the first release to have some bugs).

linouxis9, 2025-04-02:
Thanks a lot for your work @Phil1988! Looking forward the first Alphas

weed2all, 2025-04-03:
> Yes! I am working on the screen firmware already. I cant tell when there will be a stable but I expect an alpha testing phase soon :)

Thank you so much for your work!

Phil1988, 2025-04-10:
At everyone interessted in beeing part of alpha testing (like you asked for it @linouxis9):
I am starting alpha tests soon.

If you are interessted, see here for more information:
https://github.com/Phil1988/FreeDi/issues/216

dareklipi, 2025-04-18:
Hello
I am a hobbyist. 
I'm also not an expert in the field of Linux and klipper, but I have a Qidi 4 plus. 
I can beta test

rhiz0matic, 2025-05-13:
Hello Phil,

I swapped my board on my Plus4 to a Manta M8P and am running mainline klipper, so I've solved several issues (of course not all since I don't have a toolhead board anymore). however I'm using TMC2209 drivers for X&Y instead of TMC2240, but I didn't delete any of the important code and just commented it out, I also know what important changes there were from 2209 to 2204 such as the SGTHRS change from 1 to 100. I'm also using a beacon probe instead of the stock probing and haven't made changes to the screen yet.

but hopefully my files can help. I will share them when I am back home later.

Phil1988, 2025-05-14:
Hey @rhiz0matic,
thanks for your kind offer!
I have everything I need already and the firmware is running pretty promising - so you dont need to share you files :)

I am sure that the display will work with the Manta M8P as I have 2 Manta users already :)

rhiz0matic, 2025-05-20:
So I really like the stock screen layout over klipperscreen, is there any way to incorporate that?
I've found the [hmi and tft files for the screen firmware](https://github.com/QIDITECH/QIDI_PLUS4/archive/refs/tags/Plus4_V1.6.0.zip) in the plus official wiki under source code instead of the .rar, and that it is likely designed in [Nextion's free GUI for GUI design software](https://nextion.tech/nextion-editor/). In addition i've found the connection is an RS422 serial type [(i'm curruntly using the adapter to get the screen to power on with the m8p)] (https://www.amazon.com/dp/B07QN6JVG1?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1) but of course, since i'm using a M8P board the screen shows the "starts abnormally" error.

I plan to download the Nextion editor and play around with it, but I've never used it before and don't have much experience in that area. but i've made it this far, if i make progress i'll let you know.

since I have the rs422 to usb2.0 convertor i also wonder if its possible to change it to event driven instead of time based...

Phil1988, 2025-05-22:
> So I really like the stock screen layout over klipperscreen, is there any way to incorporate that? 

What do you mean by that?

>  In addition i've found the connection is an RS422 serial type 

No, its plain serial, as stated in the wiki.

> I plan to download the Nextion editor and play around with it, but I've never used it before and don't have much experience in that area. but i've made it this far, if i make progress i'll let you know.

Problem is, that its not a Nextion display ;)

> since I have the rs422 to usb2.0 convertor i also wonder if its possible to change it to event driven instead of time based...

I also dont understand what you mean here.

So in summary I didnt understand your last comment :D

As I said, your Manta will work with FreeDi.
If you want klipperscreen, go and buy a HDMI screen and use that 😅

rhiz0matic, 2025-05-22:
> > So I really like the stock screen layout over klipperscreen, is there any way to incorporate that?
> What do you mean by that?
> 
sorry sorry. and you are the expert, not I, so I'm totally open to correction but, I also wasn't clear.  the post you made a while back with pictures of the [freedi screen looked like klipperscreen](url) it's definitely not the stock screen UI. I'm saying i prefer qidi's original UI, and NOT klipperscreen, which of course, is totally your prerogative to move to do what you see fit, I just want to see if I can clone the xindi UI.

> > In addition i've found the connection is an RS422 serial type
> 
> No, its plain serial, as stated in the wiki.

As I understand it RS422 _IS_ serial, that's just the type of serial specifically, which uses differential signals (-t, +t and -r, +r)[image]
. The convertor I used(linked in previous comment) seems to work. but (dog at computer, i have no idea what i'm doing)

> 
> > I plan to download the Nextion editor and play around with it, but I've never used it before and don't have much experience in that area. but i've made it this far, if i make progress i'll let you know.
> 
> Problem is, that its not a Nextion display ;)

Interestingly actually! It (as you probably already know) is a [Taojingchi](http://wiki.tjc1688.com/product/X2_serial_compare.html) screen which seems to be a [Nextion cloner](https://www.tokopedia.com/syalis-electrical/nextion-2-8-inch-unofficial-tjc-display-resistif-capasitif-syalis-1731106359122232700), and when looking through their pages on programming their screens they [have some screenshots of the software](http://wiki.tjc1688.com/grammar/hmi_logic.html) and it looks to be a [Nextion clone](https://nextion.tech/nextion-editor/), if it's not nextion itself. ([here's a youtube video that appears to confirm it)](https://www.youtube.com/watch?v=zfbmuZtlWRQ)
so not a Nextion, but definitely a Nextion CLONE! 😅 

> 
<s >> > since I have the rs422 to usb2.0 convertor i also wonder if its possible to change it to event driven instead of time based...
> 
> I also dont understand what you mean here.
> 

what I mean is, with a conversation with stewy, he informed me that the hardware uses 912kbps serial communication right?, and it is a time polling based event handler in software that polls the serial bus to the screen 120k times per second instead of a signal/event driven, so he said if they made the hardware/software event driven with USB2 based communication interface it'd be nice.</s>

Rereading his comment, i realize a  serial to usb conversion probably won't work to change it to signal/event driver. since it seems it needs to run natively on usb 2.0

> So in summary I didnt understand your last comment :D
> 
> As I said, your Manta will work with FreeDi. If you want klipperscreen, go and buy a HDMI screen and use that 😅

But I DON'T WANNA i wanna use Qidi's UI!, lol

also if you don't mind. I'm having trouble with my chamber heater on the manta, do you mind if you can ask those manta users if I can reach out to them to help me troubleshoot?

Also thank you for taking the time to respond! you're too kind!

Phil1988, 2025-05-24:
Your first link is dead :)

This project doesnt provide support for the stock UI.
You can use the stock printer untouched with the stock screen, but you will get my UI.
Reason is, that I wanted to have more settings/options and a different menu structure and a faster responding screen.

> As I understand it RS422 IS serial [...]

Yes, there are different serial protocols and currently UART, RS232, RS422 and RS485 comes to my mind.
As not specified differently, "serial" often referes to UART and the other are called by the protocol (you would say "RS485" and not "serial").
To make it clear: It uses UART serial.

> It (as you probably already know) is a [Taojingchi](http://wiki.tjc1688.com/product/X2_serial_compare.html) screen which seems to be a [Nextion cloner](https://www.tokopedia.com/syalis-electrical/nextion-2-8-inch-unofficial-tjc-display-resistif-capasitif-syalis-1731106359122232700),

As far as I know, TJC is the manufacturer of origin that sells in CN only and nextion bought global maket rights outside china.
Its not a clone, they are acutally the same (or very similar) hardware with only slight differences in the bootloader.

About the chamber heater:
It would be better to ask somewhere else 😅
I cant to "man in the middle" communication because of time purposes.

I think discord channels would be the best idea where you can tell your problem and ask for help :)

Phil1988, 2025-06-01:
Thanks @scrps for you kind words :)

At the moment I am good but its always nice to have people in the community that can jump and help if there are hard to solve issues 👍 

For you and everyone here, I [released an open beta yesterday](https://github.com/Phil1988/FreeDi/issues/243) :)

scrps, 2025-06-02:
@Phil1988 Fantastic! I'll be flashing it in short order! Thanks again!

Phil1988, 2025-06-24:
@RedF81 @rhiz0matic @Nebur99 @weed2all @millerad7 @linouxis9 @dareklipi @scrps 
I’m excited to announce that FreeDi v2.00 is now out - and the Plus4 is officially part of the family! 🎉

You’ll find a ready-to-flash image in the [Releases section](https://github.com/Phil1988/FreeDi/releases/tag/v2.00), packed with all the new Plus4 features. 
For [setup instructions](https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Requirements), head over to our Wiki.
There’s also a [dedicated Plus4 installation page](https://github.com/Phil1988/FreeDi/wiki/Step-%232:-Flashing-the-toolhead-(Plus4)) waiting for you.

Happy printing ;)

tekknix, 2025-06-28:
Another thing:

Does this make really sense?

On the Plus4-Printer.cfg:

`sensor_list: temperature_sensor Toolhead_STM32F103XE, temperature_sensor Toolhead_STM32F103XE, temperature_sensor Toolhead_STM32F103XE, temperature_sensor chamber_sensor
`
Why 3 times the Toolhead-Sensor?