Phil1988/FreeDi #47: [Minor Bugs and Considerations] My Experience with x-Smart 3

theclue, 2025-01-06:
I spent the past nite doing the hard job on my x-Smart 3 and it seems to work incredibly better to me (still have to perform tuning and printing yet to be honest, but at least it boots great). Man, THIS IS AWESOME.

After a little nap I'm here to share my thoughts and to report some very very very minor issues but it think it's good to be reported anyway.

## 1- Mismatch versions in releases page 

Altough I've downloaded the file for x-Smart 3 v1.20 in the releases page _and the file inside the rar file *is* labelled v1.20_ this is actually the 1.03b. This is what it was flashed, and I'm totally sure it was: in facts, the LCD flashing didn't start. I upgraded OTA to v1.21 from Mainsail Web Interface and it upgraded perfectly (also the LCD flashed ok).

## 2- Printer configurations errors cause LCD stops working

After tweaking with printer.cfg, the LCD stopped working (stucked on boot screen). Launching the python script from terminal, I was able to spot that there was an error in the printer.cfg file itself (duplicated entry). 
Altough it's weird that an error in that file could trig a fatal error into the LCD, i guess it is ok...if you know it :) perhaps it could be useful to say it in the wiki. Also, having a clue there about launching the LCD start.py file from terminal for debugging would (I was aware of it only because i hang out into the x-Smart 3 threads here).

## 3- Italian translation doesn't fit

The italian translation is not ok for x-Smart 3. I guess it's because buttons are smaller here. Strings should be made shorter to fit into the buttons and boxes on this specific model of printer. I was not able to find the strings file in the source code, thou. If you can point me up there I'll be happy to push a commit :)

## 4- System Loads won't show in Mainsails until all klipper FWs are synced

The "System Loads" box in Machine section in Mainsaill didn't popup to me at first boot. Thus, I was not able to check about klipper versions on mcu and the toolhead. I was sure about the stock klipper version, so I flashed even without checking.
After the klipper versions have been put on sync, the box started to show, working perfectly. I suspect this is the normal behaviour actually, so I think an edit to the Wiki could be useful to clarifiy this.

## 5- Misleading Klipper versions hint box in Mainsail

(this is more a klipper issue I guess). After updating with the toolhead unique ID and reboot, klipper provided me the info about the mismatching versions with the MCU and the toolhead. As a first try, I used your precompiled file for the MCU for flashing, but the error for the MCU stayed.
I realized that the version didn't match for just a VERY minor element at the end (the commit sha and the 'dirty' postfix...which I actually don't know what it means, btw). It was minor but perhaps enough to make not fit (spoiler alert: it wasn't), so I compiled the MCU file by myself and re-flashed. The versions were now perfecty fitting (except for the 'dirty' postfix) but the error stayed.

I finally realized that although it's labelled "MCU", Mainsail was referring to the toolhead instead. which it wasn't upgraded yet. When I did (with your precompiled one, this way) the problem was solved. This was very time consuming to me to spot, I think it could be noted in the Wiki :)

## 6- make menuconfig for toolhead could be different

I noted a very minor difference from your screenshot and what I got in my system for configuring the toolhead FW for compiling. The last line was _"GPIO pins to set at microcontroller startup"_. Pratically, I had not the "(NEW)" postfix. It's probably nothing, but since compiling the FW for the toolhead is a very worrying step...here it is :)

## 7- Somewhat difficult to fit backup printer.cfg with the new one.

It was very time consuming to me, but very linear...except for the last part. My backup printer.cfg file has this at the end:

```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [input_shaper]
#*# shaper_type_x = mzv
#*# shaper_freq_x = 60.8
#*# shaper_type_y = mzv
#*# shaper_freq_y = 50.6
#*#
#*# [probe]
#*# z_offset = 1.010

[...]
```

I was not aware of this particular section in the klipper setup procedures. Auto-generated, __but commented__, stuff?  Perhaps it's a QIDI custom?

BTW I took those stuff and put inline in the "regular" (uncommented) syntax in your file. But I'm not sure it's the right thing to do.

I'm a bit worried about the probe offsets, since in your example file

`z_offset: 1.549`

 it differs from around 50%. This looks too big for those little differences due to diffent x-Smart 3 minor revisions.

## 8- Bed Mesh size for x-Smart 3 is not correct

Altough in your example file they are more or less ok

```
[bed_mesh]
speed:150               
horizontal_move_z: 2.6   
mesh_min:30,15
mesh_max:170,170  
probe_count:9,9     
algorithm:bicubic
bicubic_tension:0.2
mesh_pps: 4, 4
```

in the one that comes in the img, mesh_max is probably one coming from x-Max. Additionally, 9,9 points looks a bit too much to me for the smaller x-Smart 3 bed (personal opinion ofc)

## 9- Mainsail OTA upgrades availability differs from kiauh ones.

I realized that altough upgrade manager in Mainsail says everything is up-to-date, kiauh says there are some instead. I suppose these are stuff you didn't push upstream yet. I was wondering if you could clarify how the upgrade mechanism works on FreeDi: one should stays to web upgrade manager only? One should use kiauh when available with no fear? Are they equivalent?

This is for a cold start :) Now I'm going to start tuning and printing and i'll let you know if some other stuff will come. Tnx allot for this WONDERFUL firmware!

theclue, 2025-01-06:
After triggering the buttons on LCD for performing bed mesh. z offset and input shaping I've seen that the commented section at the end of the printer.cfg was indeed added with the relevant values. Yeah, I feel a bit stupid now :D

Said that, I was wondering if it's normal than the calculated value `z_offset = 1.519` doesn't show on Mainsail UI, which still shows Z-Offset: 0.000 instead...

Phil1988, 2025-01-07:
Feedback is always welcome, so thanks for taking your time to report this @theclue 

I am glad that you are happy and will work through your points :)

## 1- Mismatch versions in releases page 

That is right and I am aware of it.
I need to create the image files before I publish the new release (because they need to be included).
This means that the version at the image will always be one version before.
If you have an idea how to solve this, let me know ;)

## 2- Printer configurations errors cause LCD stops working

The duplicate error is fixed https://github.com/Phil1988/FreeDi/issues/34 and will be published with the next release.
The hint for reporting and debugging bugs is in a whole chapter [[How to report bugs](https://github.com/Phil1988/FreeDi/wiki/Bugs!)](https://github.com/Phil1988/FreeDi/wiki/Bugs!).

## 3- Italian translation doesn't fit

Unfortunately there is no such a thing as a strings file for different languages.
The display UI is completely integrated in the firmware.
I invite you to learn about the [Interaction between software and the display](https://github.com/Phil1988/FreeDi/wiki/Interaction-between-software-and-the-display).
For now you can tell/show me what to change and will fix it with one of the next releases.
For a very extended "maybe", I will think about outsourcing languages.
Reason is that this feature will lead to much work on the display firmware and a different communication structure between display and printer (+ it has to be tested) - and I think that this work is not worth the value.

## 4- System Loads won't show in Mainsails until all klipper FWs are synced
and 
## 5- Misleading Klipper versions hint box in Mainsail

I think it came from the not yet adjusted serial port for the [mcu MKS_THR].
I left a note at the wiki to cover your point `#4` and `#5`.
And sorry that this costed you some time!

## 6- make menuconfig for toolhead could be different

Thats indeed a very minor thing and I can not cover every possible detail for everyone at the wiki :)
The compiling section is for experienced users (I called it "hardliner") and just as a helping hand among the settings.
So if anyone is not familiar with it, better reach out to me.
As for you: This section was not necessary anyway :)

## 7- Somewhat difficult to fit backup printer.cfg with the new one.

For the z-offset, please perform a calibration either via the display (Settings -> Advanced -> Z-Offset):
[image]

Or via the WebUI Console by typing `PROBE_CALIBRATE`:
[image]

After this carefully check the values.
This is mostly a rough calibration.
For the fine tuning I recommend printing something and adjust the offset via the "baby steppings" feature by Mainsail or the Display.

I get your point about the printer.cfg.
The section
```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
```
Is the auto-generated section for many things like input shaper, bed mesing, z-offset etc. and marked with the sign `#*#` for the parser to notice.

Problem is that I dont know where to place a possible hint and where to stop with the "hints".
I don't think I can create a 100% complete guide that covers every aspect that a reader might stumble across.
Or to put it another way: If I included everything, the instructions would become very large and nobody would read them. 
It is already the case that some bug reports are opened because users have not read the wiki completely. 
Expanding it even further would further reduce the willingness to read it.
What do you think about this?

## 8- Bed Mesh size for x-Smart 3 is not correct

Yes I know and fixed that.
Need to publish the next release soon, but new features, bug fixes + testings needed some time.
You will then find it at https://github.com/Phil1988/FreeDi/tree/master/config_section/X-Smart3

## 9- Mainsail OTA upgrades availability differs from kiauh ones

Well kiauh is already a little advanced and a great tool for experienced users and devs.

For everyday users, the onboard Moonraker Updater (the one you see in Mainsail) should be used.
But: I tend to do updates if you want a bug fixed or missing something.
Its a little "never change a running system" advice.

For moonraker and mainsail you can almost blindly update.
FreeDi (hopefully) too, but it may lead to a waiting 4-6min for a possible display firmware update.
So dont do it, if you are in a rush and need to print something right now.

Klipper is a different thing tho:
Updating this can result in the subsystems (mainboard and toolhead MCUs) needing a klipper update as well.
This can be done in like 10min, but I have not covered this yet in the wiki (no time yet) and will do this when I see the need.

theclue, 2025-01-07:
> Problem is that I dont know where to place a possible hint and where to stop with the "hints".
> I don't think I can create a 100% complete guide that covers every aspect that a reader might stumble [...]

You're totally right, the wiki cannot be a "Klipper 101 guide". It woulsd be an infinite document.

_but..._

Since the x-Smart 3 is probably an appliance, primer machine for many users, I would only add a quick note in the wiki page when you explain the importance to sync one's backup printer.cfg with the new one. Something like

"don't copy the section blah blah blah since this will be auto generated by auto-tuning features, just like the stock firmware"

or 

"be aware that the section blah blah blah will be overwritten with possibly different values when you're going to relaunch the auto-tuning features of klipper"

in my case, for example, input shaper changed dramatically. It prints ok, so who cares, but it was puzzling at first

About the update procedures, I'll follow your advice.

It's not that I don't want to reflash the toolheads at kiauh run. I learned how to do, so I could do that :) 
It's just that I don't wont the break the web upgrade mechanism with diverging updates...

Phil1988, 2025-01-10:
Included in the updated post install guide:
https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section#auto-generated-section-on-the-end-of-the-printercfg

You can find a few changes there now and I hope you like it.
I will close this now as I think its completely covered now :)

closed 2025-01-10 by Phil1988