Phil1988/FreeDi #48: [Bug] Screen wont work properly after adding additional fans to printer.cfg

colin-small, 2025-01-06:
**Describe the bug**

My X Max 3 model came with the additional fan on the back cover for the extruder and the configuration was missing from the printer.cfg that I am using from the FreeDi github. This is covered in issue #26 

After adding the missing fan configuration to my printer.cfg, the fan started working as expected.

However, the screen no longer works. When I shutdown the machine and booted it back up, the screen is stuck at the booting display. It stays that way and never moves to the main UI interface.

I tried simply rebooting, shutdown, boot, etc.. nothing changes the behavior.

Prior to adding the fans to my printer.cfg, the screen booted with no issues at all, every time.

To confirm that it is related to the fans added to the config, I commented out the two sections that I added and rebooted the printer. This time the screen booted to the main UI with no issues, as expected.

**Expected behavior**
The screen should completely boot to the UI after the missing fan is added to printer.cfg.

**To Reproduce**
Add the following configuration to the bottom of the "Fans" section in the configuration. Reboot or shutdown and power up again etc.

## Hotend fan 2 - FreeDi issue #26
[heater_fan hotend_fan2]
pin:MKS_THR:gpio20
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

## Main board fan - FreeDi issue #26
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0

**Screenshots or picture**
[image]

**Console output**

**Base (please complete the following information):**
 - X3seriesLCD firmware version: 1.21
 - OS: Armbian-unofficial 24.2.0-trunk bookworm
 - Source: FreeDi image
 - Mainboard: Stock

colin-small, 2025-01-06:
I tried to comment out just the main board code block from the printer.cfg and rebooted. Now it is flashing the screen firmware for some reason?

[image]

Phil1988, 2025-01-06:
I dont think its a bug, but a wrong printer.cfg.
Maybe you integrated the board fan incorrectly and also with the wrong type (no [section name] given and also as "heater_fan" while it should be a "temperature_fan".

Please refer to my wiki, as its there and the go to for informations like that:
https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section#fans

Change:
```
## Main board fan - FreeDi issue #26
pin:PC4
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
off_below: 0
```
to:
```
## Main board fan
[temperature_fan MCU_fan]
sensor_type: temperature_host #This is what the MKS's Temperature sensor is known as.
pin: PC4                      #Set this to your fan pin
max_temp: 80.0                #Set below 85C, MCU can be damaged at 85C
target_temp: 45.0             #Set this to your preferred running temperature
min_temp: 0
shutdown_speed: 0.0
kick_start_time: 0.5
off_below: 0.19
max_speed: 1.0
min_speed: 0.0
control: pid
pid_Kp: 2.0
pid_Ki: 5.0
pid_Kd: 0.5
pid_deriv_time: 2.0
```

If the problem still persists, please attach your printer.cfg here (as zip) and I can fix it for you :)

colin-small, 2025-01-06:
@Phil1988 Yes, this was a configuration issue and was ultimately user error on my part. Placing the config as you suggested worked and now it boots correctly and I see the MCU fan in Mainsail as well.

I do not understand what happened with the random firmware flash, but it doesn't matter at this point. Thank you for pointing out my error, my apologies.

drewp29, 2025-01-06:
Colin - Pin PC4 is the motherboard fan on the back of the printer. It should have been named "board_fan", but they misspelled it. Just an FYI. You want that fan to run all the time or your stepper drivers will overheat. Qidi changed it to run only when the hot end is above 50C. I reduced mine to 20C to keep it on most of the time. It was their attempt at appeasing the dude on YouTube that bitched incessantly about the sound of the motherboard fan.

colin-small, 2025-01-06:
@drewp29 Thanks. The funny thing is the fan on the back was running all the time without this configuration in the printer.cfg. Unless I am looking at the wrong fan. I am talking about the one that is visible from the back cover.

Now with the configuration added that @Phil1988 suggested, it looks like it will kick on if the MCU will go above 45 degrees. So far with my running 6 hour print, it has been running for about 4 hours and has not hit 45 degrees yet, close.. but not 45. I will consider changing it though to be a lower trigger value.

I think we should close this issue though as it is not a bug as I originally thought it was.

drewp29, 2025-01-06:
@colin-small Interesting that your stepper drivers did not throw an error if it isn't running. My situation might be a little bit different, as I print mostly with the chamber temp at or around 60C, and the stepper drivers were having issues before I set it to run all the time. I have since changed to an 80mm fan. Mostly due to turning off interpolation for misrostepping and bumping the stepping up to 32 and 64 microsteps, I was having overheating issues during calibration actions. With the 80mm fan it looks like I can run at 64 and not have any problems. I have yet to test that fully, so take it with the caveat that it is untested as of yet.

Phil1988, 2025-01-07:
> @Phil1988 Yes, this was a configuration issue and was ultimately user error on my part. Placing the config as you suggested worked and now it boots correctly and I see the MCU fan in Mainsail as well.

Glad it was this easy mistake and we could solve it :)

The "random flash" may have happened most likely because you triggered a FreeDi update at some point?
When connection was established again, it flashed the new display firmware.

@drewp29 the "[heater_fan borad_fan]" is not a misspelling by @colin-small , but coming from the original /stock qidi config - yes, they missspelled it.

You raised an interessting point that was out of my sight so far, because I do have an always on fan port there:
The board fan not only cooling the cpu, but also the drivers.

My thought on this:
The fan is trying the keep the cpu below 45°C.
During a print job the cpu has to process things and the fan will turn on.
Can you verify that the above setting with `target_temp: 45.0 ` is insufficient for you?
If yes, would `target_temp: 40.0` result in adequate cooling for the drivers?

If not, then we need to establish a complex cooling logic (at least for the tools we have on klipper) which takes the steppers and the host cpu into account.

Thanks for your thoughts :)

I'll close this issue, as the initial problem was not related to FreeDi and is solved :)

closed 2025-01-07 by Phil1988