Phil1988/FreeDi #49: [Bug] Settings buttons for z-offset, mesh and shaping on LCD don't work on x-Smart 3

theclue, 2025-01-06:
**Describe the bug**
In "settings" panel on LCD, fine tuning buttons (z-offset, shaping, bed mashing) have no effect when pressed.

**Expected behavior**
They should trig an action, but most of the times they don't (altough sometimes randomly they do). In terminal log there is nothing, so it's not a macro configuration issue in my opinion. It really seems that the buttons have no effect.

**To Reproduce**
See attached video. I'll upload a couple more to see if there is some sort of consistency.

https://github.com/user-attachments/assets/a1db9c84-bd3e-4673-934c-d2510c1ed995

**Console output**
Nothing in the log

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.21
 - OS: e.g. Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17
 - Source: v1.21 freeDi image
 - Mainboard: stock Mainboard X-6

theclue, 2025-01-06:
Ok, I've found a pattern. I was able to make the buttons working: apparently not the entire surface of the button is acrtive, but only a small area a bit right from the middle, as it's shown in the video.

Same behaviours for the three buttons set.

Phil1988, 2025-01-07:
Yes thats right and a bug in the display firmware.
Thanks for the report, I just fixed it.

It will be published with the next release :)

Phil1988, 2025-01-10:
@theclue The new firmware v1.30 fixes this.
Please check it and verify. Thank you!
https://github.com/Phil1988/FreeDi/releases/tag/v1.30

closed 2025-01-11 by Phil1988