Phil1988/FreeDi #52: [Bug] Thumbnail is different from object currently printing on x-Smart 3

theclue, 2025-01-09:
**Describe the bug**
I think that a photo will explain better than a thousand word:

[image]

Basically, the object printing is different from the one shown in thumbnail on LCD screen. This behaviour tosses out in subsequent printings. Also, the data about ETA, printing times and so on doesn't fit with the ones shown in Mainsail (altough quite near, discrepance are +/- 5 minutes more or less). The filename, instead, does. The thumbnail shown on Mainsail is correct.

**Expected behavior**
If not self self-explanatory, add a clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Print an object, let's call it "A.stl", launching from the slicer
2. Print a different object "B.stl", again form the slicer
3. Select "A" again for reprinting  from LCD file explorer
4. Thumbnails shown is for object "B.stl", filename is "A.stl", printing data on LCD are wrong.

A.stl and B.stl are both in the root of the gcode_files directory (no subfolders)

**Base (please complete the following information):**
 - X3seriesLCD firmware version: v1.21
 - OS: Armbian-unofficial_24.2.0-trunk_Mkspi_bookworm_current_6.6.17
 - Source:  FreeDi image
 - Mainboard: stock Mainboard X-6

theclue, 2025-01-09:
https://github.com/user-attachments/assets/8766b449-3864-4561-b749-231eb31f636c

Phil1988, 2025-01-10:
Please check if this still persists with v1.30 and let me know:
https://github.com/Phil1988/FreeDi/releases/tag/v1.30

theclue, 2025-01-12:
Doing some intensive tests, bug seems to be squashed in v1.30. I'll let you know if I can reproduce it again :)

Phil1988, 2025-01-13:
I am closing this now to get this off my radar.
Please report back here anyways, so I get informed :)

theclue, 2025-01-22:
i spoke too early...bug is still here...

[image]
[image]
[image]

rosenrot00, 2025-01-22:
Can you reproduce it with these two gcode files?

[test_gcodes.zip](https://github.com/user-attachments/files/18512564/test_gcodes.zip)

Most of the gcode commands are removed to simulate a successful print

Phil1988, 2025-01-25:
@theclue any news on this?

Phil1988, 2025-01-28:
@theclue 
Please check the [latest release](https://github.com/Phil1988/FreeDi/releases/tag/v1.40).

Please report back if it fixes this for you (and close with comment of so).

closed 2025-01-28 by Phil1988