Phil1988/FreeDi #54: Webcam for timelapse and just viewing

pmbroth, 2025-01-10:
Configured Crowesnext for  cam for qidi cam, and get this.

[image]

[image]

pmbroth, 2025-01-10:
settings

[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1280x960                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

Phil1988, 2025-01-10:
The device might be incorrect. 

As this is not a FreeDi bug, please check the log, or if you're not familiar with it, feel free to ask for support in the Mainsail Discord channel. 
There are people there who can assist you more effectively with this. :)

closed 2025-01-10 by Phil1988

pmbroth, 2025-01-10:
Thank you - Will do..