Phil1988/FreeDi #55: Z-offset, freezes

pmbroth, 2025-01-10:
X-Max 3

When going into the offset adjustment, I'm assuming, we adjust it like before using the thin black spacer sheet.

When you move the bed up and go negative, and if I go over -1.5~ the adjustments stop working.  I have to abort and try to start again, and try not to go over the negative numbers too much, or it does it again.

I have noticed, after I save the adjustment, the  klipper error screen comes up on the printer for a few seconds, then goes to the main screen - is this because it has  updated a config  file, and  klipper is refreshing?

colin-small, 2025-01-10:
@pmbroth check your position_min: under the stepper_z section of your config. If it is 0 you probably need to set it to a different value like -2

I originally had trouble using the z offset settings due to this and after changing it I was able to configure the offset correctly.

pmbroth, 2025-01-10:
@colin-small   Thank you,  Just checked and the positon_min = -2

Phil1988, 2025-01-10:
Thanks @colin-small for assisting here :)

@pmbroth:
Change it to a lower value for the time adjusting it.

`position_min: -6` is the stock value.
You can try this, but I recommend resetting the value to -2 when adjusting the Z offset is finished to avoid accidental damage.

Phil1988, 2025-01-11:
@pmbroth  
Can you please confirm if this issue is resolved for you?

Phil1988, 2025-01-11:
Im glad its solved.
Happy printing ;)

closed 2025-01-11 by Phil1988