Phil1988/FreeDi #56: [Feature request]

theclue, 2025-01-10:
**Is your feature request related to a problem? Please describe.**
Version number in klipper and Github don't fit

**Describe the feature you'd like**
In my Home Assistant instance I've a dashboard for my 3d printer. It shows, among others, the actual FreeDi version, mined from Moonraker API.

Now it shows: V1.30-0

Also in my Home Assistant I've a set of automations based on messaging coming from GitHub. I thus can listen this repositories for new releases, feed discussions, and so on..

Now it shows "FreeDi v1.30" as last release.

They don't fit. I they do, I can create some automations for notifying about updates among other (endless possibilities).

Ofc I _CAN_ use regular expressions to transform and make them fit, so it doesn't mean they must litterally coincide. It's enough if there is a versioning name pattern which is consistent/constant in time.

I don't think is a huge issue but...who knows :)

Phil1988, 2025-01-10:
Hello,

I dont know where this "FreeDi v1.30" is taken from.
The release tag is "v1.30". Its message is "Release version 1.30"
The release tite is "FreeDi v1.30 (software and image)".
None of these matches exactly with your "FreeDi v1.30".

Im also having a hard time finding a documentation of how moonraker (update manager) precicely interacts with github.

Im not gonna lie, my biggest struggle with the last release wasnt the actual code, but with github and moonraker.
So if you can tell me how this should be done and can give me more information I will think about a suitable workflow.

Just to be precise:
Your title says "Make versions names consistent between **image names** and Gituhb
Do you mean "Make versions names consistent between **moonraker** and Gituhb "

theclue, 2025-01-10:
> I dont know where this "FreeDi v1.30" is taken from.
> The release tag is "v1.30". Its message is "Release version 1.30"
> The release tite is "FreeDi v1.30 (software and image)".
> None of these matches exactly with your "FreeDi v1.30".

[image]

For the former, apparently, the release tag is not exposed, so one must use the release title, which is probably less robust as this is a free text field by definition. But this issue is from Github API.

Same for the latter, this is somewhat related to Moonraker API here instead. In facts, in update manager I have:

[image]

which partly follows the pattern I have (except for the sha suffix).

As-is, until you're consistent in the release title, there's nothing on your side to do. A regular expression will solve the issue.

For the sake of the science, and sssuming that it's a feature useful for someone else in the world which is not me (probably not), the only way to fix this for good is to switch to a CICD pipeline for the creation of the image. But I see that the FW on github is a binary, not a source code, so it won't be easy to make it working, probably not worth the pain anyway. 

Nonetheless I'll dig into the Moonraker API to see if i can find something more consistent...

theclue, 2025-01-10:
For the two in the Universe which will need it, here's the template code to compare the current vs the latest version on FreeDI inside Home Assistant. This creates a binary sensor which can be used for automations and similar.

```
{% set latest  =  states('sensor.phil1988_freedi_ultima_versione') | regex_findall_index('v\d+\.\d+(?:\.\d+)?', 0) %}
{% set current = states('sensor.freedi_x_smart_3_version_freedi') | regex_findall_index('v\d+\.\d+(?:\.\d+)?', 0) %}
{{ latest >= current }}
```

*** Requirements
1. Github integration for Home Assistant. You *MUST* star the Phil1988/FreeDi repository to be able to get the release titles from GitHub into Home Assistant
2. Moonraker integration for Home Assistant.

This assumes that the release number in the form of vXX.XX[.XX] (the third is optional) is present in the release title field, as Github API doesn't expose the version tag into Home Assistant.

Ofc you have to change the names of the two sensors above accordingly to your system.

Hope this could help someone :)

Phil1988, 2025-01-10:
> 1. You _MUST_ star the Phil1988/FreeDi repository
Now this looks like a good guide to me :D

I am closing this and hope other folks with HA may find it.
(I want to track bugs)

closed 2025-01-10 by Phil1988