Phil1988/FreeDi #57: USB drive - not seen

pmbroth, 2025-01-10:
I plgged in a usb drive formatted to fat32

Have been using it pior to upgarde, no issues.  The system doe snto see the thumb drive./  IOd theer anythgtign I need to do to have it seen?

Thank you again

Brian

Phil1988, 2025-01-10:
I have not tested this yet, as its not my workflow (I'm using the WebIF).
Will have a look at it

pmbroth, 2025-01-10:
@Phil1988   I only use it as a backup for important stls on the machine.  I usually print over the network.   But in rare cases, have to copy off of a friends usb drive on the machine.   Thank you.

Phil1988, 2025-01-10:
Completely understandable, but its simply not tested/integrated as display functionality is/was the highest priority.

Btw: It would be really helpful if you could follow the [naming conventions](https://github.com/Phil1988/FreeDi/wiki/Bugs!) for new issues, as this helps improve clarity and organization. Also, paying a bit more attention to spelling would make the issue easier to understand and resolve.
Correcting that on every issue from you takes my time :(

Thanks for your understanding!

pmbroth, 2025-01-10:
@Phil1988   Completely agree.. I only had 2 hours sleep last night.  Apologies for the above.  I did fix them though!  :-)

Phil1988, 2025-01-10:
No worries mate :)

pmbroth, 2025-01-12:
It would be nice to have the USB support as in this photo:

[image]

Phil1988, 2025-01-28:
@pmbroth 
Please check the [release v1.40](https://github.com/Phil1988/FreeDi/releases/tag/v1.40).

I'll close this now as its now featured.
If you have any problems, please answer here and I will reopen.

closed 2025-01-28 by Phil1988