Phil1988/FreeDi #58: Not a bug, or not a Feture, just wanted to send a note about

pmbroth, 2025-01-10:
I was getting duplicate readings on my screen mainsail or fluid,  (MCU_FAN and Main board RK3328)

So, I commented out the below.  My motherboard fan still works as it is always on.  My Mainboard temp is staying around 31c

Just wanted to note this, that I did not need the below, unless you think there is a reason I need it?

[temperature_fan MCU_fan]
sensor_type: temperature_host #This is what the MKS's Temperature sensor is known as.
pin: PC4                      #Set this to your fan pin
max_temp: 80.0                #Set below 85C, MCU can be damaged at 85C
target_temp: 45.0             #Set this to your preferred running temperature
min_temp: 0
shutdown_speed: 0.0
kick_start_time: 0.5
off_below: 0.19
max_speed: 1.0
min_speed: 0.0
control: pid
pid_Kp: 2.0
pid_Ki: 5.0
pid_Kd: 0.5
pid_deriv_time: 2.0

Phil1988, 2025-01-10:
Please give more information.
I assume you started in a clean FreeDi v1.20 image and updated to V1.30 - is that right?
I assume you had this issue on v1.20 as well?
Where is this [temperature_fan MCU_fan] coming from?
I mean why did you not delete the [temperature_fan MCU_fan]  section or why did you inserted the [temperature_fan MCU_fan]  when your stock/backup config didnt had it?
See here: https://github.com/Phil1988/FreeDi/wiki/Post-install-and-config-section

I do appreciate your feedback, but I would be even happier, if you give the important information right away to save time :)

If you are worried, you can share your printer.cfg here (as zip) and I will have a look.

pmbroth, 2025-01-10:
I started with 1.2 then at the same time went to 1.3. 

Yes the same this, duplicate readings, except one said fan

Here is the printer.cfg.  Probably easier, then me typing and typing
[printer.txt](https://github.com/user-attachments/files/18381459/printer.txt)

[image]

closed 2025-01-10 by Phil1988