Phil1988/FreeDi #59: Trying to install moonraker-timlapse, and says out of disk space, on 32gb EMMC

pmbroth, 2025-01-10:
Hello,

Trying to install moonraker-timelpase onto my machine.  My eMMC is 32gb, and the install failed, due  to no disk spacee

mks@mkspi:~/moonraker-timelapse$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            379M     0  379M   0% /dev
tmpfs            92M  2.6M   89M   3% /run
/dev/mmcblk1p2  6.7G  6.7G     0 100% /
tmpfs           456M     0  456M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           456M  4.0K  456M   1% /tmp
/dev/mmcblk1p1  256M   88M  169M  35% /boot
/dev/zram1       47M  9.9M   34M  23% /var/log
tmpfs            92M     0   92M   0% /run/user/1000

It does not seem ike the 32gb emmc is fully allocated

colin-small, 2025-01-10:
You need to expand the file system. It is in the wiki

Phil1988, 2025-01-10:
Ok, this is enough @pmbroth 

I kindly asked you to use a proper naming convention:
https://github.com/Phil1988/FreeDi/issues/57#issuecomment-2583455673

Instead you keep ignoring it:
https://github.com/Phil1988/FreeDi/issues/58
https://github.com/Phil1988/FreeDi/issues/59

I appreciate feedback, but this is not an open forum for every question you have.

**Read my guide.**
I spend a good amount of time writing it and it feels like you skipping it completely and disrespecting my work.

closed 2025-01-10 by Phil1988

pmbroth, 2025-01-10:
@Phil1988   I'm sorry for bothering you.  I did not see the guide.  Only the quick start for new system,.  I just issued the command,  will reboot...

[image]

pmbroth, 2025-01-10:
All is great!  Thank you again!  I have your wiki now.. 
[image]