Phil1988/FreeDi #60: [Bug] after update Error

bronsn, 2025-01-11:
**Describe the bug**
I did a update today to your newest image over Mainsail

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Mainsail -> Machine -> FreeDi and Update

After the Update i got this Screen:

[image]

I restarted the Machine and now:

[image]

bronsn, 2025-01-11:
Thank u are right, i waited to short now after 2 min i restart and the problem was solved )

[image]

Phil1988, 2025-01-11:
Im glad it was no bug and that the update was working completele without any issues (from my side 😉)!

closed 2025-01-11 by Phil1988