Phil1988/FreeDi #61: Disp[ay current print layer out of X layers on Machine printing screen

pmbroth, 2025-01-11:
Hello,

I don't know if this is possible:

When Printing, and the printitng screen is on the display:
See current layer machine is printing, and how many layers for the entire job.

I.E.   Layer 20 out of  100

For the following reasons:

1. Change filament  color at a certain layer.
2. To see how many layers left to print.

Thank you

Brian

scottessner, 2025-01-11:
I would love this too!  I would propose that it replace the total time on the printing screen as we have both the elapsed time and the remaining time at the bottom of the screen. (Forgive my mouse drawn prototype).
[image]

leckert123, 2025-01-11:
This is already shown in Mainsail

pmbroth, 2025-01-11:
Yes, but I do not have my computer on all the time.  I upload, or reprint, without computer

Phil1988, 2025-01-11:
> This is already shown in Mainsail

Yeah, they want this feature on the display and I understand the benefit.
Will definetly be included, but I cant tell you yet when this will be finished :)

scottessner, 2025-01-11:
> This is already shown in Mainsail

By that argument we wouldn't need the screen at all.  The layer information is a valuable indicator of current progress and is very helpful to see while standing at the printer, making it valuable to have on the screen.

Phil1988, 2025-01-11:
> By that argument we wouldn't need the screen at all. 

Good idea!
Lets wipe this project and save my spare time :D

leckert123, 2025-01-11:
Not going to argue with a want, not a need.  Time remaining is important.  Unless you know the exact layer number assigned where you anticipate making a filament color change, I see little value in layer numbers.  But I have a phone. And it's on me all the time. And one browser tab is always open to the Mainsail IP address.  But if you want it on the screen, cool.  

There are things I'd prefer to be on the screen as well, which I believe are of greater value (to me) than layer number, but I'm not going to complain that the product is not tailored to me.

pmbroth, 2025-01-11:
I use layer numbers all the time.   For filament change, as well as estimating how much filament I need.  Yes time estimate is fine.  Layers are more exact then a time estimate.

scottessner, 2025-01-11:
@leckert123 I would not suggest we remove time remaining by any means.  It still remains on the screen in my suggested layout.  I appreciate that we all have different preferences and priorities.

Phil1988, 2025-01-11:
> There are things I'd prefer to be on the screen as well, which I believe are of greater value (to me)

Feel free to open a feature request! 
While I can’t promise anything, I’m always happy to explore ways to improve this project and provide greater value for everyone."

pmbroth, 2025-01-12:
Sample screens I found, one with layer x of y

[image]

breathless19, 2025-01-20:
+1 for this feature!

closed 2025-06-24 by Phil1988