Phil1988/FreeDi #62: [Bug]  Timelapse  videos show up on screen in file section

pmbroth, 2025-01-11:
Files created by timelapse show up in screen file section.. 

If you select the file, it just says cretaing thumbnail.

May  want to filter  out Timelapse videos froim showing.

Phil1988, 2025-01-11:
I dont know what  "Files created by timelapse" are.

Can you please describe in detail what it is, how you created it and share one of those?

pmbroth, 2025-01-11:
This is using the time-lapse video which is created via this site:  It allows you to have automatically a mp4 video time-lapse video created.  Se link below:
https://github.com/mainsail-crew/moonraker-timelapse

pmbroth, 2025-01-11:
I have them working perfectly.  I used the above link, and installed yesterday after I set up the Freedi image on my machine.

pmbroth, 2025-01-11:
@Phil1988 I just looked at the time-lapse copnfiguration.  It defaults to where gcode gets uploaded, I can change the directory to whwre the files are placed.  SO this may eliminate the feature change request.  Tryign it out now.

pmbroth, 2025-01-11:
@Phil1988 what is interesting the timelapse image and video is created in a different directory already, see screenshot.  The folder functon on the display is brirnging  in this directory.  

[image]

as you see this is what I have in the gcode directory under printer_data, 

[image]

Phil1988, 2025-01-12:
Thanks for the information.

The first two files on your screen (without thumbnail image) have no thumbnail and are called similar to the time lapse videos (at the end a gcode_2...).
I guess these are preview pictures of your time lapses.

I think it's because of the thumbs I'm gcodes folder subfolder. Can you please show me the contents of it?

With this information I could add an extension.
I don't think your suggested solution of completely ignoring subfolders is a good way to go.
With 1.30 exactly this (subfolder) feature has been introduced, because there are people who structure their guides on the printer in this way.
https://github.com/Phil1988/FreeDi/issues/35

As soon as I have seen the contents of your thumbs folder, I can think about a solution :)

pmbroth, 2025-01-12:
@Phil1988    ok, I cleaned out the directories, so that you can see what happens after one print...  Here are the contenst after one small test cube
[image]

[image]

orcaslicer 
[image]

pmbroth, 2025-01-12:
Also not  sure why there is a 32x32 png file being created in thumbs

Phil1988, 2025-01-12:
> Also not sure why there is a 32x32 png file being created in thumbs

Its a thumbnail generated by mainsail as you uploaded it.

Then it must be a symlink.
Can you show:
```
ls -la ~/printer_data/gcodes/
```

pmbroth, 2025-01-12:
@Phil1988   Good day!   Happy Sunday!  Here is the result of the command:

[image]

pmbroth, 2025-01-12:
[image]

pmbroth, 2025-01-12:
@Phil1988   I did another test print, and noticed on the screen, If I press the folder button twice a diffrent file is shown that says in the right corner extpic   See below

[image]
[image]

Phil1988, 2025-01-13:
> i cant even get time laps to work ??

To leave this focused on the topic, feel free to start a Discussion about this here:
https://github.com/Phil1988/FreeDi/discussions/categories/q-a

Or ask at the timelapse plugin maintainer to help :)

Phil1988, 2025-01-13:
> If I press the folder button twice a diffrent file is shown that says in the right corner extpic

Yes thats a "hidden menu" for background testing and developement.
Just dont worry about it and use the normal File-Manager menu.

Its fixed now, so these timelapse files wont appear anymore in the File-Manager menu on screen.
It will be included in the next release and you will get that via Update Manager.

I will close this now - if you, however, still have this after the next update, please report it back here :)

closed 2025-01-13 by Phil1988

pmbroth, 2025-01-13:
@Phil1988   Thank you very much!!!!   End-users  :-)