Phil1988/FreeDi #63: [Feature request] Filament change

bronsn, 2025-01-12:
Hello, me again )

Today i tried to do a manual Filament Change with a M600 Macro i use for my Vorons too.
Problem ist that after he stopps att the current Layer, there is no option to load / unload Filament so it is not possible to switch Filament.

Perhaps u can implement a feature to load / unload Filament during a Pause or M600.

Thank you
Regards

Phil1988, 2025-01-13:
Doesnt this menu work for you?
[image]

bronsn, 2025-01-13:
Hey, this menue dosnt appear anymore during the print or Pause/M600.
I can do a short video later.

Phil1988, 2025-01-13:
Thanks, that will help :)

pmbroth, 2025-01-13:
I issued a pause via the slicer -Orcaslicer (Slicing with the pause at desired layer) , set the extruder temp, then clicked settings, and was able to load/unload.    At first, I did not release I had to re-adjust the extruder temp, because when pause was issued, extruder temp is shutdown.  So, I reset the desired 220 temp, and then the load/unload buttons were not greyed out, upon set temp meant.

bronsn, 2025-01-13:
So here is the Video i took.
I tried the color change with M600 and with PAUSE in the Slicers Settings, the result is the same.

No Option to load / unload.

[https://www.youtube.com/shorts/JTxhRX9xDAE](url)

pmbroth, 2025-01-13:
ok, you are not using x-max 3, it is working on it.  Yours is a different machine, that's why..

bronsn, 2025-01-13:
o sry i forgot to write that i´m using the x-smart3 )
Thought you know me cause i posted some issues i have xD

sry

Phil1988, 2025-01-13:
> i´m using the x-smart3 ) Thought you know me cause i posted some issues i have xD

I didn’t remember that - there are too many users here and on Discord that Ive lost track 😅
Thanks a lot for the video, it was very helpful!

As I have reclassified this issue - its not a "feature request" but a bug.
Sorry for this and thanks for your patience.
It is already fixed and it will be included in the next release 🎉

I’ll close this issue now. 
If it’s still not resolved in a version later than 1.30, please tell me.

@pmbroth You described it perfectly: when pausing, the nozzle temperature is turned off for safety reasons and to prevent the filament from cooking and potentially causing clogs. This behavior can be customized in the macros, but it’s the default process I provide (safety first) 😊

closed 2025-01-13 by Phil1988

bronsn, 2025-01-13:
Thank you, glad that you could fixed it.

martin-badin, 2025-04-06:
@bronsn @Phil1988 Is this issue fixed? I have tried M0 and M600 and I don't that it is possible to see a menu with load/unload filament. Or please could someone provide a macro for M600? Thank you

Phil1988, 2025-04-08:
@martin-badin Yes it it fixed.
Can you show a video of what you get when pausing?

martin-badin, 2025-04-08:
> [@martin-badin](https://github.com/martin-badin) Yes it it fixed. Can you show a video of what you get when pausing?

I attached a video. I am using M0 command to stop printer but I don't see a menu to load/unload filament.

https://files.fm/u/vgvmtxj8w9

martin-badin, 2025-04-10:
@Phil1988 Have you had time to watch the video? Do you know what I am doing wrong? Thank you for your help. 

I saw you have a plan to release a new version.

Phil1988, 2025-04-10:
@martin-badin I just watched it.
Cant you go to the move&extrude menue?

[image]