Phil1988/FreeDi #66: [Feature request] Theming and color scheme selection for LCD

theclue, 2025-01-13:
In the process of doing a "total conversion" to FreeDI, I'm planning to paint the chassis and to remove the original QIDI Logo (spoiler alert: I'm also drawing a couple of logos :D).

Thus, a nice touch (but a huge task, I know) would be to adding some kind of theming subsystem for the LCD that would let you customize:

- The text strings (for easier translations provided by the community)
- The color scheme
- (icy on the cake) The logo in the about screen and/or the boot screen

IMHO, this means be able to externalize these assets so they are pushed from FreeDi service trough Moonraker rather than embed them into the LCD firmware itself. This one, the community would able to tune these from simple cfg files on the Moonraker side. Ofc, this applies only for the first two points in the above list.

It won't be very useful on the long run (and totally pain in the neck for the devs) to leave these embedded.

I know, it's huge task :| but I'm curious about your opinion here. It's not a very added values in terms of technical capabilities, it's more a symbolic switch I would say...

Phil1988, 2025-01-13:
Yes I have thought about this already.
And yes you are right, that is indeed a very labor-intensive feature.

Its not a "no" (yet?), but currently my focus is (in this order):

- Fixing bugs
- Adding features for everyones daily use
- Improving the current Wiki-Guide
- Doing an instructional video for the installation
- Adding the "Unleshing the beast" section
- Adding strech features that only a fraction of users might use (like this here)

Btw: What do you mean by "The logo in the about screen"? 
I dont see one.

Phil1988, 2025-01-13:
Please check this again, because I am pretty sure I did not use the klipper logo 😄 

Meanwhile I have taken a closer look at the feasibility of implementing a theming feature for the display firmware. 
Unfortunately, after careful consideration, I believe this feature will most likely never be implemented.

There are essentially three ways this could be achieved, but all of them come with significant drawbacks:

1. **Transmitting images at runtime as .bin files**
This would involve converting images to RGB565 format and storing them on the host as *.bin files. These files could then be transferred to the display at runtime. However, this would cause noticeable delays every time a page is loaded on the display, as the images would need to be re-sent each time. Given that a lot of effort has been put into ensuring a fast and responsive display, Im unwilling to compromise on this responsiveness. User experience (this includes quick reaction times) are the top priority for this project, and I do not want to introduce lag that could degrade this.

2. **Storing Images on an SD Card**
Another approach would be to store the RGB565 images on an SD card and replacing the embedded images only when an SD card is present. While this would allow faster image loading, it would require an SD card in every display. This is not a viable option for this project, as it would introduce additional complexity and hardware dependencies that is quiete the opposite if why I started FreeDi 😅
Even more concerning, it would demand extensive testing to ensure reliability across various setups, which I simply cannot justify investing in for such a feature.

3. **Loading images into RAM**
One could load the RGB565 images into RAM at the start and use them from there as needed. Unfortunately, the displays RAM is simply not large enough to hold the required images for theming. Even if we could load a few images, the limited RAM would quickly become a bottleneck.

Given the constraints and priorities of this project, I believe adding a theming feature would compromise both performance and user experience for the cost of extensive testing on my side. 
Therefore, I won’t be moving forward with this feature.

If I miss a good option to implement it, let me know.
I invite you to start a topic in the "Discussion" section :)

theclue, 2025-01-13:
Ok, got the point.

I guess the only alternative is to give the user who want to change the colors to compile the display firmware by himself from the source code.

Or to provide several different precompiled firmwares with different color schemes (just two themes come in mind to me: 1- QIDI original and 2- FreeDI). Ideally, if the firmware itself is compiled using a CICD pipeline as a Github build, it won't be a manual procedure by definition...

If you ask me, black and fucsia :D

Phil1988, 2025-01-13:
An "QIDI original" theme will never be provided by me.
I reached out to them already and was not allowed to use their logo.
I dont want to get in trouble and respect it.

You can not compile the firmware using anything.
Nor can you automate anything.
I think you believe the firmware creating process is way easier as it actually is :D

If you or anyone interessted piches a design draft I like, I will think about it ;)