Phil1988/FreeDi #77: [Bug] DFU MODE

Ecaze81, 2025-01-14:
Hello, I share here, together we will find my mistake:
So I am at the moment where I have to Updating the "THR" tool head

I install katapult, no worries, everything works fine.
I do the makemenu config, the make everything is fine.
when I do: ls /dev/serial/by-id/* I find my card.

The problem comes next when I do:
Press and hold the „boot“ button.
Press and release the „reset“ button.
Release the „boot“ button.
Now prepare everything for flashing the katapult bootloader by this:
i get : 
``` mks@mkspi:~/katapult$ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-Klipper_rp2040_95A67A96550B4F53-if00
mks@mkspi:~/katapult$ sudo mount /dev/sda1 /mnt
mount: /mnt: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.```
 
I guess I'm not in DFU mode, there is a video of the procedure on discord:
https://discord.com/channels/686160672211730511/1034422697976934411/1328605530800914532
If you have an idea, I'm interested.

Phil1988, 2025-01-14:
Hey @Ecaze81 and welcome on board! 😊  

First of all, I’d like to clarify that I don’t fully agree with the `[Bug]` tag here.  
With katapult as your base and the procedure you’re following to flash the toolhead, you're deviating from my described approach.  

That’s totally fine, but I just want to avoid the impression that your issue stems from the officially described procedure here. I hope you understand!🙏 

---

### Now, regarding your issue:  
Since you already have katapult on your toolhead, you could skip the DFU/boot mode procedure entirely.  
I’m not sure why it’s already there since I lack the necessary context.
I will not describe how, because this leads to a more complex guide.

Currently, you’re attempting to put your **mainboard** into DFU mode, but this needs to be done for the **toolhead**.  
What you're showing isn’t required at any point.  

Please open the back of your toolhead and put it into DFU mode.  
Here’s an image to help you locate the buttons:  
[image] 

---

For further steps, I kindly ask you to follow my guide.  

For example, you’re trying to flash the tool head by running:  
```
sudo mount /dev/sda1 /mnt
```
While my guide describes this:

```
sudo cp ~/FreeDi/Toolhead_RP2040.uf2 /dev/sda1  
```
It’s not a big deal, but you should be aware of what you’re doing if you choose to deviate from the guide. 🙂

Ecaze81, 2025-01-14:
I'm at work, but I think I understood with a friend, we'll test it this evening, but actually my poor command of English is the cause of the problem
I just understood with a friend who speaks English, it is not the motherboard that I have to put in DFU but the other card

I am really really sorry

Phil1988, 2025-01-14:
No problem mate!
You are welcome :)

I hope everything will work if you test this evening :)

Ecaze81, 2025-01-14:
Everything works perfectly
Thanks

closed 2025-01-14 by Ecaze81