Phil1988/FreeDi #79: [Bug]

Caretaker61, 2025-01-15:
**Describe the bug**
Hello,
I am new to this and have recently purchased a QIDI-X-MAX 3 and to date I cannot work with it.
QIDI support has given me countless tasks and already sent me two new emmc, but the printer refuses to work.
I flashed FreeDi v1.30 onto the emmc yesterday and get the following display:
Klipper meldet: ERROR

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-289-g08a1c9f12-dirty
MCU(s) which should be updated:
mcu: Current version v0.10.0-530-g3387a9c2-dirty-20221121_013639-mkspi
MKS_THR: Current version v0.10.0-530-g3387a9c2-dirty-20221210_215217-mkspi
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'MKS_THR': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u

What do I have to do to reinstall the system or how I can solve this problem?

**Base (please complete the following information):**
 - X3seriesLCD firmware version: ??
 - OS: FreeDi_v1.30_X-Max3_Armbian24.2.0_Bookworm6.6.17
 - Source: FreeDi image
 - Mainboard: X-6 V1.0
[moonraker.log](https://github.com/user-attachments/files/18429233/moonraker.log)

Phil1988, 2025-01-15:
You missed this [part](https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#flashing-the-mainboard-mcu)
[image]

Fortunately its a very easy thing to solve for you :)

Please report back how it worked out ;)

pmbroth, 2025-01-15:
@Phil1988   I was typing when your message popped up.    :-)

Phil1988, 2025-01-15:
> @Phil1988 I was typing when your message popped up. :-)

first!

Caretaker61, 2025-01-15:
Done, but the same error. The file is changed to X_4.CUR

Phil1988, 2025-01-15:
I can only help if you give me more information

I don't think it's the same error.
You probably getting the MCU reported as Klipper
v0.12.0-289-g08a1c9f12-dirty.
And you probably don't read further and flashed the tool head, did you?

Please read the whole guide mate 🙏 
There is a reason I took my time to write this down 😃

Caretaker61, 2025-01-16:
Hi Phill,
sorry, but I didn't have a good day yesterday.
I followed your instructions and now also flashed the motherboard mcu and the tool head.
Now I get this on the LCD screen of the printer:
[image]

and in Moonraker this:
Klipper meldet: ERROR

mcu 'MKS_THR': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

I restart the printer and wait 2 minutes and restart them, but nothing changed.
I tried to get the device ID via SSH, but I get the following output:
mks@mkspi:~$ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory

Caretaker61, 2025-01-16:
I tried to read out the error using your instructions:

mks@mkspi:~$ sudo systemctl stop FreeDi.service
[sudo] password for mks:
mks@mkspi:~$ cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Moonraker host address: http://127.0.0.1:80
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 1.549 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,CEB33401F645C95A,33554432-0
Display firmware: v1.30
Printer model: x-max3
Printer volume: 328x326x322
Klipper version: v0.12.0-289-g08a1c9f12-dirty
Software version: v1.30 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
self.lcd.data_update()
data.klipper_state != self.printer.klipper_state
klippy error
_NewConnectionLostPage: len: 1 data[0]: ee
ActiveScreen: = ConnectionLostPage
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()
self.lcd.data_update()

I cancelled the command after a while because the same message kept appearing.

Phil1988, 2025-01-16:
Its not a display or FreeDi software issue.
Your toolhead is not ready or incorrectly configured in klipper.

Please show me what you get at the Machine-> System Loads tab in Mainsail.
Should be something like this.
[image]

Caretaker61, 2025-01-16:
I did not find the system load in mainsail, see attached picture.
[image]
[printer.zip](https://github.com/user-attachments/files/18438662/printer.zip)

Phil1988, 2025-01-16:
Thanks for that!

Unfortunately, I cannot understand the problem. 
Since you could previously see the toolhead (MKS THR) as klipper `v0.10.0-530-g3387a9c2-dirty-20221210_215217-mkspi`, you should now be able to see your freshly flashed toolhead at 
```
mks@mkspi:~$ ls /dev/serial/by-id/*
```

For me there are currently 2 possible reasons:
1. The toolhead has hung up (boot cycle with power off solves this)
2. The wrong firmware was flashed to the toolhead.

Please give me feedback on this, because it looks to me as if information is missing here.

Caretaker61, 2025-01-16:
Phil,
what did you mean with "boot cycle with power off solves this" ?

Caretaker61, 2025-01-16:
I install the firmware again, as you wrote in your instruction.
Firmware "Toolhead_RP2040.uf2" is installed

Caretaker61, 2025-01-16:
same result

pmbroth, 2025-01-16:
You either did not flash the onboard mcu with the micro sd card, and/or you did not do the flash to the tool head.  Both need to be the version that Phil has in this directory

https://github.com/Phil1988/FreeDi/tree/master/mainboard_and_toolhead_firmwares/v0.12.0-289

Caretaker61, 2025-01-16:
I flash the mcu again with V_4.bin. After flashing the file was renamed to V_4.CUR. Is that correct or an error?
During the flashing I see no reaktion on the Printer-Screen, is that also correct?

pmbroth, 2025-01-16:
Yes that is correct.   You have to do the tool head..  Make sure it is in flashmode utilizing Phils instrucitons

Phil1988, 2025-01-16:
> Phil, what did you mean with "boot cycle with power off solves this" ?

I mean a shutdown with the switch shut off (so that no component gets power).

> I flash the mcu again with V_4.bin [...] V_4.CUR

Its X_4.bin and X_4.CUR so just a typo I guess, but yes: if its renamed to *.CUR it was flashed.

Now focus on the tool head.
Show us screenshots and provide more information please.
"same result" and "it doesnt work" does not help us to help you :)

You are doing something wrong, but if you dont tell us **exactly** what you do, we cant see what is wrong ;)

EisbacherG, 2025-01-16:
Hello Phil.
I get the same error.
i have done everything as it is described in your wiki.
MCU is flashed with downloaded X_4.bin
Toolhead is flashed with downloaded .uf2 file.

[image]

Caretaker61, 2025-01-17:
Hello,
the printer is now running without error messages.
Yesterday I updated the MCU and the tool head again and disconnected the printer from the power supply overnight. This morning I switched the power supply back on and started up the printer and, lo and behold, everything was ok!!!!!
Many thanks to Phil and everyone else who helped me.
I am a complete novice in this area and I hope I haven't annoyed you with my ignorance
Can I contact you again if I have a ‘problem’ with the printer or software?
Again many thanks!

Phil1988, 2025-01-17:
> [...] disconnected the printer from the power supply [...]. everything [is] ok!!!!! Many thanks to Phil and everyone else who helped me. I am a complete novice in this area and I hope I haven't annoyed you with my ignorance Can I contact you again if I have a ‘problem’ with the printer or software? 

So it really was only a problem flashing the mcu and toolhead and then a simple power cycle solved it 👍 
Im not annoeyd at all and invite you to contact me again if you face any problems :)
Im glad its working for you now! 
Happy printing :)

Caretaker61, 2025-01-17:
Hi Phill,
I wrote to you this morning to say that the same problem as described here had been fixed, but it occurred again after the printer was restarted.

3rd result ls /dev/serial/by-id/* 

[image]

4th result lsusb

[image]

5th result lsblk

[image]

Please find attached the printer.cfg file
[printer.zip](https://github.com/user-attachments/files/18452826/printer.zip)

Phil1988, 2025-01-17:
Problem is that your toolhead is not recocnized.
Have you seen a partition with 128MB in the past before you tried to flash it?

Something like this, but maybe named as sdb1:
[image]

It seems to me like to be a hardware issue.
Have you had any problems with the toolhead in the past?

My best guess right now is either:
your toolhead is defective
or the cable to the toolhead has a loose contact/is faulty.

Besides that you have a wifi dongle in usb mode like this
https://github.com/Phil1988/FreeDi/issues/23

Lets do an install to install the wifi dongle, but I would be very surprised if this solves the issue with the toolhead:
```
cd ~/FreeDi && ./install.sh
```
Show me the while output please :)

Caretaker61, 2025-01-17:
Phill,
I try to install the WLAN with "cd ~/FreeDi && ./install.sh" and get the question "Do you use the stock Mainboard? (y/n)"
What must I answer?

Caretaker61, 2025-01-17:
I answer with "y". The installtion is finished, I checked the the cable to the toolhead and I restart the printer.
But I get the same message:
Klipper reports: ERROR

mcu 'MKS_THR': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Can I restore the printer to its factory settings?

Phil1988, 2025-01-17:
The answer "y" (yes) was correct for you.

You forgot to 
`Show me the whole output please :)`
`Have you seen a partition with 128MB in the past before you tried to flash it?`
`Have you had any problems with the toolhead in the past?`
I am still missing the answers to my questions, so I dont know how I should help you :(

You can roll back to your stock system if you want to, but I am sure, that the hardware issue will remain.
Should I tell you how to go back to stock or do we proceed?

rosenrot00, 2025-01-17:
> I answer with "y". The installtion is finished, I checked the the cable to the toolhead and I restart the printer. But I get the same message: Klipper reports: ERROR
>
> mcu 'MKS_THR': Unable to [...]

Besides the output that is still missing and if you like to proceed - issues are solved more easily when they are broken down into smaller pieces. Let's go step by step to identify your issue.

Scenario A:
1. Remove any attached USB drive
2. Start your printer just normally
3. Provide the output of "lsusb", "lsblk" and "ls /dev/serial/by-id/*"

Scenario B:
1. Remove any attached USB drive
2. Open your toolhead and keep the BOOT button pressed
3. Start your printer just normally
4. After 10s release the toolhead's BOOT button
5. Provide the output of "lsusb", "lsblk" and "ls /dev/serial/by-id/*"

There is likely to be an issue with your hardware it is either a bad flash of your toolhead or a bad connection.

pmbroth, 2025-01-17:
When I updated my toolhead, at first it did not work properly.  For me, I unplegged all usb from the board except the toolhead and then it worked properly.  I know it should have worked piror, but I kept having the flash iussues until I unplugged the other usb devices.

Phil1988, 2025-01-19:
> I unplugged all usb from the board except the toolhead and then it worked properly. I know it should have worked prior, but I kept having the flash iussues until I unplugged the other usb devices.

This is not necessary. You may have accidentally tried to flash a USB drive or the wifi dongle instead of the actual toolhead. However, unplugging all USB devices can indeed help reduce the chances of making mistakes.

@Caretaker61 Any news on that from your side?

Caretaker61, 2025-01-19:
Hi Phil,
sorry, for the late answer.

After cd ~/FreeDi && ./install.sh I forgot to make a scrennshot from the output.

Before flasing I had seen the partion with 128 MB and I had this problems before:

[image]

At the moment I try your scenario A and here the date from

mks@mkspi:~$ lsusb
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mks@mkspi:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0  28.9G  0 disk
├─mmcblk1p1  179:1    0   256M  0 part /boot
└─mmcblk1p2  179:2    0   6.9G  0 part /var/log.hdd
                                       /
mmcblk1boot0 179:32   0     4M  1 disk
mmcblk1boot1 179:64   0     4M  1 disk
zram0        251:0    0 455.5M  0 disk [SWAP]
zram1        251:1    0    50M  0 disk /var/log
zram2        251:2    0     0B  0 disk

mks@mkspi:~$ ls /dev/serial/by-id/*~
ls: cannot access '/dev/serial/by-id/*~': No such file or directory

I will try now the scenario B

Caretaker61, 2025-01-19:
Result from scenario B

mks@mkspi:~$ lsusb
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 2e8a:0003 Raspberry Pi RP2 Boot
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mks@mkspi:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda            8:0    1   128M  0 disk
└─sda1         8:1    1   128M  0 part
mmcblk1      179:0    0  28.9G  0 disk
├─mmcblk1p1  179:1    0   256M  0 part /boot
└─mmcblk1p2  179:2    0   6.9G  0 part /var/log.hdd
                                       /
mmcblk1boot0 179:32   0     4M  1 disk
mmcblk1boot1 179:64   0     4M  1 disk
zram0        251:0    0 455.5M  0 disk [SWAP]
zram1        251:1    0    50M  0 disk /var/log
zram2        251:2    0     0B  0 disk

mks@mkspi:~$ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory

Phil1988, 2025-01-19:
> I had this problems before:
> 
> [image]
> 

I absolutely cannot understand that.
I have never used Klipper v0.11 and this screenshot shows that you are completely wrong.

You seem to be mixing up different guides or don't know exactly what you're doing. 
**Where does the OS come from, how did you configure it?**

And from now on, please don't do anything other than what we're discussing, because otherwise it's too time-consuming.

Next steps:
On scenario B your toolhead is in boot mode.
Now please do this:
```
sudo cp ~/FreeDi/Toolhead_RP2040.uf2 /dev/sda1 
```
Shut the system down (sudo shutdown) **and shut it off - switch off or cable disconnected**.
**Wait a few seconds** (at least ~10s) and turn the printer on.

Then do this again:
`ls /dev/serial/by-id/*`

And attach your printer.cfg (as txt file or zipped)
I will be gone for the next ~10h so please be patient if you are still getting errors.

Caretaker61, 2025-01-19:
I done it.
mks@mkspi:~$ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory

[printer.zip](https://github.com/user-attachments/files/18468485/printer.zip)

Caretaker61, 2025-01-19:
Phil,
here's an overview of what I did before I installed your software.
I got the printer, connected it and started it up. I then connected to the printer via a browser and found the programs Klipper, Moonraker with the note “update available” in the “Software” section of the “Fluiid” program interface. As it turned out later, I had carried out the updates by mistake.
I then searched the internet for solutions and found this “GitHub - billkenney_revert_qidi_software_ revert klipper and moonraker to working versions from qidi's repo, and install the latest working version of fluidd” and did it.
QIDI sent me an emmc chip on which the operating system was stored. I exchanged the emmc chips and tried to install the latest firmware provided by QIDI support. When that didn't work either, I installed your software.

rosenrot00, 2025-01-19:
I try to help a bit, we can solve this. Could you please post the output of everything after you run the commands? you can export these data from the console tool you are using.

In order to start from a common ground, did you install the FreeDi image on your emmc? If not, please do that. Backup your current emmc and overwrite it with the FreeDi image.

When you are in scenario B, you have to flash the toolhead

`sudo cp ~/FreeDi/Toolhead_RP2040.uf2 /dev/sda1 ` isn't 100% what is needed as you need to redirect the path of the uf2 file to where it actually is. Could you verify that the Toolhead_RP2040.uf2 file is somewhere on your system? If so, please change the command accordingly and report the output of this command here.

Caretaker61, 2025-01-19:
Hi @rosenrot00,
you are german? Then write it in german, please

All steps that you provide, I done it thsi morning.
Do I repeat them ?

rosenrot00, 2025-01-19:
We can switch to german but if you have discord we should continue our discussion there. Could you share your discord name?

The ping-pong here takes too long.

Caretaker61, 2025-01-19:
I must create one, wait a minute.

Caretaker61, 2025-01-19:
I don not know how discord works, but we can try.
My discord name is Caretaker61_82592

Caretaker61, 2025-01-20:
Hi Phil,
rosenrot00 try to help me, but it failed.
He thinks that the tool head is defective because he could not establish a connection to it.

Caretaker61, 2025-01-20:
Phil,
I would like to restore the printer to its original state in order to complain about it to QIDI.
Can you please send me the klippers.bin? Version 0.10 was installed.

Phil1988, 2025-01-21:
> My best guess right now is either: your toolhead is defective or the cable to the toolhead has a loose contact/is faulty.

If rosenrot00 and I share the same opinion, this is most likely the case. I'm sorry to hear that, and I hope Qidi provides you with a replacement.

Find my latest commit and make sure to download them as "raw file":
https://github.com/Phil1988/FreeDi/tree/master/mainboard_and_toolhead_firmwares/v0.10.0-530_stock

Keep us updated on how it goes.

Caretaker61, 2025-01-22:
Good morning Phil,
the reset almost worked, only the screen shows your start screen, nothing else. Is there a reset routine for that too?

pmbroth, 2025-01-22:
Have you tried to  access using a browser, based on the ip?  I had this happen, when I had some bad configurations, the display froze on the first screen on the machine.  I then went in via IP and saw the error message, I then used mobaxterm to correct my typo.

Phil1988, 2025-01-22:
As I'm currently don't have access to my files, I can give you one a link to billkenney 

Please download this:
https://github.com/billkenney/update_max3_plus3/blob/main/800_480_max3_4.3.15.tft

Save in on a SD card.
Switch the printer off.
Insert the SD card to the backside of your screen panel.
Turn the printer on.

Wait until it's flashed.

Switch the printer off.
Remove the SD card.
Turn it on.

Phil1988, 2025-01-23:
@Caretaker61 the one that I have is a v4.3.12 so it doesnt change much.
Simply use the one from billkenney and please report back.

closed 2025-01-23 by Phil1988

Caretaker61, 2025-01-24:
Hi Phil,
I try to install the version from billkenney, but it does not work. After switch one the printer, only the screen shows your start screen, nothing else.

Caretaker61, 2025-01-24:
Yes, with mobaterm

pmbroth, 2025-01-24:
If you try to access the machine say with orcaslkicer,  does it connect, and can you look at the device tab, does it show an error?    Review the configurations (Printer.cfg, Macro and mainsail)

Caretaker61, 2025-01-24:
Yes, for the toolhead, but nothing for the screen on the printer.
mcu 'MKS_THR': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Phil1988, 2025-01-24:
As far as I understand this is the current status:
- Stock QIDI OS is flashed on the EMMC
- mainboard MCU is flashed with stock Klipper v0.10.x
- toolhead ist flashed with stock Klipper v0.10.x
- LCD Firmware is still FreeDi and not stock.
- 
@Caretaker61 ist this right?

Make sure your SD card is formatted and it hat the *.TFT file on it that I gave you.
Make sure you have downloaded it as raw file.
To be on the save side, format the SD card with SDFormatter (on windows) and turn adjust option ON

Caretaker61, 2025-01-24:
Yes, you are right

Caretaker61, 2025-01-24:
that´s the file

[image]

pmbroth, 2025-01-24:
looks good

Phil1988, 2025-01-24:
format the sd card like this with sdformatter if not already done (turn format size adjustment option ON)
https://github.com/Phil1988/FreeDi/wiki/Requirements#3-microsd-card

Caretaker61, 2025-01-24:
how long does the updating take?

Phil1988, 2025-01-24:
flashing the display starts instant after power on.
It takes ~ 2min

Caretaker61, 2025-01-24:
Phil,

it's not working.

[image]

[image]

Could the motherboard be defective?

Phil1988, 2025-01-24:
No, the problem is you, not the mainboard :D

That's exactly why I wanted the picture.
Insert the SD card into the slot of the display please ;)

Phil1988, 2025-01-25:
Thanks, mate :)

I'm happy to see you're still learning and haven't lost your motivation. 
Hardware failures are always "unfortunate," and I'm sorry you're dealing with that.

I heard that your printer never worked properly and you got it recently directly from QIDI. I hope they will soon help you out. For me the support was very good and I hope you experience the same.

Just for everyone stumbling across this: 
If your printer has hardware issues, report it to QIDI and ask for replacements. 
Don't convert it to FreeDi, as this can't be resolved. 

As the landing page states: "You're facing hardware issues. FreeDi is software and can't do magic - even if it sometimes still feels magical for me ;)"