Phil1988/FreeDi #81: [Problem] MKS_THR unable to connect

Phil1988, 2025-01-17:
> Hello Phil.
> I get the same error.
> i have done everything as it is described in your wiki.
> MCU is flashed with downloaded X_4.bin
> Toolhead is flashed with downloaded .uf2 file.
> 
> [image]
> 
> 
>  when i try to compile the files by myself and flash MCU and Toolhead with the self made files, there is the same error.
> 
> I don't know, if it is necessary, but when booting, the printer screen shows the picture from the x-max3 and not the x-plus3.
> 
> [image] 

 _Originally posted by @EisbacherG in [#79](https://github.com/Phil1988/FreeDi/issues/79#issuecomment-2597114628)_

Phil1988, 2025-01-17:
Hello @EisbacherG,

to track this down and help you better I need your assistance ;)

1. Can you please **make sure** the toolhead is currently flashed with my provided firmware?
It is included in FreeDi already but in case you have overwritten it with the self compiled one, you can find it here: [Toolhead_RP2040.uf2](https://github.com/Phil1988/FreeDi/blob/master/mainboard_and_toolhead_firmwares/v0.12.0-289/Toolhead_RP2040.uf2)
  I kindly ask you to do this again (I know you did it before, but we need to make sure on this critical point).
  While you do the tool head flashing, please show me  a screenshot of the **whole command inputs you do and the output you get** (the whole process please).

2. Then please shut the printer down (`sudo shutdown`) and turn it off completely (switch off / or power cable unplugged)
   This is to make sure the toolhead is not stuck in a boot mode state.
   Turn it on again after at least ~10s shut off.

If the error still persists, lets track it down better. Show me screenshots of the results of:

3. `ls /dev/serial/by-id/*`
4. `lsusb`
5. `lsblk`

This information will hopefully make it easier to understand what the current situation is :)

EisbacherG, 2025-01-17:
Hello Phil.
Thanks for trying to help.

Yesterday, when i typed the command: ls /dev/serial/by-id/* i got a 16 char long alphanumerical number.
wie auch schon im Forum beschrieben, habe ich bei einer Nachschau 
As already described in the forum, when I looked in the directory "~/dev/serial/by-id/" I only found a link with the name "usb-Klipper_rp2040_12345-if00".

If I now enter the command "ls /dev/serial/by-id/*" after the system has started, I receive this entry as feedback.

[image]

I have now changed the entry in the "printer.cfg" file accordingly and everything seems to be working.

[image]

Thanks again for your offered help and for your time and the great project.
I now need to add the camera and make the adjustments you described in the "Post install and config section".

Phil1988, 2025-01-17:
> Hi Phill, I wrote to you this morning to say that the same problem as described here had been solved, but...

Please dont mix the threads :)
Please report it on the one you started. It looks similar, but It can have different causes :)

Phil1988, 2025-01-17:
> usb-Klipper_rp2040_12345-if00

Haha! 
So your serial is seriously usb-Klipper_rp2040_**12345**-if00 ? 😄
I never even took in consideration this might be a valid serial number !

Glad this was really just that simple - (but I am still confused, as it normally is a 16 digit hash)

closed 2025-01-17 by Phil1988

rosenrot00, 2025-02-24:
@Caretaker61 we found a solution to this issue you reported. Last thing I heard from you was that you are about to file an issue to Qidi.

Look here for the solution: https://github.com/Phil1988/FreeDi/issues/165